xenomorphales/xenomorphales #1: Cadre administratif

astralien3000, 2015-05-28:
Bon, on va décider ici si on prend un cadre administratif ou non.
Il faut que la décision soit prise pour septembre.

Elioty, 2015-05-28:
On a évoqué par mail que l'on créerait une association si on trouvait un partenaire financier. Mais je me demande quand même si ça ne faut pas le coups d'en créer une même sans partenaire financier.
Premier point (sans vouloir mettre le feu à quoi que ce soit), on peut donner la propriété du matériel et la propriété/paternité du code (plutôt paternité vu qu'on s'oriente vers du libre et l'open source) à l'association. En cas de clash futur, ça devrait limiter les problèmes et arranger ceux qui restent, tout du moins pour le matériel acheté en commun pour l'association. (Pas très plaisant comme point je sais mais bon, on ne sait jamais)
Second point justement à propos du matériel et plus précisément de son achat, ne peut-on pas être exempté de la TVA en achetant au nom d'une association ?

Ce sont les deux points auxquels j'ai pensés jusqu'à maintenant pour soutenir l'idée de former une association (enfin, quoi que le second est plus une interrogation).

astralien3000, 2015-05-29:
Pour la paternité du code : +1
Pour le financement, en gros, on mettrait tous une certaine somme dans l'asso, c'est ça ? (si on a pas de sponsors, ou que c'est pas assez, bien sûre).
En cas de clash, le problème est que vu le nombre qu'on est actuellement, je pense pas que l'asso survive... Donc je sais pas si ça réglerai le problème. Après ça dépends des personnes qui nous rejoindraient éventuellement...
Pour la TVA, j'ai regardé, normalement on peut avoir des récup sur les achats uniquement si on vends des produits, ce qui ne sera pas notre cas (c'est pour ça qu'eirbot n'a pas de récup aussi d'ailleurs).

Autre point : il faudra rédiger les status, etc... , et c'est chiant ! XD

Elioty, 2015-05-29:
Pour le matériel, on peut très bien utiliser du matos perso donc là pas de problème, chacun achète son truc, l'embarque dans le robot, s'en sert et puis le récupère ensuite. Par contre si on a besoin de quelque chose que personne n'a réellement besoin sur le plan personnel, je pense que ça serait une bonne idée de partager les frais entre les membres (ou utiliser les fonds des sponsors dans un premier temps si on en trouve) et d'acheter dans ce cas, au nom de l'association. Vous en pensez quoi comme ça ?

C'est vrai qu'en cas de clash à trois, ça ne sera pas facile à régler... Mais effectivement si au moins deux ou trois personnes nous rejoignent, ça rajoute bien de la sécurité.

Tant pis pour la TVA... Et c'est vrai que c'est logique d'ailleurs, la TVA est une taxe pour le consommateur final, si on ne transforme pas les produits achetés pour les revendre, on est un consommateur final, donc on doit payer la taxe.

Dernier point, la paperasse effectivement, je n'ai pas était au bureau donc je ne sais pas trop ce que ça représente mais par écrire status, tu entends juste dire qui fait prèz, qui fait trésorier et qui secrétaire ? Après il y a le dossier à faire à donner à la préfecture, aucune idée s'il est long ou pas mais ce n'est en principe à faire qu'une fois donc on devrait pouvoir surmonter cette tâche, non ? :yum:

astralien3000, 2015-05-31:
Alors, pour les status, il faut aussi décrire deux trois choses en plus, genre la façon dont sont désignés/élus les membres du bureau, et deux trois autres trucs. C'est sûre que c'est à faire qu'une fois, mais ça reste un bon gros dossier à remplir.

Si non ça me va plutôt bien comme ça.
Par contre, on va sûrement devoir faire des achats assez rapidement (encodeurs, moteurs, ...), il va falloir s'arranger sans pour le moment.

Elioty, 2015-07-24:
Maintenant que le nom a été décidé, il faudrait s'y mettre pour faire la paperasse pour créer l'association.

astralien3000, 2015-10-04:
On est paru au journal officiel !

closed 2015-10-04 by astralien3000