xenomorphales/xenomorphales #2: Caméra

astralien3000, 2015-05-28:
Il faut faire une choix et des tests de caméra

nbelin, 2015-05-28:
Vu à l'instant au Leclerc de Saint-Chamond !

"Webcam E C 305 avec microphone clip"
6.90 €
Webcam USB, focus de 8cm à l'infini (hmm à tester !)
Capteurs CMOS 1.3M (penser à comparer les technos)
640x480 / 30FPS (résolution suffisante je pense, à tester aussi, et le frame rate est respectable)
USB2 + Jack (micro)
Mode vision nocture ?!

astralien3000, 2015-05-28:
Je pense pas que la compatibilité posera problème.
Par contre, si tu veux faire de la reconaissance de motifs, il faudra bien faire gaffe à a résolution.
En effet, il faut pouvoir détecter le motif à ~3 m, pour une balise de 8 cm de large.
Faudrait faire un calcul avec l'angle d'ouverture de la caméra pour savoir combient ça fait.

nbelin, 2015-05-29:
Je pense que ça va être très compliqué d'obtenir les infos sur les angles d'ouverture des webcams, mais je testerai avec celle intégrée à mon PC je pense pas que les différences soient énormes.
En effet, si à 3m la balise fait 10 pixels sur 10 pixels, c'est pas reconnaissable !

Elioty, 2015-05-29:
Pensez aussi à étudier le cas du [module caméra](https://www.raspberrypi.org/products/camera-module/) officiel de Raspberry (il peut faire du 1080p30, fréquence suffisante donc je pense, la question de la résolution restant à être évaluée selon le motif et l'angle d'ouverture). Au moins, aucun problème de drivers en principe avec lui et possiblement pas mal de support sur le net.

astralien3000, 2015-06-06:
Comme l'a dit Clément dans l'issue #4, on pense à trouver une caméra grand angle, mais il se posera le problème de la déformation des lignes sur les côtés.
OpenGL possède un algo pour ça. Il suffit de le calibrer avec un damier (si ça te rappelles des choses, Nico). 
Après les questions sont : est-ce que l'algo sera assez rapide ? Est-ce qu'on trouvera une camera avec les bonnes caractéristiques ? Est-ce que le prix sera abordable ?

Elioty, 2015-06-06:
À noter que l'algo de "redressage" des lignes devra tourner sous OpenGL ES (Il n'y a pas de support d'OpenGL sur Rapsberry Pi). Juste un point de détail, je ne pense pas que ça pose problème mais il ne faut pas l'oublier quand même.

nbelin, 2015-06-07:
Une caméra grand angle ne devrait pas poser trop de problème...... Mais on sait pas trop encore !
OpenGL permet de le faire ? Ok, sinon OpenCV aussi hein ^^ (ça me rappelle l'issue #8 où parle des dépendances, et moi OpenCV je trouve ça quand même pratique !)
La calibration risque d'être un peu chiante à faire, mais l'algo de correction de distorsion ne devrait pas être trop coûteux (c'est quelques divisions et multiplications pour chacun des pixels de la caméra, la formule sur http://docs.opencv.org/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html).
Mais une caméra grand angle aura moins de "pixel par area" vers les bords d'image (si vous voyez ce que je veux dire ?) du coup une moins bonne résolution sur les bords.

Elioty, 2015-06-07:
OpenCV fonctionne sur le CPU je suppose. Utiliser OpenGL ES permettrait peut-être d'avoir de meilleure performance si ça peut bien utiliser complétement les 12 unités de calcul du GPU.

nbelin, 2015-06-07:
Alors je sais qu'il y a un module GPU sur OpenCV, mais je sais pas comment il fonctionne.... Et ce n'est pas ma prio pour l'instant :)
Et là je dois avouer que dernièrement j'essaie des algos un peu plus poussés d'OpenCV qui seraient très chiants à implémenter.

Elioty, 2015-06-07:
Pour faire de la reconnaissance dans une image, je ne pense pas. Mais pour le traitement genre pour appliquer des algos de rotation, distorsion, miroir, floutage et autre, je pense que oui vu que c'est quand même fait un peu pour ça à la base un GPU (et donc les APIs qui te permettent de t'en servir). Mais c'est vrai que tu peux juste implémenter ton algo avec OpenCV sur CPU pour l'instant comme tu as l'habitude de faire puis on verra ensuite pour paralléliser (on a quand même 4 coeurs CPU) voire externaliser certains traitements sur le GPU.

nbelin, 2015-06-07:
Et puis comme on dit, faut pas chercher à optimiser avant même que ça ne marche !
J'aimerais bien avoir un truc qui marche sur vos vidéos le weekend prochain.

Pour en revenir au thème de la discussion (quand même !), en effet les caméras n'ont pas un angle très important, il faudrait des caméras avec un angle de vue de 90° de manière à ce que les 2 caméras qui sont dans les angles aient chacune une vue globale de la table (allez, 80° si vous voulez) pour bien trianguler, et faire en sorte que la 3ème caméra ait la meilleure vue possible pour ajouter de l'information aux deux premières caméras.
L'angle de vue dépend de la lentille utilisée et de la distance focale (et peut-être d'autres facteurs), et j'ai peur que les webcam classiques soient peu configurables.

Ah et quand j'ai vu la qualité des images produites par la webcam de l'option robot, j'ai halluciné. C'est quasiment pas flou  ! Du coup faudra peut-être reconsidérer ce que j'ai dit sur des motifs seulement horizontaux...

astralien3000, 2015-06-07:
Après on a fait des vitesse faibles. Je sais pas si on peut en tirer des conclusions.

nbelin, 2015-06-07:
Tout de même je suis pas sûr que ma webcam aurait fournit ces images ! Bon, bonne nuit !!

astralien3000, 2015-06-08:
Salut !
J'ai trouvé un BON plan !
http://www.genius-europe.com/fr/produktdetail.php?ID2=108&ID=31&ID3=805
 Une webcam avec un angle de 120° ASCII paraît. Jusqu'à 30 fps. Et une résolution max de 1920x1080 px.

http://www.amazon.fr/gp/offer-listing/B008JIYGDM/ref=sr_1_2_olp?ie=UTF8&qid=1433788356&sr=8-2&keywords=genius+webcam&condition=used
On peut l'avoir d'occaz pour 21€ !!

Donc je m'en suis achté une pour tester, et la première chose que j'ai fais : 
https://www.dropbox.com/s/ljzckoniy7z3ufo/2015-06-08%2019.57.46.jpg?dl=0
(je peux pas m'en empêcher !)
Niveau driver sous linux j'ai pas eut de problème (à par peut être un, je pouvait pas avoir une résolution autre que 640x480...)

J'ai fais quelques tests pour voir si on pouvait la faire tenire dans les balises (parce qu'elle est large quand même) : 
Si on considère la diagnoale d'un carré de 8*8 (ça fait 11.3) : 
https://www.dropbox.com/s/f17cniyvux8dunm/2015-06-08%2020.18.34.jpg?dl=0

Si on essaie de la mettre quand même dans du 8*16 : 
https://www.dropbox.com/s/q8354fbjmsbyt5h/2015-06-08%2020.18.56.jpg?dl=0

Et enfin, j'ai fais des test d'angle de vue (en horizontal et en vertical) : 
https://www.dropbox.com/s/73bg64l5r4ew400/2015-06-08%2020.13.02.jpg?dl=0
https://www.dropbox.com/s/wyxkweljqkn37hk/2015-06-08%2020.04.48.jpg?dl=0

J'ai pas encore calculé les anggles que ça fait, mais ça a l'air pas mal, il y aura peut être même moyen de voir la balise dans l'autre coin de la table.

Vous en pensez quoi ?

nbelin, 2015-06-08:
Excellent ! Le prix n'est même pas exorbitant ! C'est dommage que tu ne postes pas d'images capturées par la caméra. Ça donne quoi ? Les couleurs sont belles, les images sont floues si tu bouges trop vite ? etc etc !

Sinon en passant, là je fais une détection de lignes horizontales et sur les images d'une des vidéos où t'as le bazard du local en arrière plan, j'arrive à détecter principalement que les lignes des balises. Bon t'as toujours des compromis : trop permissif tu détectes des lignes un peu partout, pas assez permissif et tu détectes plus forcément tes balises.
Je peux pas encore trop dire si les lignes horizontales sont vraiment une bonne idée (j'aurais peut-être imprimé des bandes légèrement plus fines ?), mais pour l'instant ça a l'air exploitable si je couple ça à la détection des couleurs.
Faut encore que je trouve comment les coupler correctement, mais bon ;)

astralien3000, 2015-06-08:
Un aperçu : 
https://www.dropbox.com/s/4xa1c3fvx4wnp97/2015-06-08-211323.webm?dl=0

J'ai pas de balise chez moi par contre, c'est bête. En tout cas pour la luminosité, ça a l'air de bien s'adapter. On pourra faire plus de test le weekend, ou un soir, si j'ai le courage de trainer mes fesses jusqu'à Eirbot. Mais bon que pense-tu de cet échantillon ?

nbelin, 2015-06-08:
Haha merci beaucoup pour la vidéo et l'effet larcen ! ;)
Non non déjà que je culpabilise de pas bosser autant que vous, alors si en plus vous passez toutes vos soirées au local pour tourner des vidéos, ça va pas aller !

Hmm je crains un peu que les algos marchent pour une caméra, et ne marche plus pour une autre caméra (par exemple la ta caméra a l'air de moins bien supporter les objets en mouvement que la logitech...)

Et sinon Claire me pose une question toute bête...pourquoi on fait pas des balises lumineuses ? D'accord c'est un peu chiant à faire (souder pas mal de LEDs par exemple ?), mais vous en pensez quoi ?

Elioty, 2015-06-08:
J'y avais pensé aussi à l'idée du "phare" lumineux :yum: Si tu penses que ça peut faciliter le travailler de l'algo (i.e. un algo plus simple et/ou plus rapide), pourquoi pas.

nbelin, 2015-06-09:
Je sais pas trop... il faudrait des lumières qui forment un motif (spatial et/ou temporel) reconnaissable. Et je pense que pour pouvoir se baser sur la luminosité, il faut pouvoir faire en sorte que la caméra ait un shutter/temps d'exposition fixe (pas sûr que ce soit configurable sur beaucoup de webcams...). Et puis suivant le thème de l'année, ils seraient capables de mettre plein de spots lumineux dans la salle, sans compter les flashs des appareils photos...

astralien3000, 2015-06-09:
Mouai faire des motifs avec des leds... T'aura pas de contours déjà, du fait soit de l'éblouissement de la caméra, soit du fait qu'une led est un petit spot. Tu va avoir un nuage de points à la place de vrai motifs. Et en plus l'emmerdement des balises lumineuses est proportionnelle au nombre de leds qui la composent.

Si non pour le fait que ce soit flou, faudrait refaire un test à Eirbot, parce que pour le coup, il y avait moins de luminosité (je pense que ça joue un peu). En plus la caméra était mouvante, et les vitesses de déplacement étaient plus grandes que lors des tests du weekend. Donc l'un dans l'autre, c'est vrai qu'on peut pas vraiment comparer...

astralien3000, 2016-10-17:
@nbelin, tu peux mettre un lien vers la caméra et la lentille pour les balises ici ?

nbelin, 2016-11-14:
Caméra : https://www.amazon.fr/Raspberry-Pi-1080p-Module-Cam%C3%A9ra/dp/B01ER2SKFS/ref=sr_1_1?ie=UTF8&qid=1479155728&sr=8-1&keywords=pi+camera

nbelin, 2016-11-14:
Lentille : https://www.amazon.fr/Universal-Objectif-t%C3%A9l%C3%A9phone-smartphone-Samsung-Noir/dp/B01G2WVJW2/ref=sr_1_17?ie=UTF8&qid=1479156151&sr=8-17&keywords=lentille+smartphone (à vérifier)

nbelin, 2016-11-14:
Batterie : https://www.amazon.fr/Poweradd-Chargeur-t%C3%A9l%C3%A9phone-Adapteurs-Appareils/dp/B00N5LP35I/ref=sr_1_1?ie=UTF8&qid=1479158339&sr=8-1&keywords=smartphone+batterie+5000+mah

closed 2016-11-21 by astralien3000