xenomorphales/xenomorphales #3: Filtre à particules

astralien3000, 2015-05-28:
Discussion sur le filtre à particule

Elioty, 2015-05-28:
Je l'ai dit par courriel mais je le note ici afin d'éviter de l'oublier.

Donc concernant le filtre à particule, de ce que j'ai compris, c'est qu'on récupère toutes les infos de tous les capteurs possibles et imaginables (enfin, surtout de ceux à notre disposition au final) et on détermine la position et l'angle du robot en fonction de l'état t-1 et de comment le robot a dû se déplacer pour arriver aux valeurs données par les capteurs à l'instant t.

On avait également évoqué une histoire de pondération de l'importance que l'on attribue à une source d'information. À part, considérer ou totalement ignorer l'information, je ne vois pas trop comment ça pourrait marcher entre les deux.

Néanmoins, si BigBrother fonctionne bien, on connaitra la position (plus ou moins approximativement) des robots sur la table. À partir de ça et de la position du robot à l'instant t-1, on pourrait adapter de manière dynamique l'importance des capteurs de distance. C'est à dire que si la demi droite sur laquelle un capteur de distance fait sa mesure est en collision avec un robot*, on l'ignore, sinon, on considère sa valeur. Histoire d'être sûr d'éviter que l'algo se mette à diverger (bref, pour essayer d'éviter l'application de la loi de Murphy). Vous en pensez quoi ?

*les robots étant simplement représentés par leurs coordonnées (x,y), on considérera, pour le calcul de collision entre la demi droite et les robots, que le robots sont des disques (tout comme dans le système de collision dans le path finder ou le système de collision dans l'asserve pour arrêter le déplacement en cas de danger).

astralien3000, 2015-05-28:
L'algo a son propre système de pondération des capteurs. ^^
Par contre, je me rend compte que faudra pas chômer si on veut tout les systèmes opérationnels...
Ca fais pas mal de boulot ! Surtout que c'est pas ma priorité pour le moment.

Elioty, 2015-05-28:
Hum, d'accord, je peux bien imaginer que l'algo est capable de pondérer lui même l'importance de chacune des valeurs s'il se rend compte par lui même qu'un capteur dit de la "merde". Néanmoins, si on connait la position des robots, ne serait-ce tout de même pas un bon point d'ignorer les capteurs de distances qui ont un robot sur leur trajectoire de mesure ?
Parce qu'à ce moment là, on sait pertinemment que ces mesures seront erronées et ne connaissant ni la forme du robot se trouvant sur la trajectoire, ni son angle, on ne peut pas "adapter" la valeur "attendue" pour tel ou tel capteur.
Donc autant l'ignorer pour être sûr que l'algo ne la considère pas et limiter le risque qu'il se plante. Parce que je suppose que si une majorité de capteurs donne des valeurs erronées mais consistantes entre elles, l'algo va diminuer l'importance de celle qui sont en fait correctes ? Ou je me plante totalement sur mon hypothèse de comment l'algo de pondération fonctionne ?

nbelin, 2015-05-31:
Je suis d'accord avec Elioty. Tu peux faire l'algo que tu veux, plus sont propres les données en entrée et mieux il fonctionnera.
On travaillera effectivement en 2 dimensions, je pense qu'intégrer une 3ème dimension ne serait pas un cadeau !
Pour ce qui est de la pondération des informations, je prends par exemple le cas de BigBrother : si une caméra te dit que le robot est en (x1, y1) et que l'autre caméra te dit (x2, y2) avec deux fois plus de confiance, tu peux imaginer dire que le robot est en ((2*x2 + x1)/3, (2*y2 + y2)/3). Je me rends compte que mon exemple est pas top parce qu'en vrai une caméra te dira sur quel axe est le robot avec une bonne précision normalement, mais sera moins précise quant à la distance qui la sépare du robot.

astralien3000, 2015-05-31:
Je dis pas le contraire, quoique les algo de type filtre de kalman et filtre à particules sont quand même faits pour rendre des données bruitées un peu plus précises. Mais on vas pas faire exprès de prendre des données crades pour autant.

C'est juste que je vois pas pourquoi on prendait pas les données GP2 quand un robot est proche, surtout que les GP2 sont faits pour ça. En gros, d'après ce que j'ai compris, vous avez peur que si BigBrother dit que le robot adverse est à 30cm devant et que le gp2 voit un truc à 20cm devant, ça fasse diverger l'algo.
En effet, si le robot est à 30cm, le gp2 le verra à 15-20cm selon sa forme/taille/etc...
Sauf que l'algo est fait pour ça !
La question n'est pas : Où est le robot adverse ?
La question est : Si le robot adverse est là, est-ce que mes données capteur correspondent à la situation ?

Et là, pour l'histoire du GP2, il y a plus de problème. En effet, dans l'algo, il y aura un cas où le robot est à 30cm devant. On compare avec les données de BigBrother : C'est bon. On compare avec les données GP2, en prenant en compte un taille moyenne : C'est plutôt bon (avec un certain poucentage). Donc cette possibilité a une grande chance d'être bonne, elle sera sûrement sélectionnée pour le résultat.

Donc voila, le problème n'est pas que un problème de pondération des données, et il n'y a pas besoin de virer les données GP2 quand un robot adverse est devant soi.

nbelin, 2015-05-31:
Hmm je pense pas que Clém parlait de ça. Il pensait à l'algo qui fusionnerait les données caméra et GP2 pour avoir une position plus précise du robot sur la table, en considérant que le GP2 te donne la distance à un bord de table, auquel cas il ne faut pas que l'algo considère que le GP2 te donne vraiment la distance au prochain bord de table puisque ce dernier est en train de voir un robot. (en tout cas je l'avais compris comme ça)

Elioty, 2015-05-31:
La taille moyenne et le certain pourcentage de marge d’acceptance restant à déterminer.

Mais il y a aussi la situation où le GP2 voit "à travers" le robot adverse (robot à pattes et la trajectoire du GP2 passe entre ou le robot est plus petite que la taille moyenne et la trajectoire du GP2 passe à côté), mais dans ce cas de figure là, je pense que l'algo pourra simplement considérer le GP2 comme retournant une valeur faussée et l'ignorer.

Dans tous les, je pense qu'on a assez discuter sur ce point. Donc on fera l'implémentation du système tel que Loïc le décrit. Et après de nombreux tests, s'il y a des problèmes, on implémentera ma petite option supplémentaire (qui ne devrait pas prendre des semaines à implémentait de toute manière) pour ignorer un capteur de distance si on sait pertinemment qu'il y a un robot devant.

astralien3000, 2015-05-31:
Erf, les GP2 sont pas faits pour ça par contre...
Faut considérer qu'à 1m, on peut oublier leur valeur, c'est pas génial pour voir les bords de la table.
Il faudrait plutôt les capteurs que Clément a trouvé à 150€ pour ça... Et encore vu la précision de 2cm, je sais pas s'il faut pas mieux un télémètre laser plutôt, quite à prendre un capteur cher. (L'idée de RCVA est sûrement bonne, mais il faut dépenser pas mal de sous pour faire la même chose.)

Et puis, pour revenir aux GP2, ça revient au même, si tu connait la position du robot adverse avec BigBrother, et que tu connais la distance qui te sépare de ce robot avec le GP2, ça te donne une info sur ta propre position aussi, moins précise qu'avec un mur, mais bon...

Par contre, oui, l'histoire du robot transparent peut poser problème. Mais c'est rare.

Elioty, 2015-05-31:
Beh il me semblait que l'idée était de peupler le filtre à particule d'un maximum de capteurs, et de se positionner nous (c'est le but du filtre à particule non ?), pas détecter les robots adverses ou les éléments de jeux avec. Après si tu limites l'utilité des capteurs de distances (peu importe le type) à la détection des autres robots, on devrait avoir BigBrother pour ça déjà (on ne détecte pas les éléments de jeu là par contre). Ça ne nous empêche pas d'en mettre quand même au cas où BigBrother ne soit pas encore au point pour cette année et comme on l'a dit aussi hier, plus il y en a, mieux c'est et au moins, quand on voudra s'en servir, ils seront déjà là.

astralien3000, 2015-05-31:
En fait, pour que le filtre à particules ne diverge pas, il faudra aussi qu'il essaie de localiser les éléments de jeu et les robots adverses. Parceque pour que l'algo marche, il faut que tout les éléments mobiles du monde soient inclus.
Si on limite la fusion des capteurs à la localisation de nos propres robots, il vaut mieux faire un filtre de kalman couplé avec BigBrother.
D'ailleurs on peut faires les deux en même temps ! C'est pas interdit, et ça fait de la redondance en plus !
Mais c'est une autres sujet.

Elioty, 2015-05-31:
Beh sinon on se positionne à l'odométrie de manière habituelle (plus éventuellement BigBrother en support) et on utilise les capteurs de distance sur les côtés uniquement lorsque l'on sait que l'on longe un mur (plus proche de l'idée d'origine de RCVA). Les capteurs de distance de face et de derrière servant à détecter les robots adverses (en redondance avec BigBrother) ou les éléments de jeu.

astralien3000, 2015-06-01:
Je préfère en effet.

closed 2015-06-01 by astralien3000