xenomorphales/xenomorphales #4: Test de balises

astralien3000, 2015-05-28:
Il faut qu'on filme plusieurs types balises avec des couleurs et motifs différents.
On pourrai le faire cette semaine ou la prochaine.
Comme caméra, je pourrai me procurer celles de l'option robot, et j'en ai une personelle qui est vraiment pourrie.
Donc déjà, faisons la liste des balises possibles !

Elioty, 2015-05-28:
Vous pensez quoi d'un cylindre de 8cm de diamètre et 8cm de haut (le plus grand qui rentre dans les contraintes donc) avec un motif 2D (comprendre pas un simple dégradé uniforme ou une couleur unie) sur sa circonférence ?

Comme caméra pour prendre des échantillons de test, j'aurai celle de mon smartphone si ça vous parait pertinent d'essayer. Sachant qu'elle prend des photos en 3264*1840 (16:9) ou 3264*2448 (4:3) et des vidéos en 1920*1088 à 19.715386 fps selon VLC.

astralien3000, 2015-05-28:
Ah ouai j'avais pas pensé aux téléphones !

Ouai, on prend des verres qui trainent dans eirbot (les mêmes qui ont été utilisés pour la RDS), ils font 8*8*8. Ensuite on imprime des couleurs/motifs sur des feuilles, et on les enroule autours des verre. 30 min au four, et c'est cuit !

Faut réunir les image des motifs à imprimer.

Elioty, 2015-05-28:
Ah beh oui, c'est vrai qu'on a déjà les "verres" de la RDS qui font les dimensions que j'ai évoquées. je les avais zappés tiens :yum: 

Reste plus qu'à déterminer les motifs effectivement mais je ne m'y connais pas ensuite à propos de comment ça va réagir avec OpenCV alors à part le quadrillage (qui est limite le motif le plus basique), je ne sais pas quoi proposer. Et encore faut voir pour détermine la cote des carreaux dans le quadrillage !! Et voir pour les couleurs aussi. Il vaut mieux un fort contraste entre les deux couleurs ou plutôt des couleurs exotiques relativement contrastées tout de même pour éviter d'entrer en "collision colorimétrique" avec l'environnement ambiant ?

nbelin, 2015-05-29:
Si on part bien sur une caméra couleur (ce qui est le cas de toutes les webcams je pense), je pense qu'il y aura deux stratégies d'approche pour détecter la balise :
- détecter les contours (donc le motif) à l'aide d'OpenCV par exemple, puis éliminer les faux positifs (à l'aide de la couleur, de la taille, etc...)
- détecter la couleur dominante de la balise, puis vérifier sur les zones trouvées que le motif de la balise est bien présent

Dans le premier cas, deux couleurs trop proches rendront impossible la détection du contour à mon avis.
J'aimerais faire des tests préliminaires, mais en effet pourquoi pas un damier vert/rose ?

Elioty, 2015-05-29:
Pour ton premier point à propos de la détection du motif, je vois un problème potentiel (peut-être que je me trompe ne connaissant OpenCV), il faut bien se rappeler que le motif sera sur la circonférence d'un cylindre et donc il sera perçu comme "étiré" sur les bords gauche et droite. À toi de nous dire si ça pose un problème pour la première solution.

Pour le second point, c'est là qu'il faut bien trouver des couples de couleurs pour éviter la "collision colorimétrique environnementale" afin d'élimer le plus rapidement possible le plus de faux positifs. Mais ensuite même problématique pour le motif que pour le premier point.

Sinon avec ton deuxième point, j'ai pensé à un autre motif. La balise fait 8cm de hauteur, on pourrait mettre une couleur unie sur les 2 premiers cm en haut et en bas et un damier avec seulement deux lignes de 2cm chacune au milieu. La largeur des carreaux étant à adapter pour tomber juste sur un diviseur du périmètre du cylindre. On pourrait donc détecter dans un premier temps les deux bandes de couleur unie en haut et en bas de la balise puis vérifier ensuite le motif entre les deux.
Sachant que les deux bandes unies pourraient avoir la même couleur sur toute les balises nous permettant de prendre une couleur "exotique" (ie peu présente dans l'environnement), les couleurs des motifs nous permettant ensuite de déterminer l'identité des différentes balises détectées.

astralien3000, 2015-05-29:
Pour le problèmes d'étirement, normalement openCV peux détecter un motif même s'il n'est pas face à la caméra, par contre, sur un cylindre (de cette taille en plus), c'est pas évident.

D'ailleurs pour la détection de motif, j'ai utilisé lors de mon stage 1A une lib utilisant OpenCV pour détecter des motifs de type "codebar 2D" (un peu plus grossiers que ceux qu'on peux trouver dans le commerce). Ca marchait pas mal du tout ! C'était plutôt stable. Et en plus, avec un seul motif sur une surface plane, on pouvait déterminer l'orientation de la surface ! Par contre, ces motifs seront peut être pas utilisable sur du 8*8... Mais on pourrai les mettre au moins sur une paroi de nos robots.
Le nom de la lib est ARToolkitPlus (AR pour Augmented Reality). Par contre il y a deux ans, elle était déjà plus en développement, difficilement trouvable, et j'ai galéré à la compiler. Mais une fois que ça marche, ça fait des merveilles ! A savoir que ARToolkitPlus est juste une surcouche à OpenCV normalement, donc c'est faisable avec OpenCV seul, mais il faudra travailler pour avoir le même résultat et la même stabilité !

J'aime bien ton idée des deux bandes sur les balises Clément.

nbelin, 2015-05-29:
J'aime aussi l'idée de Clément (et j'y avais déjà pensé aussi :D)
L'idée d'un code barre (ou tout motif à très haute fréquence et haut contraste) associé à une détection de couleurs peu communes pourraient très bien marcher je pense.
J'ai juste hâte de voir quelle taille a visuellement un cylindre de 8x8x8 pour une webcam classique à différentes distances.

ET nouveauté : j'ai pensé à un truc qu'on a vu en TD : l'apprentissage de background. Si chaque caméra apprend son background, la balise sera plus facile à trouver car forcément sur une zone qui diffère du fond. Ou alors s'inspirer de cette idée pour par exemple faire la différence de chaque image avec la précédente, et l'emplacement de la balise devrait apparaître clairement comme deux (ou plus en fonction du motif) segments verticaux. Plus la balise a bougé, plus ces segments seront épais. Couplé à du tracking, on devrait facilement suivre l'évolution de la balise !

Elioty, 2015-05-29:
Hum, dans un environnement statique, ton idée d'apprentissage de background me parait très efficace. Néanmoins, à la coupe, le fond risque d'être quelque peu agité entre les arbitres et les équipes autour de la table, sans parler de l'éventuel public (qui risque être bien dans le noir et flouté selon la caméra donc à voir si ça dérange quand même l'algo ou pas).
À voir/tester/comparer les perfs avec et sans l'apprentissage de background lorsque le background est justement agité.

nbelin, 2015-05-29:
C'est pour ça que je parle plutôt de m'inspirer de l'idée de background. Surtout que si le robot ne bouge pas au début, il pourrait être pris pour un élément du fond ;p
Toute façon c'est un truc de plus à tester !
Ce qui est certain, c'est que les images seront liées temporellement entre elles, et ça serait dommage de faire une détection de balise seulement spatiale (c'est à dire grâce à une image) et de ne pas la coupler avec une information temporelle (en utilisant les images passées).

Exemple de résultat attendu avec à chaque fois une différence pixel par pixel entre l'image en cours et l'image précédente : 
 - le fond bouge bizarremennt (typiquement le public) : ton image "différence" va avoir pas mal de pixels à valeur non négligeable éparpillés, sans structure, et aléatoires. Un filtre anti-bruit devrait enlever ça (par exemple une érosion/dilatation si tes souvenirs sont bons Loïc)
 - la balise s'est clairement déplacée : tu vas avoir une grosse bande nette de pixels à haute valeur dans ton image "différence", ajoutant une information à la détection de la balise
 - un arbitre au t-shirt blanc passe derrière la table : tu vas avoir une bande nette de pixels à haute valeur, il faut donc forcément de quoi éliminer ce faux positif.

Elioty, 2015-05-29:
Je n'avais pas pensé à un filtre anti bruit effectivement pour virer le public. Après pour les positifs comme les faux positifs, ne peut-on pas simplement passer l'algo classique de détection juste sur les zones où on a détecté du mouvement ?

Sur le fait que le robot soit considérer comme un élément du background, je ne vois pas le problème, le problème étant si la balise, elle, le devient. Et je pense que l'on peut simplement contourner le problème, on intègre systématiquement la zone où se trouvait la balise à la photo précédente dans la liste des zones à traiter par l'algo classique de détection. C'est possible sans trop perdre en perf ?

nbelin, 2015-05-29:
Je ne pense pas non plus que ce soit problèmatique que le robot soit considéré comme background. C'est juste notable !
D'une le background peut être un truc dynamique qui se met à jour régulièrement, de deux tu couples bêtement avec l'algo de détection et basta !
Question perf n'ayant jamais de traitement d'image sur Rasp je sais pas trop ce qui va être possible ou pas :/

Question procédure d'algo, un truc bâteau serait : 
 - première détection de la balise, on a une minute avant le début du match, donc c'est pas grave si c'est coûteux en temps
 - ensuite on prend la position de la balise dans l'image précédente, et on la cherche dans un voisinage proche dans l'image courante - voisinage à ajuster en fonction du frame rate, robustesse, distance de la balise à la caméra, etc etc...

La question de voisinage peut être une liste de positions à tester, liste à laquelle tu peux rajouter les endroits où l'image "différence" présente de grosses tâches.
Il faut aussi que la Rasp soit capable de repasser par une phase de recherche de balise si elle la perd, où si Murphy intervient quoi !

nbelin, 2015-05-30:
On peut raisonnablement penser que les images capturées seront un minimum floutées. Le robot se déplaçant toujours lattéralement par rapport à la caméra, je pense qu'il est nécessaire d'avoir un motif horizontal. Cela limitera la déformation de celui-ci même si le robot se déplace trop vite.
Pourquoi pas : 
 - bande supérieure de couleur COLOR1 de 2cm de hauteur
 - succession de bandes noires/blanches de 1cm de hauteur
 - bande inférieure de couleur COLOR2 de 2cm de hauteur
COLOR1 et COLOR2 peuvent être magenta et jaune par exemple, et les inverser en fonction de l'équipe du robot. On peut également les changer si on se rend compte que ce sont les couleurs choisies par le règlement de la coupe pour différencier les équipes.
Sous réserve qu'une bande de 1cm soit distinguable à 3m par une webcam...

Je pense juste à un truc : la balise de notre robot ne pourra être à la hauteur classique d'une balise, vu que cet espace sera réservé à la balise de l'autre équipe ?

astralien3000, 2015-05-30:
En effet, les balises sur nos robots seront 8cm plus bas.
En tout cas, on comptait faire les tests la semaine prochaine, si on a le temps de tout préparer avant. En gros, on a la semaine pour faire les motifs et les imprimer. Et samedi prochain, c'est séance video !
Au pire, même si on a d'autre idées debalises après, on poura refaire des tests, mais tu aura des données sur lesquelles travailler.

Elioty, 2015-05-30:
D'ailleurs oui pour les motifs, on en a bien parler aujourd'hui également. Tu te charges de les faire en vectoriel Loïc et je me chargerai de la mise à l’échelle et de l'impression ?

Concernant les balises, je pense que ça serait bien qu'elles soient toutes les quatre uniques (par le motif ou les couleurs, peu importe) afin de pouvoir les identifier. Je pense que ça pourrait aider si on pousse suffisamment loin la stratégie et la partie d'anticipation de ce que pourrais faire l'adversaire ! (Certainement pas pour cette année je vous l'accorde mais je pense que ça serait bien).
Ne penses-tu pas qu'il serait quand même mieux d'avoir toutes les balises avec les bandes haute et basse de la même couleur pour simplifier la "trouvaille" de balise par l'algo puis les bandes intermédiaires de 1, 1.33 ou 2cm de couleurs permettant l'identification de la balise concernée ?

nbelin, 2015-06-01:
Comme l'a rappelé Loïc, les balises de nos robots seront 8cm plus bas (à la place du mat, c'est ça ? de 35 à 42cm du sol ?). Je pense que c'est un critère suffisant de distinction entre les 2 équipes. Une petite formule donnera la hauteur de la balise en fonction de sa position dans l'image et de son éloignement. On peut même s'amuser à faire en sorte que la bande colorée du bas des balises adverses et la bande colorée du haut de nos balises soient toujours à peu près sur la ligne centrale de pixels de la caméra. Ça facilite sa détection, et ensuite il y a plus qu'à savoir si la balise se prolonge au dessus ou en dessous pour savoir à quelle équipe on a à faire :)

Ensuite on peut jouer sur l'alternance des bandes (commencer par noir ou par blanc par exemple) pour différencier les 2 robots d'une même équipe ?

J'ai fait quelques tests hier. Rien de très rigoureux, mais pourtant ça marchotait déjà pour une balise que j'ai coloriée sur un bout de papier. J'ai pas encore passé assez de temps pour exploiter correctement le motif d'alternance de bandes, du coup je m'en servais pas hier, mais je pense qu'il y a moyen !
(je pense en particulier à faire une détection de contours VERTICAUX pour détecter les bandes, et une détection de contours HORIZONTAUX pour ne PAS détecter les bandes ! (en effet, sur les tests préliminaires quand j'affichais les résultats de contours verticaux il y en avait un peu partout dans l'image, MAIS si j'enlève toutes les zones où apparaissent en plus des contours horizontaux, je pense que ça fait une bonne descrimination)

ps : attention entre le moment où ça va effectivement marchoter dans un environnement à peu près maîtrisé, et le moment où ça sera robuste même face à des changements radicaux de lumière, de fond, etc... il risque de s'écouler pas mal de temps ^^

Elioty, 2015-06-01:
Ah oui pas bête de se baser sur l'altitude de la balise sur déterminer m'équipe. On a donc ensuite besoin que de deux motifs différents, chacun devant donc être imprimés en double.

Et donc au final, il vaut mieux des bandes verticales ou horizontales ? Je demande juste par curiosité, je vous imprimerai vos motifs dans tous les cas.

Le ps me rappelle un air d'un autre système de détection et de localisation... :yum:

nbelin, 2015-06-01:
Alors tout dépend de la qualité de la Webcam, mais avec les premiers tests que j'ai fait avec la mienne, je peux te dire que si le robot se déplace un poil trop vite, la balise devient vite floue. Du coup comme le robot se déplacera toujours latéralement par rapport à la caméra (et non de haut en bas), je préconise les bandes horizontales (car même floutées ça reste des bandes horizontales, alors que des bandes verticales deviendraient un amat grisâtre).

Pour la sélection des couleurs, j'ai une idée d'expérience à réaliser : si on récupère suffisament de vidéos issues de la coupe, on pourrait s'amuser à construire un histogramme de la teinte des images, et ainsi déterminer quelles teintes sont les moins fréquentes. Sinon pour l'instant je pense que magenta serait une couleur intéressante.

Ah et point important : il faut vérifier dans le règlement qu'on ait bel et bien le droit d'utiliser la zone qui se trouve sous le porte balise de notre robot. J'ai peur que cet espace ne puisse être utilisé que par un système qui sert à détecter les robots adverses ? (du coup est-ce que si on fait un gros mat de 8cm de diamètre, coloré, serait accepté ?)

Elioty, 2015-06-01:
Oui effectivement en prenant en considération les déplacements possibles des robots, des bandes horizontales paraissent plus adaptées.

Pour l'idée des couleurs, tu pousses l'idée un peu loin là. Mais ce n'est pas bête quand même et ça n'empêchera pas Murphy d'habiller le mec de l'équipe contre laquelle on joue en magenta...

Pour le règlement (de 2015) :
[Section E.2 page 20]
> In any case the height of the main robot and the secondary robot must never exceed 350 mm. The emergency button is allowed to exceed this limit but must stay below 375 mm in height.
> For Eurobot Open this height excludes the beacon support, sensors and electronic circuits that can be placed below the beacon support.
> No components of the robot and playing elements manipulated at any moment by the robot should exceed 350 mm in height, in order not to disturb the beacons.

[Section E.4.ii page 23]
> The beacon support should at all times comply with the following specifications:
> - It needs to have an 80x80 mm square surface, located at 430 mm above floor level. This is where the opponent's beacon will be placed.
> - The surface of this platform has to be entirely covered with Velcro (the rough "hook" side)
> - The structure supporting the platform (mast) must stay within the vertical projection of this platform.
> - The mast can only host sensors and electronic circuits that also need to stay inside the platform's vertical projection.
> - The mast should be stable and must be able to support a weight of at least 400 g.

Il semblerait que rien n'interdise de mettre un cylindre (d'autant plus totalement passif dans notre cas) temps que ça reste dans la projection du porte balise (de 80x80mm donc) car le règlement parle de "the structure supporting the platform" ce qui est plutôt générique quand même. Aucune contrainte sur la couleur n'est donnée aussi donc ça doit être ok de ce côté là.

nbelin, 2015-06-02:
Les tests avancent et la détection des lignes horizontales commence à être exploitable je pense. D'ici quelques jours j'aurai peut-être mon tracking de balise sur un fond pas trop méchant ! Faudra voir le frame rate que j'obtiens, et tester sur des fonds plus complexes...
Puis je pense que ça sera pas simple d'intégrer la détection de plusieurs balises dans le champs de vision (par exemple quand est-ce que tu t'arrêtes ? tu en as vu 3, ça suffit ou tu persistes pour trouver la 4ème ?)

Elioty, 2015-06-02:
Beh dans l'idéal, il faudrait chercher les quatres. Mais la recherche devrait être relativement rapide si les balises et les caméras sont bien placées de telle sorte que les balises sont toujours à plus ou moins x pixels d'un axe sur l'image, non ? Le soucis étant si tu détectes les deux notre puis une adversaire, je pense qu'il est plutôt important d'aussi chercher la dernière (d'autant plus que c'est une posée sur un adversaire).

nbelin, 2015-06-02:
Si la détection d'une balise se passe très et est plutôt rapide, il n'y aura pas de souci. Cependant si c'est coûteux en temps, est-ce qu'on pourra se permettre d'attendre 150ms pour s'assurer qu'il n'y a pas d'autre balise dans le champ de vision ?
(je pense pas trop que ce point posera problème, mais sait-on jamais !)

En fait l'autre souci je pense sera si deux robots sont proches dans le champ de vision, et que les balises apparaissent confondues. Problème !

Elioty, 2015-06-02:
Ça c'est à tester et à adapter selon les performances obtenu donc.

Pour ton dernier point, la différence de couleurs ne te permet pas de détecter qu'elles sont proches ? Je veux dire que si le contraste est suffisant, on devrait bien voir un bord verticale (mais éventuellement flouté avec le mouvement). Mais dans ce cas ci (ou pire, l'alignement exacte), je ne pense pas que ça soit dramatique si une seule des trois caméras ne voit pas une balise.
Le vrai problème se présente lorsque deux caméras et deux robots sont alignés ou lorsque deux caméras et un robot sont aligné et qu'un autre robot cache le premier du point de vue de la troisième caméra. Murphy sera-t-il assez fort pour faire en sorte que cela se réalise ?
(reste aussi à voir les angles d'ouvertures des caméras et les angles morts induits)

astralien3000, 2015-06-05:
Bon, j'ai un début de balises : 
https://github.com/astralien3000/projet2016/tree/gh-pages/doc/balise

Si vous voulez en rajouter, c'est pas compliqué, vous installer inkscape, et vous modifiez une existante pour en créer une nouvelle.

nbelin, 2015-06-05:
Sympa ! Désolé j'ai pas trop eu de temps cette semaine, pas mal de papiers à faire. J'ai pas trop réfléchi aux balises du coup non plus.
La détection de motifs noirs/blancs est plus compliquée que simple couleur, mais je pense que c'est plus robuste à un changement de luminosité. Du coup je pense que c'est pas mal ça permet de coupler différents types d'informations. Vous pouvez donner vos avis et réfléxions ça aide à mieux voir le truc :)

astralien3000, 2015-06-05:
Je sais pas si détecter du noir/blanc sur un fond noir sera vraiment robuste...
D'ailleurs on aura du mal à faire un fond noir ou même sombre pendant nos tests de ce weekend.

Elioty, 2015-06-05:
Faites chauffer l'imprimante ! Enfin, je vais de ce pas là faire tourner pour qu'on puisse te prendre des photos demain Nico.
Sinon, une Raspberry Pi 2, ça chauffe un peu aussi lorsque ça tourne à gogo pour compiler des trucs haha.

nbelin, 2015-06-06:
La détection couleur ça marche bien, mais t'as des limites : 
 - Si la couleur que tu cherches est pas mal présente autour, tu peux ne pas t'y retrouver. Détecter 2 ou 3 couleurs "rares" peut permettre d'y remédier (quelle est la proba que ces couleurs soient présentes dans le même ordre autour de ta balise ?)
 - Si les conditions de lumière changent, ou ta webcam qui rappelons-le est pas de super qualité et peut peut-être s'amuser à changer quelques-uns de ses paramètres en live, ça peut modifier les teintes perçues.
La détection des bandes blanches/noires demande un peu plus de traitement, MAIS : 
Comme on se base sur une détection de contours, même si à un moment les teintes sont faussées par je ne sais quel éclairage, les contours eux seront toujours là. J'ai fait quelques tests pour ne détecter que des contours horizontaux, et c'est plutôt descriminant pour l'instant (en arrière plan j'ai le bazard de l'appart qui ne provoque que très peu de contours horizontaux, et la balise elle en provoque des propres).
Tant que j'aurai pas fait plus de tests avec plus d'images et que j'aurai pas plus avancé mes algos, je peux me tromper (et l'expérience montre que tu te trompes rarement quand on a deux avis différents :p), mais je pense que coupler une info couleur et une info contours nous préparera mieux aux différents aléas qui peuvent arriver.

Si on se rend compte que même au soleil ou dans la pénombre, les teintes de la balise sont toujours vues avec précision, ça sera peut-être la solution la plus simple.
En tout cas continuons de confronter nos idées, ça ne peut que nous aider à prévoir de futurs problèmes !
ps : en fonction des orages prévus ce weekend, j'avancerai plus ou moins :p

astralien3000, 2015-06-06:
On a fait les videos de test !
Voici les miennes : 
https://www.dropbox.com/sh/yb6wbfawun4bc72/AADbWfYhQQld9jt-zGG2h4pqa?dl=0

La balise sur le carton rouge est un peu trop haute, genre 1cm, mais c'est le plus proche qu'on ai trouvé.
Donc en gros, on a mis 3 caméra à peu près aux emplacements des balises fixes.
Il y a la webcam logitech de l'option robot à la zone de départ de l'équipe verte.
Mon téléphone est du côté jaune, coin du mur.
Le téléphone de Clément est du côté jaune, coin des fenêtres..

On s'est trompé en mettant certaines balises à l'envert... Mais bon, je pense que tu a déjà de quoi faire avec ça.

Elioty, 2015-06-06:
Pour mes vidéos, je me mettrai à les upload demain... Elles sont quand même vachement plus lourdes que celles de la webcam ou de l'iPhone de Loïc. Compter presque 200Mo par vidéo pour les miennes (Full HD oblige). Au moins, ça te fait différents paramètres pour voir si ton algo est robuste et tient la route :)
Tu vas certainement le voir avec les vidéos mais il y a pas mal d'angle mort donc on a évoqué avec Loïc d'avoir des caméras grands angles. Mais à voir si ça rentre dans les contraintes de volume et si tu peux aussi t'en servir correctement.

nbelin, 2015-06-07:
Super pour les images, merci ! Je vais pouvoir pas mal m'amuser avec tout ça ! J'espère que vous avez pas pris trop de temps pour moi :)
Aujourd'hui pas d'orage, du coup je passerai la journée dehors :p 
Déjà je vais voir si je peux considérer facilement une vidéo comme une entrée live de Webcam ?

Elioty, 2015-06-07:
De robots "autonomes"... :yum: 
Amuse toi bien avec celles de Loïc. Les miennes sont en route et question qualité (tout du moins résolution), elles envoient du lourd. Par contre, évite de trop écouter le son... On (enfin je surtout) sort pas mal de conneries haha.

Elioty, 2015-06-07:
Voila pour les miennes, avec quelques photos collectors haha
https://www.dropbox.com/sh/jo6egp955oarm4p/AAD4TAGUV1da7YSLuaNWgegPa?dl=0

nbelin, 2015-06-08:
J'ai un premier résultat où on peut suivre 2 balises (là flemme de traiter les autres couleurs pour l'instant :p). Le code est hyyyper moche, le temps d'exé pas terrible, mais au moins j'ai une piste qui peut marcher :)
(vous voulez le code ? où ?)

astralien3000, 2015-07-07:
Bon, les video sont là, s'il y a d'autres discussions, on peut créer une nouvelle issue.
Déjà qu'on a un peu débordé !

closed 2015-07-07 by astralien3000