xenomorphales/xenomorphales #5: Encodeurs

astralien3000, 2015-05-28:
Il faut de nouveaux encodeurs quadratiques pour Cérès !
Ceux que J'avais acheté coûtaient 40€, pour 512 périodes / tours (soit 2048 impultions / tours).
Mais ils sont quasiment impratiquable pour faire des roues codeuses externes. Si ils ont un contrainte mécanique sur l'axe, le cache peut se décaler par rapport aux capteurs optique, et faire qu'une partie du tour n'est pas vu.
Je les ai acheté là : 
http://www.mouser.fr/ProductDetail/Avago-Technologies/HEDS-5540A14/?qs=sGAEpiMZZMvy8cVzszrmR%252bE9KRCgbURJyLLqyTrO6yY%3d

Donc, je conseille d'acheter des encodeurs avec au moins l'axe intégré, et encore mieux, avec la roue intégrée (je pense que c'est très rare). Si on peut diminuer les emmerdes avec ça...

Elioty, 2015-05-28:
Oui, il faut absolument de nouveaux encodeurs pour Cérès, ce n'est pas possible d'avoir des "trous" dans le comptage. Les tiens avaient 512 incrément/décrément par tour alors, c'est bien ça ? C'est à dire sur une roue imaginons de 8cm de périmètre, ça ferait une précision de 8/512 = 0.015625cm ? :astonished: 

Pour le conseil que tu donnes, je suis parfaitement d'accord, essayons tout de même d'éviter les emmerdes (et Murphy) au maximum. Je vais essayer de chercher un petit peu de mon côté.

astralien3000, 2015-05-29:
Hehe, Les roues que tu vois sont pas celles qui toucheront la table ! XD
En fait ces composant sont ce qu'il y avait dans mes encodeurs.
Faut des encodeurs avec un axe et une roue en plus, pas un encodeur avec la coque en moins !

astralien3000, 2015-05-31:
Ce genre d'encodeur me paraît pas mal : 
http://www.robotshop.com/en/remote-controlled-armored-drink-carrier-3806-500B-5-24f.html

Elioty, 2015-05-31:
Ils m'ont l'air pas mal par contre faut faire attention à la taille oui... De l'encodeur en lui même mais aussi du diamètre de l'axe pour que ça se cale bien avec la roue fixée de l'autre côté.
Le premier, bien qu'ayant moins d'impulsion par tour, est plus court que le second mais peut surtout encaisser pas mal plus de force (d'où la différence de prix je pense). Faut faire le calcul de la précision qu'on peut avoir avec dans tous les cas selon le diamètre de la roue que l'on met dessus.

Elioty, 2015-08-18:
Ping @astralien3000 : on prend lesquels alors d'encodeurs qu'on puisse fermer l'issue (ou la taguer en "décidée mais en attente d'achat") et que je sache avec lequel traiter pour faire l'élec ? Tu as pu modéliser ceux que tu voulais et pu les faire rentrer dans la méca ?

astralien3000, 2015-08-18:
On prends ceux là : 
http://www.robotshop.com/en/6mm-rotary-encoder-1024-p-r.html

astralien3000, 2015-09-29:
Fait aussi !

closed 2015-09-29 by astralien3000