xenomorphales/xenomorphales #6: Moteurs

astralien3000, 2015-05-28:
Les moteurs actuels tournent à 120 rpm soit 2 tours par secondes (avec les roues de 8cm de diamètre, ça fait environs 30cm/s max). C'est peut être un peu léger...
Le couple est de 10kg-cm environs, ce qui fait qu'avec les roues, le robot pouvait développer une force d'un peu plus de 1Kg, ce qui normalement est suffisant.

Il y a aussi une contrainte mécanique forte. En effet, si on ne prend pas de renvoi d'angle, le moteur ne peut faire que 6 cm de long max ! Je peux vous dire qu'à cette taille, il va être difficile de trouver de meilleurs performances.

Mais juste avant la coupe (soit une semaine), je me suis procuré des renvoi d'angle et toutes les liaisons qui permetterai de solidariser un axe de moteur avec le renvoi et le renvoi avec la roue.
Donc on peut taper rapidement dans du moteur plus grand sans se poser de question !

Elioty, 2015-05-28:
Un renvoi d'angle, c'est le système à deux engrenages qui fait 90° comme sur les maxons d'Eirbot ?
Ça serait donc intéressant de se trouver des moteurs plus rapides car 30cm/s, c'est quelque peu lent. Une toupie à cette vitesse ne fera pas grand effet... (cf la toupie du robot aspirateur des 3A 2014 haha, ok c'était encore plus lent eux mais bon)

Pour les histoires de couple et de force développé, je sais que plus c'est élevé, plus les moteurs peuvent pousser de charge et "forcer contre les frottements" mais à part ça, je n'ai aucune idée de ce que représente vraiment telle ou telle valeur.

astralien3000, 2015-05-29:
Alors pour la force développée : Prenons un moteur qui a un couple de 10kg-cm. On lui attache une roue de 10cm de diamètre. On attache à la roue une ficelle (erf, une ficelle !!!) qui s'enroule autours de la roue. Et on laisse pendre un poids à l'autre bout de la ficelle (dans le vide, bien entendu). Le système pourra soulever une poid de 1Kg.

Pour un robot de 1Kg par exemple, on a pas forcément besoin d'une poussée de 1Kg, il faut juste pouvoir dépasser la force de frottement avec le sol (quand le robot est seul, ou quand il doit pousser une objet).

Pour le calcul de vitesse, c'est simple : 
v_robot (cm/s) = diametre_roue (cm) * PI * vitesse_motoreducteur (tour/s)

Petit exemple de calcul : 
Pour faire du 3m/s (vrooouuuummmm !)
Avec une roue de 8 cm
Il faut 300 / (8 * PI) ~ 12 tr/s

Pour le calcul de force, c'est un peu plus compliqué ^^ (je maitrise pas du tout les forces de frottement). Je pense que développer 1Kg de poussée environs est pas mal ! Mais ça peut être moins... D'ailleur il est peut être possible de faire une test sur Céres actuellement pour voir ce qu'on a besoin ! TODO ?

astralien3000, 2015-06-02:
Alors, d'après les résultats de #7, j'ai trouvé deux moteurs qui peuvent correspondre : 
http://www.selectronic.fr/motoreducteur-fixe-919d-501.html
(toujours avec une roue de 8 cm)
Couple : 7.6 Kg.cm
Vitesse : 4.4 tr/s -> 1.11 m/s
Prix : 30€

http://www.selectronic.fr/motoreducteur-mfa-970d-471.html
Couple : 7.2 Kg.cm
Vitesse : 4.7 tr/s -> 1.18 m/s
Prix : 40€

Faudra peut être faire plus de recherche, mais ceux-là ont l'air acceptable.

Elioty, 2015-06-02:
Question vitesse, c'est honorable :)
Question couple, c'est si grave d'en avoir un peu de trop ? Parce qu'ils n'ont l'air excessivement cher et je ne sais pas si ça fera baisser le prix tant que ça.

astralien3000, 2015-06-02:
Je pense pas que ça change grand chose. Et puis vaut mieux en avoir trop que pas assez parceque c'est des valeurs théoriques, donc en vrais je sais pas ce que ça vaut.

Par contre : 
http://www.dx.com/p/25ga-25mm-12v-300rpm-powerful-high-torque-dc-gear-box-motor-replacement-motor-grey-375953#.VW10n6jIWq4

V = 300rpm -> 1.25 m/s
C = 5~10kg -> à la louche, on sait pas trop...
Prix : 8.83€

Ils me feront toujours rêver ces chinois !

nbelin, 2015-06-02:
Je ne sais pas si c'est un paramètre important, mais si vous trouvez des moteurs qui peuvent facilement s'intégrer à des solutions mécaniques simples, c'est d'autant mieux non ?
(je pense aux moteurs cylindriques qui sont peut-être pas facile à utiliser...)

astralien3000, 2015-06-02:
En effet, c'est un paramètre important. Mais il semble que les vendeurs se sont mis d'accord pour pas proposer de trucs standards !! Genre un vendeur va proposer des moteurs et des renvoi d'angle, et aucun coupleur d'axe qui colle parfaitement aux deux parties (qui n'ont pas le même diamètre, d'axe bien sûre).

Et les rares moteurs qui s'intègrent facilement sont souvent des parties de kit de robot. Donc les caractéristiques de moteurs sont pourries (ils s'adressent à des débutant, ils vont pas leur vendre de la qualité !)

Ou si non l'autre solution est de concevoir soit même sa méca et de la faire faire par un industriel... Mais c'est pas le même prix. Quoiqu'il est possible qu'on en arrive là parce que pour le moment c'est pas la joie...

Elioty, 2015-06-02:
Ce n'est pas la joie concernant le coupleur d'axe effectivement. Mais pour l'intégration du moteur en lui même dont parle Nicolas, les deux moteurs sur Selectronic sont montés avec un châssis en acier avec quatre trous à vis donc facilement intégrable.

astralien3000, 2015-08-03:
Les MFA 970d471 ont été achetés !
(en soldes)

closed 2015-08-04 by astralien3000