xenomorphales/xenomorphales #7: Test pour le couple moteur

astralien3000, 2015-05-29:
Il suffit d'attacher une ficelle à Céres (sans ses roues), de la mettre sur la table, et de faire pendre la ficelle sur un des bords. En suite, il suffit d'attacher des masses au bout de la ficelle et de voir à partir de quelle masse le robot commence à bouger. Il faudra donner un peu plus de masse au robot que ce qu'il a actuellement (Cérès est un peu vide...), et on pourra aussi ajouter des éléments de jeu pour savoir la force qu'il faut pour les bouger.

Avec ça, on peut déterminer la force de frottement statique, qui est plus grande que la dynamique.

Elioty, 2015-05-29:
Ça parait simple à tester. Par contre pour les éléments de jeu et le fait de "combler" Cérès (sans arrière pensée aucune...), ça sera approximatif sachant qu'on ne connait ni les éléments de jeu de la prochaine coupe ni tout ce que l'on va embarquer dans Cérès.

astralien3000, 2015-05-29:
Bah, il faudra juste faire des tests dans le pire des cas ^^. Donc à la place de "combler" Céres, on va la "bourrer" (ok,  là ça deviens vraiment dégueu) avec des batteries par exemple.
Et pour les éléments de jeu, si on les a encore, on peux utiliser les triangles, qui ont vraiment une grande surface de contact avec le sol, ou même un foyer de l'an dernier, avec 3 triangles dedans. On aura du mal à trouver plus de surface de contact et plus lourd.

Elioty, 2015-05-29:
Oui, on fera ça avec un foyer avec trois triangles dedans, plus Cérès comblée.

Par contre, ça vient de me traverser l'esprit, on va bien tester la force requise pour déplacer Cérès avec des forces frottements induites par ce qui peut trainer sur la table. Mais concernant la non présence de force de frottement, comprendre par là une peinture avec une adhérence différente sur la table, comment peut-on faire pour essayer d'éviter tout problème de "glisse" ?
Il me semble que notre robot 2014 avec l'asserve en rapide dérapait/partait de côté un peu au démarrage. Avoir une plus haute fréquence d'asservissement est le seul moyen/suffirait pour réduire le problème ?

astralien3000, 2015-05-29:
Je pense qu'il suffit de régler la quadramp pour éviter une accélération trop rapide, donc un patinage artistique. La quadramp peut limiter la vitesse et l'accélération.
Mais de toute façon, les tables de la coupe sont beaucoup plus adhérentes que la plupart des tables. Donc on devrait pas avoir de problème de ce côté là. D'ailleurs c'est une raison supplémentaire de prendre large pour les frottement, parce que la table d'Eirbot est plus glissante.

astralien3000, 2015-05-30:
D'ailleurs, il faudra une balance ou tout autre moyen de mesurer un poids pour ce test.
Il en faudra une aussi pour le test #9.
J'en possède pas, tu en a une Clément ?

Elioty, 2015-05-30:
Je vais voir si je peux emprunter la balance de cuisine de ma mère, c'est la seule qu'on ait après le pèse-personne mais ce dernier sera loin d'être suffisamment précis :yum:

astralien3000, 2015-05-30:
Alors pour le test avec Céres, et un élément de jeu avec BEAUCOUP de surface, et un poids conséquent, il faut 2 batteries puissance, 1 batterie logique, et une autre (pas standard).
Clément les a emmené pour déterminer leur masses.

Si on me fait pas confiance sur le compte des batteries :
https://www.dropbox.com/s/uf6ooy7ccbmdewq/2015-05-30%2017.16.40.jpg?dl=0

Et la configuration du robot : 
https://www.dropbox.com/s/6nlnx5gohvdywkj/2015-05-30%2017.16.33.jpg?dl=0

Si on dimensionne  les moteurs pour posser ça, on aura pas de problème ! D'ailleurs on y a peut être été trop bourrin... À voir. Au pire on refera des tests.

nbelin, 2015-05-30:
Vous faites quoi à l'école à cette heure le weekend ? xD
Je confirme, si un robot est capable de pousser ce qu'on voit sur la photo, ça sera pas suffisant pour battre RCVA au démarrage, mais ça permettra de faire un usage plus que correct du robot !
Si j'ai compris l'expérience, la masse (en N) du bac à batteries devrait donner un bon ordre de grandeur de puissance pour la somme des moteurs ? (c'est à dire qu'il faut diviser par 2 pour un moteur ?)
Sauf que ça va nous donner la puissance requise pour DÉPLACER le robot. Comment tu convertis ça pour connaître la puissance requise pour aller à telle vitesse ou fournir telle accélération par exemple ?

astralien3000, 2015-05-30:
Le couple et la vitesse sont deux paramêtres différents (voir #6).
En gros, il faut trouver le moteur qui a le couple suffisant tout en ayant la vitesse suffisante.
En sachant que pour un même moteur, si on augmente le couple, on diminue la vitesse (en mettant un réducteur de rapport plus ou moins grand). Les moteurs qui ont beaucoup de couple ET une grande vitesse sont assez rares, et chères (genre des maxon) !

Si non, ouai, on peut diviser la force par deux pour un moteur.

Elioty, 2015-05-30:
**!!!Roulement de tambour!!!**
[image]
2955g pour les bigleux!

astralien3000, 2015-05-30:
Cool !
Tu pourra aussi peser une batterie puissance ? Pour savoir le poids qu'on a ajouté à Cérès et à "l'élément de jeu".
D'ailleurs, c'est sur quoi que tu hébèrege tes images ?

Elioty, 2015-05-30:
Pour la première : [image]
Pour la seconde : [image]

L'hébergeur : http://fr.tinypic.com/

astralien3000, 2015-05-30:
Merci ! J'ai pu mettre les images sur mon post !
Par contre, je viens de capter que j'ai oublié de te laisser buggybot pour le peser... Par grave, la prochaine fois peut être !
Faudrai faire les calculs, mais là j'ai la flemme. Je ferai ça demain.

Elioty, 2015-05-30:
Non mais pour peser Buggybot, je penserai à bien amener la balance la prochaine fois... :yum:

Elioty, 2015-05-31:
Lorsque tu feras les calculs Loïc pour la valeur du couple, il faudrait faire une page pour les spécifications de Cérès sur le wiki et mettre la valeur là bas. On pourra ensuite fermer la (première) issue :D

astralien3000, 2015-05-31:
Oups, je voulais savoir si on pouvait mettre deux Assignee sur une issue, mais c'est pas le cas...
Ouai, je le ferai. D'ailleurs je comptais aussi faire un tuto/une explication de notre test sur le wiki, donc deux pages à faire ^^.

Elioty, 2015-05-31:
Ça fera toujours de la doc en plus pour Eirbot :) Plutôt limité aux PMI par contre parce que la question ne se pose pas tellement avec les Maxons :yum: 
Dommage pour la limite à un seul Assignee. Mais tu voulais me faire faire quoi de toute façon ? haha Sinon faut faire une autre issue et bien décrire les deux tâches.

astralien3000, 2015-05-31:
Nan bein c'est juste qu'on a fait les tests/mesures ensemble ^^.
Après c'est pas important, maintenant on sait.
Par contre je sais pas si j'aurai le temps de faire les calculs ce soir encore...

astralien3000, 2015-06-02:
Bon, j'ai fais le calcul : 
En supposant qu'on aura une roue de 8 cm
m = 2.955 Kg
d = 8 cm
r = d/2 = 4 cm

Ctotal = m * r = 11.82 Kg.cm
Cmoteur = Ctotal / 2 = 5.91 Kg.cm

En N.cm : 
g = 9.81 N/Kg
Cmoteur = 57,879 N.cm

Elioty, 2015-06-02:
Ok, j'intégrerai ces résultats dans les spécs mécanique de Cérès.

Elioty, 2015-06-02:
Haha, j'allais le faire là mais Loïc est déjà passé le faire. Je clôture l'issue.

closed 2015-06-02 by Elioty