xenomorphales/xenomorphales #8: Architecture logiciel sur Raspberry 2

Elioty, 2015-05-29:
Discussion pour étudier l'utilisation de tel ou tel OS sur les Raspberry 2 ou de certaines techniques particulières, le tout dans un soucis d'optimisation des perfs.

La "référence" en terme de distributions : http://elinux.org/RPi_Distributions
Point à part, [la référence pour la config de la board](http://elinux.org/RPiconfig) (on pourra éventuellement overclocker légèrement le CPU des boards si besoin, en paramétrant l'OS en "on-demand").

La Raspbian est pratique du point de vue du support disponible sur le net. Néanmoins, je n'ai vu nul part de mention explicite d'un support "étendu" de l'ARMv7 (cad kernel et packages logiciels bien tous compilés pour ARMv7). De plus, elle est relativement volumineuse en terme d'espace disque (même s'il semblerait exister des variantes alléger, à étudier). Le gros avantage tout de même est sa base de packages compilés qui en contient plus de 35000.

Il y a ensuite Gentoo, qui d'après la "référence" a même une utilisation de RAM moindre que Raspbian (~23Mo vs ~30Mo)\*. On peut trouver [un tuto d'installation simple à suivre ici](https://wiki.gentoo.org/wiki/Raspberry_Pi_Quick_Install_Guide). Le tuto fait référence à ARMv6j-hardfp mais les dépôts contiennent bien les images en ARMv7a-hardfp donc il doit suffire d'adapter. Concernant la base de packages, elle doit être "très" grande, plus que Raspbian mais à le défaut d'être une base de packages source, il faut donc se taper les compilations (en terme de temps, pas de manip', le gestionnaire gère tout tout seul comme apt peut le faire).

Enfin, j'ai également remarqué Arch Linux qui a bien une version adapté à l'ARMv7-hardfp également avec [un tuto simple pour l'installation](http://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2). L'utilisation de RAM après installation n'a pas été mesuré ici par contre et la base de packages se limiterait à environ 9000 paquets, mais qui sont, contrairement à Gentoo, déjà compilés.

\*Vous allez me dire que sur 1Go, on a de la marge quand même... Mais si Raspbian consomme plus, c'est certainement parce qu'il utilise plus de services de base (donc plus d'entrées dans la table des tâches du scheduler) ou des services simplement plus demandeur en RAM. N'ayant pas de board sous la main pour l'instant pour l'essayer, je ne m'avancerais tout de même pas plus.

Après les OS, j'ai aussi regarder du côté du GPU et d'OpenCL. Et malheureusement, le GPU ne supporte pas OpenCL... Dommage, le GPU dispose de 12 units tout de même. Mais j'ai peut-être trouvé autre chose qui utilise [une méthode plus terre à terre](http://petewarden.com/2014/08/07/how-to-optimize-raspberry-pi-code-using-its-gpu/). Je ne sais pas si on aura le temps de bien étudier et utiliser cette méthode par contre (du moins pour cette année).

Voila, n'hésitez pas à partager vos avis ou les résultats de vos recherches.

astralien3000, 2015-05-29:
Alors déjà première remarque : J'ai utilisé raspbian cette année sur une raspberry 1. C'est pas mal; installation facile, énormément de logiciels dispo. Mais même si les binaires ont été compilés, ça ne veut pas dire qu'ils fonctionneront tous ! En effet, par exemple, Blender, qui est dispo dans les dépôt, se lance, mais fait une erreur de segmentation avant même d'afficher une fenêtre.

Autre point important : l'OS mettait 2 minutes à démarrer. C'est très long ! Surtout juste avant un match ! J'avais sûrement une config pourrie (j'ai pris celle par défaut), mais il faudra faire attention à ce point.

Et pour Gentoo, déjà que pour l'installer sur mon PC, ça m'avais pris une plombe, mais sur une rasp ! Le fait de tout devoir compiler me gêne un peu parce que ça prends beaucoup de temps. S'il y a une solution pour cross compiler, ça me plairais mieux. En tout cas, si c'est le choix qui est fais, faudra commencer à compiler tout de suite ! XD

nbelin, 2015-05-29:
Je viens juste m'incruster pour dire que Raspbian boot en 10 secondes sur Raspberry 2 :)
(et a priori fait un bon usage des 4 coeurs)
Et si t'as Raspbian sur une Raspberry 1, t'as 2 commandes à lancer (lien à retrouver) pour mettre l'OS à jour et le faire marcher sur la Raspberry 2, sans perdre aucune donnée.

Elioty, 2015-05-29:
Rien à dire sur Raspbian après le message de Nico.

Pour Gentoo, la différence c'est que sur PC, tu dois compiler toi même ton kernel (sans parler de sa configuration avant qui peut te prendre des plomb si tu veux vraiment bien fine-tune'r le truc). Or, pour Raspberry, le kernel est fourni directement par la Raspberry Pi Foundation donc installation relativement rapide (et il est en plus relativement à jour, en 3.18 actuellement). Il y a normalement toute une suite complète pour la cross-compilation sur Gentoo (mais faut s'en installer un du coups, mais on peut utiliser portage (le système de paquets de Gentoo) sur d'autre système que Gentoo donc on doit pouvoir faire ça sur n'importe quel Unix (y compris Mac OS X)) donc ça doit pouvoir se faire sans problème. Reste quand même à voir si, en utilisant bien les quatre coeurs de la Raspberry 2, les compilations ne sont pas trop longues, l'élément bloquant risquant d'être la vitesse de la carte SD ensuite.

Remarque : avec mon PC portable à 4 coeurs, la compilation du kernel n'avait même pas pris une demi-heure (en configurant bien le système pour lui dire qu'il peut utiliser les 4 coeurs pour compiler).

ps: il faut que je m'achète une Raspberry 2 en urgence !

nbelin, 2015-05-30:
Je veux juste préciser que j'ai Raspbian sur Raspberry 2 au boulot, et que je pourrais tenter d'exécuter quelques codes dessus pour voir les perfs'. Par exemple lancer quelques parcours d'image, des filtres, tout ça. Tu voudrais tester des choses en particulier ou ça va ?

Elioty, 2015-05-30:
Personnellement ce que je voudrais tester, c'est surtout les différents OS et comment fonctionne leur différent gestionnaire de paquets. Dans tous les cas, on en a discuté avec Loïc et il ne veut pas entendre parler d'administration système. Il veut juste un truc qui marche sur lequel il y a ses dépendances\* de dispo et sachant que j'ai reçu la formation la plus adaptée de ce côté là, je vais étudier les trois OS que je citais dans le premier poste et je vous partagerai une image (avec toutes les dépendances installées ou un script de setup pour automatiser leur installation) de carte SD toute prête pour flasher vos cartes. J'ai juste à commander ma carte demain avec une flotte de 4-5 cartes microSD et je me lance dans cette étude :wink: 

\* Pour la liste des dépendances, on a listé pour l'instant :
 - la compile chain (gcc, g++, binutils)
 - Qt5 core et xml
 - OpenCV
 - OpenDDS
 - Python (3 ou 2 d'ailleurs ?)
Et j'en ai peut-être oublié. Vous avez besoin de chose en plus ?

Elioty, 2015-06-05:
Bump!

Pas de retour sur la liste des dépendances nécessaires ?

astralien3000, 2015-06-05:
Prend python 3. Après si t'a déjà le 2 garde le.
Pour python encore, ajoute numpy.

Si non pour les lib C/C++, j'ai pas d'idée...

Elioty, 2015-06-05:
Ok, j'ai ajouté numpy à la liste. Pas besoin de scipy avec ?
Pour le C++, avec Qt5 core et xml, on a déjà de quoi faire et c'est plutôt complet/pratique à utiliser.
Pour le C par contre... Qui s'en sert encore de cet ancêtre ? :yum:

astralien3000, 2015-06-05:
J'ai pas utilisé scipy l'année passée... Donc je pense pas qu'on en aura besoin.
Bein en fait j'ai failli te dire boost tout à l'heure, mais Qt5 Core fait le boulot.

Elioty, 2015-06-05:
Ok, j'ai mis à jour le premier post pour que vous puissiez voir mon avancement et aussi pour lister les dépendances ou les trucs sympas que j'ai testés.
D'ailleurs dernière chose, y a-t-il besoin de avr-gcc/libc-avr et gcc-arm-none-eabi (et autres outils pour flasher les boards embarquées) ? Ou on les programmera uniquement depuis des PC ?

nbelin, 2015-06-06:
OpenCV n'est pas absolument nécessaire pour ce qui est des filtres d'image (ça sera ptet un poil moins performant écrit à la main), MAIS pour ce qui est de la lecture de flux vidéo ou de debug via l'affichage des images, c'est priceless j'ai envie de dire. (sauf si vous me trouvez un remplaçant)

Elioty, 2015-06-06:
Concernant le debug via l'affichage, je ne pensais pas mettre un serveur X sur les boards :/ tu peux t'en sortir autrement ? Genre en sauvegardant les images puis en les récupérant par ssh ?

Elioty, 2015-06-08:
Qt5 core fonctionne bien sous Gentoo, j'ai même pu compiler Aversive++v2014 (pas réussi à compiler la v14.11 par contre).

astralien3000, 2015-06-09:
Travail nocturne hier ?

Si non, il est possible qu'il y ai un petit croisement Qt4/Qt5, je suis pas sûre. (pas taper !) Ca compile sur ton ordi avec Qt5 ?

Elioty, 2015-06-09:
Je n'ai pas testé avec mon PC, il était tard et j'ai déjà testé 5 min sur ma Rasp donc je me suis arrêté là. Mais en gros, j'avais beau faire autant de `make mk/all.mk` et de `make update` que je voulais, `make sasiae` me disait très rapidement (avant même de compiler quoi que ce soit) qu'il manquait une cible. Bon, tout ça de mémoire, et vu que je suis quelque peu fatigué, je peux me tromper !

Elioty, 2015-06-14:
Ping @nbelin 
Pour OpenCV sur Gentoo, j'ai tout plein d'options disponibles et je ne sais pas lesquelles tu as besoin :
- python (activé par défaut) pour utiliser la lib en python je suppose, tu en as besoin ou je peux désactiver cette option ? (sachant qu'il n'y a que le support de python 2.7 et pas de 3.3)
- ffmpeg, ça doit être pour décoder des vidéos encodées en h264 ou autre, nécessaire ? (pour la reconnaissance en direct, je ne pense pas mais si tu veux faire des tests avec des vidéos préalablement enregistrées, je pense que oui)\*
- java, je ne pose même pas la question... :yum:
- jpeg, pour décoder des images en jpeg
- png, pour décoder des images en png
- jpeg2k, pour décoder des images en jpeg2k
- tiff, pour décoder des images en tiff
- threads, je ne sais pas trop à quoi sert cette option sachant qu'il y a déjà OpenMP dans le lot (et d'activée par défaut), je l'active dans le doute ?
- v4l, apparemment absolument nécessaire pour faire du traitement en direct

\* Pour ffmpeg, le support du h264 n'est pas activé par défaut... Ni le xvid d'ailleurs. Mais vu que OpenCV ne demande pas que ces options soit activées sur le paquet ffmpeg, je me dis qu'OpenCV ne doit de toutes manières pas supporter ces encodages. Si la réponse est oui pour activer l'option ffmpeg sur OpenCV, as-tu besoin de ces deux codecs dans ffmpeg ?

Il y a également une option OpenGL, qui installe un GROS paquet de dépendances mais je ne suis pas sûr que ça inclut bien le support de OpenGL ES (vu que OpenGL tout court n'est pas supporté par le matériel). Mais en cherchant quelques tuto sur le net à propos de OpenCV sur une Raspberry Pi, il semblerait que ça soit possible d'activer cette option. Je l'active donc ?

Pour @astralien3000, numpy par contre supporte bien python 2.7 ET 3.3 ;)

nbelin, 2015-06-14:
Hmm je comptais pas faire de Python, donc je pense pas que ce soit utile (sauf si on a trop de mémoire et qu'on sait pas quoi en faire ? ^^)
Pour les vidéos, je sais pas plus que vous quel format est nécessaire :/ Le fait est que pour l'instant je lis les vidéos webcam d'astra sans souci.
Les formats d'image ça peut être pas mal franchement, parce qu'on aura peut-être logs d'image en plus de vidéo (ET que ça peut sûrement nous permettre d'enregistrer des images qu'on visualisera plus tard !) (après jpeg et png devraient suffire non ?)
Pour thread aucune idée, si tu as de la place active dans le doute ^^

Pour les derniers points, pile ou face ? J'aimerais qu'à terme le truc ne dépende plus d'OpenCV (à part pour le flux vidéo (sauf si t'as des solutions ?), lire/sauver des images, tout ça...). C'est juste qu'au début je teste pas mal de trucs et je suis content d'avoir des algos à portée de main, le temps de trouver exactement ce que je veux faire..

Elioty, 2015-06-14:
Pour Python, ok il vire.
Pour les vidéos, je vais embarquer ffmpeg dans le doute alors mais avec juste ces options par défaut (donc pas de h264 et de xvid).
Pour les images, ok pour jpeg et png. Je pense que ça doit permettre d'en sauvegarder aussi mais ça sera à voir (je n'ai aucune idée de l'API d'OpenCV).
Pour threads, je vais l'embarquer dans le doute alors.

Pour lire le flux vidéo, ça doit pouvoir se faire si on attaque directement l'API de v4l. Pour lire/sauver des images, avec les libjpeg et autres, ça doit se faire sans problème.

Enfin pour l'accélération sur GPU, [j'avais trouvé ça](http://petewarden.com/2014/08/07/how-to-optimize-raspberry-pi-code-using-its-gpu/) et je dois bien avouer que je nage un peu dans ce long billet de blog et toutes les ramifications qu'il y a. J'ai vu des posts/billets où les gars finissaient carrément à faire de l'assembleur pour le GPU (comme c'est le cas il me semble pour le lien cité plus haut). Mais [j'ai aussi trouvé celui-ci](http://robotblogging.blogspot.fr/2013/10/gpu-accelerated-camera-processing-on.html), il utilise simplement les shaders OpenGL et n'a pas l'air de toucher à l'assembleur, ce qui me parait tout de même vachement plus réalisable.
Donc je pense que je vais installer OpenGL. Comme ça, tu l'auras à disposition pour plus tard pour pousser l'optimisation à son comble et pour l'activer sur OpenCV, c'est déjà un peu de gain en performance s'il peut s'en servir.

astralien3000, 2016-09-02:
Je considère ce projet comme abandonné, ou en pause très longue durée. Tu pourra rouvrir si tu reprends.

closed 2016-09-02 by astralien3000