xenomorphales/xenomorphales #9: Poids de buggybot

astralien3000, 2015-05-29:
J'ai besoin de connaître le poids de buggybot pour dimensionner les capteur que je vais recevoir du sponsor.
En gros j'aurai le choix entre 10N, 40N, 60N ou 100N (avec une marge de plus de 200%) pour les capteurs.
Il faut juste que le capteur arrive à encaisser le poid normal du robot, et le choc (si on le fait tomber ou on le pose comme un bourrin).

Elioty, 2015-05-29:
Juste par curiosité, tu parles en Newton ? Ça me rappelle des souvenirs du lycée ça !

astralien3000, 2015-05-29:
Ouai c'est des Newton ^^

nbelin, 2015-05-30:
À quoi te serviront ces capteurs? J'imagine que tu parles de capteurs de pression ?

astralien3000, 2015-05-30:
Ils permettent de capter la force de contact appliquée sur un dôme.
Et la force est en 3D ! C'est à dire que on peut détecter la force perpendiculaire à la surface de contact, mais aussi la force tangente (pour détecter d'éventuels frottements).
Et c'est des capteurs UN PEU chères... Merci le sponsoring !

astralien3000, 2015-08-04:
Bon, on connais toujours pas le poids exact du robot, mais les capteurs sont là.
Et ce n'est pas très important, je clos l'issue.

closed 2015-08-04 by astralien3000