xenomorphales/xenomorphales #11: Nom

astralien3000, 2015-05-31:
Allez, choisissons un nom !
Peut être juste pour l'équipe, peut être pour l'éventuelle association.

J'ai deux propositions à faire : 
Astrabot
Xenobot

Vous pouvez faire d'autres propositions. Personellement, comme je vous l'ai dis, j'aime vraiment bien Astrabot. Et aussi, j'aimerai beaucoup garder le logo de l'alien dans l'engrenage. C'est pour ça que je propose aussi Xenobot (pour Xenomorphe).

Elioty, 2015-05-31:
J'ai essayé de me creuser la tête pour essayer de rester dans le thème "alien" vu qu'on a déjà le logo et rien trouvé de passionnant jusqu'à maintenant. Par contre, je préfère Xenobot personnellement !
Moyen de mettre le logo sur la home page du wiki d'ailleurs ?

astralien3000, 2015-05-31:
Ouai, j'me doute, Astrabot fait plus référence à mon pseudo qu'à la série aliens (mais mon pseudo fais référence à la série aliens ! Donc par transitivité...). Ça me pose pas de problème.

Autres idées rapidement :
Robot's hive
Hivebot

"hive" fait référence à la ruche des aliens, et fait référence à la ruches des abeilles aussi !

Par contre, est-ce que le thème "alien" vous dérange aussi ?

Elioty, 2015-05-31:
Le thème ne me dérange pas du tout, ça va. Par contre avec le mot "hive", la connexion est loin d'être directe :p
Xenohivebot ? Hahaha, ça fait presque trop long là :X
Xenov'bot ? (en contractant celui d'au dessus, ça fait aussi penser à innov(ation), c'est Astien qui va être content)

astralien3000, 2015-06-29:
Bon, je pense qu'on est pratiquement d'accord pour Xenobot.
On le décidera ce weekend, quand on sera chez nicolas.

astralien3000, 2015-07-18:
Xenomorphal ?

Elioty, 2015-07-20:
Il est marrant effectivement :laughing: 

De plus, le nom de domaine est dispo avec toutes les extensions que propose OVH !

xenomorphal.xxx ? haha
xenomorphal.alsace ? oui oui, ça existe comme extension
.flowers aussi !!

Mais je crois qu'on va se restreindre au .com, ça fait plus pro quand même. Le .org est dispo aussi d'ailleurs mais un poil plus cher.

astralien3000, 2015-07-24:
C'est décidé pour le nom ! Le nom du dépôt est changé.
Nous sommes les Xenomorphales !
Il va falloir changer quelques liens sur le site et les issues par contre...

closed 2015-07-24 by astralien3000