xenomorphales/xenomorphales #13: Servomteurs 20kg-cm

astralien3000, 2015-05-31:
Les servomoteurs de 10Kg-cm sont inssufisants pour le deuxième segment des pattes de buggybot. Il faut que je m'en achète avec plus de puissance. Le problèmes est qu'ils sont beaucoup plus dur à trouver...

astralien3000, 2015-06-05:
J'ai trouvé un servo à 21.3kg.cm !
Voici le [servo](http://www.feetechrc.com/product/hv-digital-servo/0-15sec60degree-21-3kg-cm-hv-metal-case-digital-servo-fi8622m/).
Il semble avoir un protocole de type Dynamixel-AX12, mais il y a peu de doc, et la personne qui m'a répondu semble avoir un anglais assez... approximatif.

Mais il coûte que 25$ !!
Par rapport aux AX12 qui coûtent 50€ pour 1.5N.m soit 15kg.cm, ça vaut le coup de tenter.

Elioty, 2015-06-05:
Ça ne serait pas du chinois magique encore une fois ?
Parce que leur site est magique en tout cas en ce moment pour moi :
> Error establishing a database connection

astralien3000, 2015-06-05:
MDR !
Leur site est mort entre le moment ou j'ai envoyé le lien et où tu es allé voir.
Deux visiteurs, c'était peut être trop ! C'est l'attaque DDOS la plus cheap de l'histoire ?

En tout cas, je vais essayer d'en acheter 1 ou 2 pour tester. Enfin s'ils me répondent (j'ai attendu 3 semaines pour avoir une réponse...)

Elioty, 2015-06-05:
Leur site est toujours down pour moi... Je ne sais pas où ils sont basés mais s'ils sont Européen, ça serait inadmissible (pour leur CA) de rester plus de 4h down en pleine journée... Puis ce n'est même pas le serveur web qui est hs, c'est le sgbd (ou le lien entre les deux) d'après le message d'erreur. Et je n'ai jamais vu un sgbd tomber en rad perso >_< C'est l'admin réseau qui s'est planté en ajoutant un ligne dans son iptables hahaha "Oh beh mince, pourquoi ça marche plus ?!" :yum:

Bref, pas très constructive tout ça vis à vis du sujet du thread. Et s'ils mettent trois semaines à livrer un mail, qu'est-ce que ça sera pour du matériel physique ?... Ça serait intéressant de tester leurs servos mais faudrait déjà qu'ils veulent nous les vendre d'abord.

astralien3000, 2015-07-07:
Pour info, j'ai pas encore pris les servos du site feetech, mais j'ai pris trois servos du site pololu.

Un à 17kg.cm (25$) : 
https://www.pololu.com/product/1057
C'est pas du 20kg.cm, mais ça sera peut être suffisant (faut tester). Le prix reste correcte.

Un à 21kg.cm (60$) : 
https://www.pololu.com/product/2147
Exactement le bon couple, mais beaucoup plus cher que pour le site chinois.
Il reste cependant dans les pris que j'accepte de payer pour un servomoteur à 21kg.cm

Un à 40kg.cm (60$) : 
https://www.pololu.com/product/2375
Une bombe ! Par contre, la taille est pas standard, et il peut consommer jusqu'à 9 Ampères, ça fait un peu mal.

astralien3000, 2015-07-29:
J'ai testé les 21kg.cm et 17kg.cm précédents.
Le 21 faisait un bruit pas génial, on dirait que les engrenages ont un défaut de fabrication. J'ai peut être pas eut de chance, mais pour 60€ je suis pas sûre d'avoir envire de la retanter.
Si non, il tient la charge.

Le 17 est pas mal, quand il est tout seul... En effet, quand on met tout les servo en même temps, il a encore moins de force qu'un 10kg.cm (je dirai même beaucoup moins). On dirait que tout les autres servo lui pompent l'énergie. Je sais pas si c'est un système de sécurité pour éviter les chutes de tension, mais ça fait que lors du test, le robot était incapable de marcher à cause de ce servo... Mais peut être que si je dimensionne correctement la carte alim, ça irai ? Ca donne pas envie d'en mettre partout en tout cas.

### Si non

Oh la vache !
J'ai faillis en foutre partout en voyant ce servo !

http://hobbyking.com/hobbyking/store/__27749__HobbyKing_8482_Mi_Digital_High_Torque_Servo_HV_MG_60g_31kg_0_15.html

31kg.cm à 7.4V, pas besoin de convertisseur de tension, et pour un prix de 35€ !!!
Les dimensions sont bonnes.
Le seul truc qui est pas précisé, c'est l'ampérage max. Avec une LiPo sans convertisseur de tension, ça doit passer. Et au pire, je fais un mix avec les autres servo.

Elioty, 2015-07-29:
L'ampérage manque dans la description mais s'il peut tenir ses promesses, il défie toute concurrence à ce prix là ! Et je pense qu'ils ne précisent pas l'ampérage parce qu'au voltage où il fonctionne, on peut le brancher directement sur une lipo sans convertisseur de tension. Une personne en commentaire a tout de même mesuré qu'il peut tirer jusqu'à 4.5A en 6V.

Par contre d'après les commentaires, pour certains utilisateurs, le fait que tous les engrenages dedans soit en aluminium leur posent des problèmes de robustesse et de solidité (on comprend d'où ils peuvent sortir un tarif aussi attractif). Ne serait-ce pas un problème pour nous aussi ?

astralien3000, 2015-07-29:
Hmmm, je ne sais pas du tout pour le engrenages en alu. Pour moi l'alu pourrai faire des pièces correctes pour des jambes, mais vu la taille des dents des engrenages, cela peut poser problème.
Tout ce que je peux faire, c'est tester.

astralien3000, 2015-09-03:
Les servomoteurs à [31kg.cm](http://hobbyking.com/hobbyking/store/__27749__HobbyKing_8482_Mi_Digital_High_Torque_Servo_HV_MG_60g_31kg_0_15.html) sont validés.

Ils tiennent  très bien la charge (même à 5V), ont une bonne précision, ont une vitesse équivalente aux autres servos. Et, comme je l'ai dit plus haut, 35€ pour ça, c'est honête.
Le seul défaut trouvé est qu'ils ont que 90deg d'amplitude en non hackés.
Mais lorsqu'ils sont hackés, ils peuvent tourner à 360deg, et peuvent être asservis sur 330 deg environs, ce qui est pas mal !

Elioty, 2015-09-04:
Tu l'as hacké sans problème avec une Teensy 2.0 pour l'asservir alors ? :smiley:

astralien3000, 2015-09-05:
Non, mon arduino.

Par contre, j'ai fais aussi un test avec les servo qui ont une mauvaise précision, et c'était pas à cause du système de contrôle. Leur capteur est pas bon du tout !

astralien3000, 2016-03-13:
Enfin !

closed 2016-03-13 by astralien3000