xenomorphales/xenomorphales #14: Architecture électronique de la base roulante de Céres

astralien3000, 2015-05-31:
Il faut faire le cahier des charges de la base roulante. Déjà, faisons la liste des capteurs et actionneurs qu'il y a déjà (ou qui étaient prévus).

Capteurs : 
2 encodeurs externes (2 TIMERS avec lecture de signal quadratique)
4 fins de course (GPIO)
4 GP2 (ADC)
2 Ultrasons (2*GPIO)

Actionneurs : 
2 moteurs CC (1 TIMER avec 2 output compare/pwm)

Elioty, 2015-05-31:
Concernant les capteurs de distance (parce qu'à entendre toutes les emmerdes avec les GP2, ça ne donne pas trop envie... haha), je suis tombé sur ça ce matin :
http://www.teraranger.com/products/teraranger-one/

Et ça à l'air vachement balaise ! (Le prix aussi, 150€ l'unité...)
Par contre, ils sont français !!! On les contact pour un sponsoring ? :yum: 

Pour le reste, 2 encodeurs, 4 fins de course et 2 moteurs CC, je suis ok.

astralien3000, 2015-05-31:
Alors, comme je t'ai dis, ça me dérange pas de leur demander, ça coûte rien.
Par contre, avoir 2cm de précision... Je sais pas... Après j'ai pas trouvé d'autre appareil à moins de 500€ qui fasse la même précision...

Pour continuer dans le même style : 
http://www.robotshop.com/en/lidar-lite-laser-rangefinder-pulsedlight.html
Télémètre laser (classe 1), 2.5cm de précision, 90€, 60m de portée.

Tu en pense quoi ?

astralien3000, 2015-05-31:
Pour le cas des GP2, si on part sur le fait de vouloir se positionner par rapport aux bords de la table, il faut plus les mettre sur la base roulante, mais plus haut, de façon à ne détecter que les robots adverse.
On est d'accord là dessus ?

Elioty, 2015-05-31:
C'est mignon, ça fait jumelle. Ça ne pose pas problème pour la coupe la classe du laser ? Et ils ne précisent pas (il me semble avoir bien tout lu) si la lumière ambiante peut le déranger ou non donc à étudier (et à mettre au noir avec un cache parallélépipédique ouvert dans la direction du signal).

Pour les GP2 et du fait de les mettre plus haut, c'est pour éviter les éléments de jeu que tu parles je suppose ? Faut les mettre le plus haut possible et qu'ils soient toujours à hauteur des bordures.

astralien3000, 2015-06-01:
Non, ce que je veux dire, c'est que les GP2 ne serviront plus du tout à détecter les bords de la table.
Si tu veux les remplacer par un capteur de distance précis, biensûre.

Elioty, 2015-06-01:
Ah, je vois ce que tu veux dire. Des capteurs précis avec une relativement longue distance de fonctionnement en bas pour les bords de la table et des GP2 plus haut uniquement pour détecter les autres robots.

On peut espérer une précision de combien avec les GP2 sous les 50cm de distance ? Histoire d'avoir une idée pour comparer parce que pour se positionner en longeant un mur, 2cm de précision, je trouve ça loin d'exceptionnel (voir même d'acceptable). On va peut-être se limiter à l'odométrie pour se positionner et au fin de course pour se recaler contre des murs entre chaque action (comme RCVA l'a évoqué également).

astralien3000, 2015-06-01:
Disons qu'il est possible d'avoir une précision assez correcte pour les GP2 si l'obstacle est dans la zone où la caractéristique du GP2 a la plus grand pente. Mais je ne peux pas donner de chiffre, et ça peut être moins bien que les capteurs dont on parle. Et surtout, il faut savoir que la valeur pour une même distance peux changer selon la luminosité. Il y a peut être des moyens pour calibrer le robot et avoir des mesures correctes, genre faire un test avec un mur juste avant un match, mais je ne m'y risquerai pas avant que les autres systèmes fonctionnent.

Par contre, je viens de voir qu'il faut laisser tomber les ultrasons pour la base roulante, ils fonctionnenet pas bien aussi près du sol... Heureusement que j'ai pas eut le temps de les utiliser !

Elioty, 2015-06-01:
Ah oui pour les ultrasons si près du sol qui posent problème. A cause de signaux rebondissants/fantômes, c'est ça ?

astralien3000, 2015-06-01:
Ouaip, je viens de le lire sur une description de capteur ultrason, c'est pas les mêmes que les miens, mais le problème s'applique aussi je pense.

D'ailleur la source : 
https://www.pololu.com/product/1605

Elioty, 2015-06-01:
Oui, j'ai lu ça ailleurs sur le net aussi, sur des pages qui parlaient juste du principe physique donc c'est la même chose pour tous capteurs ultrason. À savoir qu'un capteur ultrason ne verra rien s'il a un objet qui absorbe le son devant lui, genre une éponge (c'était un exemple donnée).

Bon sinon, avec/après les discussions qu'on a eu sur d'autres threads, on reste sur l'ensemble de capteurs que tu as décris dans le premier post minus les deux capteurs ultrason ? Si oui, reste à définir le mcu à utiliser (checker si la Teensy 3.1 en est capable, on pourrait utiliser la mienne) puis à faire les achats. Après avoir retravaillé la base roulant du point de vue méca bien sûr.

Elioty, 2015-07-22:
### Brève étude de faisabilité avec la Teensy 3.1

- [Manuel](https://www.pjrc.com/teensy/K20P64M72SF1RM.pdf)
- [Datasheet](https://www.pjrc.com/teensy/K20P64M72SF1.pdf)
- [Schematic](https://www.pjrc.com/teensy/schematic.html)
- [Pin assignment](https://www.pjrc.com/teensy/pinout.html)

Pour les encodeurs (bien détaillé sachant que c'est le point le plus critique) :
- Page 769 du manuel, il y a le chapitre sur les timers FTM.
- Page 811 du manuel, il y a la description du registre de contrôle pour le mode "compteur de quadrature" des timers FTM.
- Page 888 du manuel, il y a toutes les explications pour paramétrer les timers en compteur de quadrature. Dans notre cas, il faut les registres : - FTMEM à 1 - QUADEN à 1 --Et (page 890)-- - QUADMODE à 0 - PHAPOL à 0 - PHBPOL à 0. Les pages suivantes expliquent relativement bien comment traiter les situations d'overflow du compteur (qui est sur 16 bits).
- Page 62 et suivantes de la datasheet, il y a la "table de multiplexage" des différents pins du mcu.
 - Pour FTM1, on peut brancher la phase A sur PTA12 ou PTB0, et la phase B sur PTA13 ou PTB1.
 - Pour FTM2, on peut brancher la phase A sur PTB18 et la phase B sur PTB19.
 - Note : il faut que je regarde dans le manuel comment configurer le multiplexeur.

Je propose donc d'utiliser les pins PTA12 et PTA13 (soit les pins 3 et 4 sur la board) pour l'encodeur gauche et les pins PTB18 et PTB19 (soit les pins 32 et 25, au dos de la board) pour l'encodeur droit.

Pour les quatre GP2 : je propose d'utiliser les pins PTD1, PTC0, PTB0 et PTB1 (soit respectivement les pins 14/A0 à 17/A3 sur la board). D'après les features du module ADC (page 647 du manuel), il supporte bien les interruptions. Néanmoins, il n'y a que deux modules ADC (spécifié page 97 du manuel) donc faut voir comment ça marche exactement mais il faudra certainement alterner la configuration des modules pour changer de pin. J'ai tout de même bien attention à répartir "la charge" sur les deux modules. En effet, d'après la datasheet, PTC0 et PTD1 sont respectivement sur ADC0_SE14 et ADC0_SE5b (donc sur le premier module ADC) et PTB0 et PTB1 sont respectivement sur ADC1_SE8 et ADC1_SE9 (donc sur le deuxième module ADC).

Pour les deux moteurs : je propose d'utiliser les pins PTC1 et PTC2 (soit respectivement les pins 22 et 23 sur la board) qui sont câblés sur FTM0_CH0 et FTM0_CH1.

Enfin, pour les quatre fins de course : je propose d'utiliser les pins PTD7, PTD4, PTD2 et PTD3 (soit les pins 5 à 8 sur la board). Et de ce que je comprend du manuel page 219, tous les pins supportent les interruptions ce qui peut être pratique.

Donc pour résumer, on utiliserait (sur la board) les pins 3 à 8, 14 à 17, 22 et 23, et 25 et 32. Je ferais une infographie pour représenter simplement les pins utilisés et pour quelle utilité chacun d'eux sera utilisés.
J'ai essayé de ne pas trop utiliser des pins qui pourraient être utilisés à autre chose d'utile si jamais on en a besoin par la suite. Je ferais un récapitulatif complet de ce qu'il nous reste comme PWM et ADC une fois l'infographie réalisée. Pour les moyens de communications, le CAN n'est plus disponible ainsi que le série/l'UART 3. Il nous reste donc toujours deux UART, deux I2C et un SPI.

Elioty, 2015-07-22:
### Recherche de connecteurs

#### Puissance

##### Molex

[Mâle, 1€](http://www.conrad.fr/ce/fr/product/742708/Cnc4Pts-mle-pour-Disque-Dur?ref=list)
[Femelle, 1€](http://www.conrad.fr/ce/fr/product/742236/Cnc4Pts-FemPour-Disque-Dur)

#### Logique

##### Jack 3.5mm

[Mâle, 0.70€](http://www.conrad.fr/ce/fr/product/731480/Jack-35-mm-BKL-Electronic-1107003-mle-droit-Nombre-de-ples-3-stro-argent-1-pcs?ref=list)
[Femelle, 0.80€](http://www.conrad.fr/ce/fr/product/733962/Jack-35-mm-embase-femelle-horizontale-Nombre-de-ples-3-stro-argent-1-pcs?ref=list)

Problèmes :
- sans protection électrique adéquate, le connecteur ne peut pas être branché à chaud sous risque de court-circuit du fait de la physionomie du connecteur en lui même.
- si le connecteur n'est pas branché, il pourrait causer un court-circuit s'il traine là où il ne devrait pas.

##### Micro-USB

[Mâle, 1.90€](http://www.conrad.fr/ce/fr/product/730794/Fiche-Micro-Usb-5Br-Trapzodale?ref=list)
[Femelle, 0.70€](http://www.conrad.fr/ce/fr/product/1307069/Embase-femelle-Micro-USB-USB-20-embase-femelle-horizontale-1-pcs?ref=list)

Problèmes :
- si le connecteur n'est pas branché, il pourrait causer un court-circuit s'il traine là où il ne devrait pas.
- le prix un peu plus élevé que les autres.

##### "Embase" 4 pins (connecteur type ventilateur de PC)

Pour ce connecteur, c'est le mâle qui est fixait sur le PCB.
Mâle (au choix) :
- [droit, 0.60€](http://www.conrad.fr/ce/fr/product/1088232/Embase-mle-droite-pour-circuits-imprims-Pas-254-mm-1-x-4-ples-1-pcs-Wuerth-Elektronik-WR-WTB-61900411121)
- [coudé de "dos", 0.60€](http://www.conrad.fr/ce/fr/product/1088231/Embase-mle-coude-pour-circuits-imprims-Pas-254-mm-1-x-4-ples-1-pcs-Wuerth-Elektronik-WR-WTB-61900411021)
- [coudé de "face", 0.85€](http://www.conrad.fr/ce/fr/product/1088234/Embase-mle-coude-pour-circuits-imprims-Pas-254-mm-1-x-4-ples-1-pcs-Wuerth-Elektronik-WR-WTB-61900419521)

Femelle :
- [boitier, 0.20€](http://www.conrad.fr/ce/fr/product/1088233/Boitier-femelle-Pas-254-mm-1-x-4-ples-1-pcs-Wuerth-Elektronik-WR-WTB-61900411621)
- [contacts, 0.10€](http://www.conrad.fr/ce/fr/product/1088222/Contact-femelle-1-pcs-Wuerth-Elektronik-WR-WTB-61900113722DEC)

astralien3000, 2015-07-23:
# Recherche de connecteurs (suite)

## Puissance

Les connecteurs Molex que tu a choppé sont pas mal pour brancher les moteurs, mais ça serai bien de différentier les connecteurs batteries.

Ceux que j'ai trouvé sont plus cher...

### T

[Mâle/Femelle, 1.49$](https://www.pololu.com/product/925)

C'est ce qu'il y a sur mes batteries LiPo.
Ils sont dur à plugger/déplugger, par contre...

### XT60

[Mâle/Femelle, 1.75$](https://www.pololu.com/product/2174)

C'est les connecteurs de batterie de Rob'otter.
Pas trop d'expérience avec ceux là...

Elioty, 2015-07-23:
Je pense qu'on peut prendre les connecteurs T pour la batterie puissance (sachant que l'alimentation logique viendra depuis la Rasp par USB je pense) histoire de rester consistant avec Buggybot. D'autant plus que s'ils sont dur à plug/déplug, au moins ça ne bougera pas tout seul. Ils sont aussi moins chers que les XT60.
Juste un détail, pour rester consistant et par sécurité, c'est le femelle ou le mâle qui est au bout du câble de la batterie ? Par soucis de sécurité, il vaudrait mieux que ça soit le femelle pour que rien ne se prenne un coups de jus si le connecteur traine (et donc le mâle sur le PCB comme pour les embase).

astralien3000, 2015-07-23:
C'est bien une femelle qui est sur le câble de batterie.
Si non la lipo aurai déjà pris feu depuis longtemps, vu que c'est moi qui la manipule ^^ !

Elioty, 2015-07-23:
Ok très bien donc je pense qu'on peut partir sur le connecteur T pour la batterie (mâle sur le PCB), un seul molex pour alimenter les deux moteurs (femelle sur le PCB) et embases 4 pins (avec code couleur et mâle sur le PCB).

Me reste à voir les pins de sens à sortir de la teensy pour les drivers moteurs. Et d'ailleurs, on prend quoi comme pont-h ? (que je sache combien de pin de sens il me faut)

Et je pense qu'on prendre des embases 2 pins aussi pour que je câble sur le PCB les trois UARTs (tant pis pour le hardware control) et les deux I2C.
(et peut-être des 3 pins aussi selon comment je me débrouille pour le routage et le câblage)

Note : Eirbot devrait vraiment se mettre aux embases plutôt que de rester aux simple sécables d'ailleurs, le détrompeur de sens devrait permettre d'éviter pas mal d'erreurs de branchements.

astralien3000, 2015-07-23:
Alors niveau pont en H, j'ai actuellement 2 L298N achetés sur un site Chinois, mais j'ai peur qu'ils tiennent pas les nouveaux moteurs.

##### L298N

[datasheet](http://www.st.com/web/en/resource/technical/document/datasheet/CD00000240.pdf)
[ceux que j'ai acheté, 5.37€](http://www.dx.com/p/l298n-stepper-motor-driver-controller-board-module-143633#.VbDe09DIXsY)

Avantage : beaucoup de boards trouvable sur internet, donc remplaçable facilement.
Désavantage : Tiendra peut être pas le choc.

##### LMD18200

[datasheet](http://www.ti.com/lit/ds/symlink/lmd18200.pdf)
[seul board tout faite trouvée, 13.56$](http://www.ebay.com/itm/New-PWM-Adjustable-Speed-Motor-Driver-Module-LMD18200-for-Arduino-Robot-Project-/391044212854)

Au moins avant de les cramer, faut y aller, ou les brancher à l'envert.
Par contre, il y a très peu de cartes dispo sur internet.
Je serai presque d'avis de faire la boards (soit en indus, soit à eirbot).

##### Remarque

Il serait peut être intéressant de mettre des [capteurs de courant](https://www.pololu.com/product/2197) sur les sorties moteur. Ca nous donne des info qui peuvent être intéressantes afin d'éviter de cramer des trucs (entre autre), et en plus j'en ai BEAUCOUP !

Elioty, 2015-07-24:
Concernant les capteurs de courant, ça pourrait nous être utile uniquement pour éviter de cramer le pont en H si on fait un pic supérieur à 3A pendant plus de 200ms.
Si c'est juste un ampérage au dessus des 6A supportés en pic par le pont en H, on aura un fusible pour ça.
Et si c'est pour cramer au sens bruler, le pont en H arrête tout tout seul lorsqu'il atteint 170°C mais il a aussi un pin qu'il met à 1 lorsque la température est supérieure à 145°C, que je pourrais router sur un pin du µP si tu penses que c'est utile (par contre de ce que je comprends de la datasheet (je n'en suis pas sûr là, faudrait demander à un élec), l'état haut du pin d'alerte de température est à 12V, il faudrait donc un "réducteur"/"abaisseur" de tension de 12V à 3.3V, un transistor pourrait faire l'affaire non ?).

astralien3000, 2015-07-24:
Ça peut permetre de savoir si le moteur force, donc si il est bloqué (même si n peut le savoir par d'autre moyens). Et puis si on intègre, ça permet de savoir l'énergie consommée par le moteurs au cours du temps, donc d'avoir une idée des niveaux de batterie.

D'ailleurs pour savoir le niveau de batterie, c'est mieux de le mettre au niveau des batteries le capteur.

Si non je suis d'accord avec ce que tu viens d'écrire. Pour l'abaisseur de tension, un diviseur ira très bien ^^ (deux résistances à mettre, ça va).

Elioty, 2015-08-01:
### Système anti-rebond matériel

Alors tout d'abord, voici un article (en deux pages) qui explique très très bien le phénomène et comment éviter les problèmes qu'il induit de façon hard ou soft : http://www.ganssle.com/debouncing.htm

Donc pour la solution choisi : c'est le "RC debouncer" qui va être utilisé.
[image]

Donc d'abord, le choix de la bascule de Schmitt : www.conrad.fr/ce/fr/product/143049
Elle est nécessaire vu que les entrées digitales de la teensy n'en sont pas équipées (et très peu de µP le sont d'ailleurs). [Et sa datasheet](http://www.produktinfo.conrad.com/datenblaetter/125000-149999/143049-da-01-en-IC__SN74LS14D__SOIC14__TID.pdf).

Avant de régler le cœur du problème, on voit (page 5) qu'on peut lui tirer jusqu'à 8mA en sortie et qu'elle sort en moyenne du 3.4V, donc R3 = 3.4 / 0.008 = 425Ω. Par précaution, on arrondi à la résistance supérieure facilement trouvable dans le commerce, en 430Ω : http://www.conrad.fr/ce/fr/product/442698

Maintenant, le cœur du problème, le filtre RC. L'article (qui a fait une très longue étude en première page) suggère de prendre un t = 20ms pour être large, et sûr.

On commence par calculer la résistance R2 qui décharge de la capa : R2 = -t / (C \* ln(Vth / Vinitial))
Pour cela, on pose d'abord C = 1µF. Vth est le pallier (thresold) haut de notre bascule de Schmitt, qui est de 1.6V en moyenne. Enfin, Vinitial est simplement notre +5V qu'il y a en entrée.
On a donc R2 = -0.020 / (1\*10^-6 \* ln(1.6 / 5)) ~= 17553Ω. Encore une fois, on arrondi à la résistance supérieure facilement trouvable dans le commerce, en 18kΩ : http://www.conrad.fr/ce/fr/product/439988

Ensuite pour la charge, on a : R1 + R2 = -t / (C \* ln(1 - Vth / Vfinal))
Vth est ici le pallier bas, qui est de 0.8V en moyenne dans notre cas. Vfinal est ici aussi notre +5V.
Ce qui nous donne donc R1 = -0.020 / (1\*10^-6 \* ln(1 - 0.8 / 5)) - 18\*10^3 ~= 96709Ω. Encore une fois, on arrondi à la résistance supérieure facilement trouvable dans le commerce, en 100kΩ : www.conrad.fr/ce/fr/product/439017

Note : je n'ai pas retranscrit les calculs de puissance ici, mais je les ai bien faits et les résistances listées ci-dessus supporteront bien la puissance qu'elles prendront.

Enfin, pour finir l'étude, on peut faire les calculs inverses pour trouver les temps réels de latence que notre anti-rebond va induire (ils étaient supposés comme étant de 20ms auparavant).

Donc pour la décharge, lorsque l'interrupteur est enfoncé :
t = -R2 \* C * ln(Vth / Vinitial) = -18\*10^3 \* 1\*10^-6 \* ln(1.6 / 5) ~= 20.5ms

Et pour la charge, lorsque l'interrupteur est relâché :
t = -(R1 + R2) \* C \* ln(1 - Vth / Vfinal) = -118\*10^-3 \* 1\*10^-6 \* ln(1 - 0.8 / 5) ~= 20.6ms

Question : tu voulais pouvoir mettre deux fins de course supplémentaires (en plus des quatre évoqués initialement), or la puce qui embarque la bascule de Schmitt fait six canaux, et je comptais me servir de ce système pour la tirette également. Donc, reste-t-on sur quatre fins de course plus la tirette soit 5 canaux nécessaires (donc une seule puce) ou je dois câbler le shield pour supporter deux fins de course supplémentaires, soit 7 canaux (donc deux puces nécessaires) ?

astralien3000, 2015-08-01:
Haha, ça c'est du pavé !
Je t'avouerai que j'ai pas eut le courage de tout lire cette fois-ci !

Mais si non pour ta question des 2 fins de course en plus, non, fais en sorte de rester sur une seule puce. Mais tu peux peut être mettre la 6e pour la complèter. C'est toujours bien d'avoir des ports en plus.

Elioty, 2015-08-01:
Tu me fais confiance et on utilise donc ce système comme anti-rebond ? (que j'update le premier post pour acter le choix)
Et ok, je câblerais le sixième canal, reste à voir où il arrivera sur la Teensy 3.1 (sachant qu'il me reste encore à voir où je sors les pins de sens pour les drivers moteurs).

astralien3000, 2015-08-02:
Nan bein je viens de lire tes explication, et une partie de l'article que tu donnes, et ça me va.

Elioty, 2015-08-03:
Bon, pour le système anti-rebond, ça a été bien validé (on a fait le tour des datasheets et tout) par les élecs de ma boite donc c'est tout bon. Juste une remarque, la résistance en sortie de la bascule de Schmitt n'est pas nécessaire.

Pour l'étage d'alimentation en logique, ils préconisent une simple capa de 100nF en parallèle et au plus proche des différents composants logiques (les GP2 et la puce bascule de Schmitt). On a même vérifié sur la schematic de la teensy, elle a déjà la sienne. Reste donc juste à évaluer le fusible logique.

Pour les connecteurs, on a regardé les embases et les types T et ça a l'air d'aller. Il avait juste un doute sur la résistance aux vibrations et aux chocs des embases (ie, est-ce que ça se détache) mais je lui ai dit qu'on a fait ça longtemps avec de simples sécables sans problème et qu'on passait à ça surtout pour le détrompeur, donc c'est ok.

Dernier point, il peut nous donner des contacts indus pour faire imprimer la/les cartes si besoin ;)

Elioty, 2015-08-03:
Je viens de penser à une dernière chose d'ailleurs, on embarque quoi comme batteries au final ? Je pensais à une LiPo (au moins 12V de préférence, pour les LMD18200) et une batterie externe USB classique (Li-ion donc, en 5V comme il faut) pour la logique. Sachant que pour la logique, je ne parle que de la base roulante, je pense qu'on devrait avoir une autre batterie USB dédiée pour la Rasp.

Elioty, 2015-08-18:
### Alimentation

#### Logique

| Device | Ampérage (moyen/max) | Condensateur |
| ------ | -------- | ------------ |
| [Batterie](http://www.amazon.fr/dp/B00RCGYTVG/ref=asc_df_B00RCGYTVG27910755?smid=A2UZUMHNJKCOJU&tag=clubic-mp-electronics-21&linkCode=df0&creative=22822&creativeASIN=B00RCGYTVG) | 2.1A | aucun |
| [GP2](https://www.pololu.com/file/download/gp2y0a21yk0f.pdf?file_id=0J85) | 30mA/40mA | 10µF (recommandé) |
| [Teensy 3.1](https://www.pjrc.com/store/teensy31.html) | _500mA_\* | déjà équipée |
| [Bascule de Schmitt](http://www.produktinfo.conrad.com/datenblaetter/125000-149999/143049-da-01-en-IC__SN74LS14D__SOIC14__TID.pdf) | 12mA/21mA | 100nF |
| [RC anti-rebonb](https://github.com/astralien3000/xenomorphales/issues/14#issuecomment-126942873) (les six) | < 1mA | aucun |
| Pullup flag de chauffe (les deux) | 1mA | aucun |
| Encodeurs | ??? | _a priori 100nF_\* |
| Total | ??? | |

_\* a priori 500mA maximum mais je n'ai pas pu trouvé de datasheet pour la board en elle-même._
_\* a priori 100nF par défaut, valeur recommandée dans la datasheet s'il y a une._

Condensateurs :
- [100nF](http://www.conrad.fr/ce/fr/product/454613)
- [10µF](http://www.conrad.fr/ce/fr/product/1278809)

Fusible : ??? (vaguement/certainement 1A)

#### Puissance

Condensateur : recommandé par la datasheet, au moins 100µF par ampère, donc il nous faudrait au moins environ 300µF avec nos moteurs mais je vais vérifier cela avec les élecs de ma boite.
Fusible : je pensais à [ceux là](http://www.conrad.fr/ce/fr/product/532889), un par pont en H.

Elioty, 2015-08-19:
Bon l'étude pour "l'alimentation" au dessus est complète. Subsistent deux questions pour les fusibles :
- Pour la logique, le fusible est-il vraiment nécessaire ? Sachant que, théoriquement, la batterie logique ne peut sortir au maximum que 2.1A, or le fusible présenté peut mette jusqu'à 5 secondes (au maximum) à rompre à 200% de son intensité nominale (donc à 2A). Autant dire que ça a largement le temps de prendre feu. Autre défaut des fusibles, ils font un drop de tension et celui ci fait 200mV. Tous nos composants acceptent de fonctionner à 4.8V mais c'est juste que ça réduira un peu l'autonomie de la batterie pour le coup.
- Pour la puissance, la question est 4A ou 6A ? Sachant que le drop en tension est ici négligeable (et le PID sera là pour compenser) et que les LiPo ont ici largement assez de patate pour faire fondre le fusible quasiment instantanément en cas de cours circuit, que ce soit pour le 4A ou le 6A je pense. Je pense aussi limiter la PWN à 90% dans un premier temps (je la débloquerais peut-être selon la réactivité de l'heuristique de "je patine ou je bloque, arrêtez tout !") vu que la datasheet de nos moteurs ne précise pas le courant qu'ils tirent en pointe à 15V (2.85A à 12V pour référence) et que les LMD18201 ne sont pas sensés sortir plus de 3A (ou seulement en pointe jusqu'à 6A pendant au max 200ms, d'où les tests de réactivité à faire pour voir si on peut quand même sortir des PWM de 100%).

astralien3000, 2015-08-19:
Pour la logique... Je pense que ça devrait le faire sans fusible.
Pour la puissance, il y a pas à réfléchir, si ? Si les LMD18201 peuvent sortir 3A, vu qu'on en a 2, ça fait un fusible à 6A ?

Elioty, 2015-08-20:
Pour la logique, ok, pas de problème.
Pour la puissance, je comptais leur mettre chacun un fusible mais on peut faire avec un seul aussi. Du coups en 6A, ça devrait le faire ;)
Je valide tout ça.

Elioty, 2015-09-06:
La schématique est a priori complète, je viens de l'envoyer pour validation à Eirbot, Ausbee et Robotter. J'attends les retours avant de considérer cette étape comme définitivement terminée.

Elioty, 2015-09-25:
Liste de composants :
- [Résistance CMS1206 22Ω 0.25W](http://www.conrad.fr/ce/fr/product/434909)
- [Résistance CMS1206 82Ω 0.25W](http://www.conrad.fr/ce/fr/product/437839)
- [Résistance CMS1206 10kΩ 0.25W](http://www.conrad.fr/ce/fr/product/433084)
- [Résistance CMS1206 18kΩ 0.25W](http://www.conrad.fr/ce/fr/product/434291)
- [Résistance CMS1206 20kΩ 0.25W](http://www.conrad.fr/ce/fr/product/434575)
- [Résistance CMS1206 100kΩ 0.25W](http://www.conrad.fr/ce/fr/product/433096)
- [Capa céramique CMS0603 1nF 50V](http://www.conrad.fr/ce/fr/product/454419)
- [Capa céramique CMS0603 10nF 50V](http://www.conrad.fr/ce/fr/product/454532)
- [Capa céramique CMS0603 100nF 25V](http://www.conrad.fr/ce/fr/product/454613)
- [Capa céramique CMS0603 1µF 25V](http://www.conrad.fr/ce/fr/product/450696)
- [Capa céramique CMS1210 10µF 25V](http://www.conrad.fr/ce/fr/product/1280051)
- [Capa électrolytique radiale 330µF 35V](http://www.conrad.fr/ce/fr/product/445240)

Elioty, 2015-10-11:
J'ai corrigé tous les angles droits sur le routage (il ne me semble pas en avoir zappé, j'ai vérifié deux fois mais je ne garantis plus rien après deux heures du matin). Je vais juste rajouter les références sur tous les composants demain et puis ça part par mail pour vérification par tout le monde ;)

Par contre, j'ai des via avec un trou de 0.40mm et une pastille de 0.60mm, ça ne passera certainement pas à l'école mais vu que tu voulais la faire indus dès le départ de toute manière, on n'aura pas trop le choix je crois.

Je retrouverai la liste des sites/entreprises pouvant nous usiner ça demain aussi et je les mettrai ici.

Elioty, 2015-10-13:
Possibles industriels pour imprimer la carte :
- [Cimulec](http://www.cimulecgroup.com/fr/)
- [PCB-POOL](https://www.pcb-pool.com/ppfr/index.html)
- [ProtoElectronique](http://www.protoelectronique.com/)
- [Emsproto](http://www.emsproto.com/fr/)

astralien3000, 2015-10-14:
Question, est-ce que c'est important de mettre des pistes aussi petites ?
Plus une piste est petite, plus elle a de chance d'être mal imprimée (c'est de l'indu, mais quand même).
De plus, on voit pas ton plan de masse. Nan parce que si ça fait des ilots, il va falloir ajouter (encore) des vias.

Elioty, 2015-10-14:
Pour la largeur des pistes (de 0.250mm), c'est la valeur par défaut de Kicad donc je suppose que ça doit bien sortir sans problème en indus. Puis si c'est mal imprimé, on peut toujours refuser le PCB qu'ils n'auraient pas dû envoyer de toute manière. Sinon question capacité électrique, les pistes logiques supportent un peu plus de 800mA donc largement suffisant de ce côté là.

Je n'ai pas prévu de plan de masse, c'est surtout pour ça qu'on ne le voit pas. Puis on m'a dit que c'est surtout pour limiter des problèmes de court-circuit si des choses tombent sur le PCB mais si on fait vernir la carte, ça limite tout de suite l'utilité du dit plan de masse.

astralien3000, 2015-10-15:
Ok pour les pistes. Après pour le plan de masse, je pensais que ça stabilisait un peu les signaux...

Elioty, 2015-10-15:
Je reposerai la question cette après-midi et on pourra demander à des anciens ce soir aussi dans le doute ;)

astralien3000, 2015-10-27:
On m'a parlé d'un site chinois pas cher : 
 - [pcbway](http://www.pcbway.com/)

Elioty, 2015-10-28:
J'ai fini les derniers détails a priori sur le routage. Y compris le baptême de la board. Hermès pour les intimes, messagers des dieux, le gardien des routes et des carrefours et des voyageurs, entre autres. Ce qui je trouve va bien pour une carte gérant la base roulante. Après la mythologie romaine avec Cérès, grecque avec Hermès, je chercherai un nom égyptien pour la Raspberry (genre dieu de la guerre, de la stratégie ou quelque chose comme ça).

Sinon à part ça, je viens de faire deux devis en ligne sur PCBWay et je pense qu'on va commander là bas.
Les deux devis sont identiques à l’exception de deux paramètres :
- la largeur minimale/l'espacement minimale des pistes (0.100mm vs 0.150mm)
- le diamètre minimal des trous (0.2mm vs 0.3mm)

Mais sachant qu'on a des pistes logiques de 0.250mm de large et que j'ai travaillé avec une grille de 1.27mm, il y a normalement de la marge, même pour deux pistes obliques en parallèle au plus proche selon ma grille.
Pour les vias, le perçage est à 0.4mm donc encore une fois, c'est a priori bon.

Question quantité, c'est 5 minimum, tant mieux à vrai dire.
Et pour le tarif (des PCB seuls), c'est 83$ pour l'impression la plus précise, 13$ pour la moins précise, qui devrait être bien suffisante.

Je leur ai uploadé mes gerber files et c'est en attente de vérification là donc je vais voir s'il valide l'impression la moins précise, si oui, on ne risque pas grand chose à ce tarif et je me lance dès que possible.

Concernant la commande de composants, une fois le PCB commandé, je la lance de mon côté si elle n'est pas déjà partie avec Eirbot.

Elioty, 2015-10-29:
Ils ont validé les deux commandes. Je vais donc tenter avec l'impression la moins chère, je passe commande ce soir. La livraison coûte plus cher que l'impression de 5 boards... Enfin bref, il y en a pour 38$ au total, donc frais de douane ou pas, je vais tenter le coups. Et il accepte bien PayPal il semblerait donc tout est ok.

astralien3000, 2015-10-29:
Sur PCB Way ?

Elioty, 2015-10-29:
Oui sur PCB Way, j'en ai eu pour 35.79€ tout compris pour 5 PCB donc. On verra ce que ça donne.
Je vais voir pour passer la commande de composants dans la soirée aussi.

Elioty, 2015-10-29:
Les composants ont été commandés (tout en double) sur Mouser ce soir également.
J'attends réception et validation des articles pour valider les deux sub-tasks.

Il ne restera donc ensuite (si tout ce se passe bien) plus qu'à souder tout ce petit monde sur la board et faire tous les câbles. À ce propos, quelqu'un aurait un câble USB avec une prise mâle type A classique d'un côté dont il ne se sert plus ? C'est pour faire le câble d'alim logique. Je dois certainement en avoir en rab aussi cela dit.

Elioty, 2015-11-02:
Composants bien reçus, en bon nombre voire même quelques excédents. J'ai validé la sous-tâche.
Je vérifie un peu plus les PCBs après manger.

Elioty, 2015-11-02:
Bon les PCBs sont bons, j'ai testé les quelques contacts où j'avais un doute et tout est ok.

Elioty, 2016-03-02:
Juste un message pour "réveiller" un peu l'issue. Maintenant que la base méca est nickel toute propre et surtout avec le nécessaire pour fixer fins de courses et GP2, il ne reste plus qu'à faire leur câblage (certainement ce vendredi ?) pour cette issue. Et puis on pourra ensuite tout monter, et clore l'issue :)

Elioty, 2016-03-06:
Le câblage des GP2 et fins de course a été fait ce vendredi comme évoqué. De plus, les 4 entrées GP2 et les 4 entrées fin de course ont été testé, avec succès :) On avait zappé l'entrée tirette mais je viens de la tester aussi sans problème.

Je clos l'issue !!

closed 2016-03-06 by Elioty