xenomorphales/xenomorphales #15: Système de path finding sur Cérès

Elioty, 2015-06-01:
Concernant le path finding sur Cérès, je suggère d'utiliser un A* avec une représentation vectorielle de la table (comme on l'a fait en 2014). La résolution de la matrice de l'A* restant à déterminée selon les performances de l'algo sur la Raspberry Pi 2 embarquée. Je pense viser une résolution de l'algo sous les 500ms en moyenne.

Des suggestions, des avis ou d'autres idées sur le sujet ?

nbelin, 2015-06-01:
Je pense que le principe général de 2014 était pas mal : 
 - connaissance vectorielle de l'environnement
 - ajout dynamique d'adversaires
 - test rapide pour savoir si la trajectoire directe rencontre un obstacle
 - si oui exécuter A*
 - parcourir le chemin !

Concernant l'A* en particulier, je suis plutôt content de la solution de 2014 car on avait du environ 200ms de calcul sur AVR avec une grosse optimisation de la complexité en espace. Du coup sur Rasp, en changeant les accès pour passer d'une optimisation espace à une optimisation temps, la barre des 10ms doit être facilement atteignable.
Ce qui va jouer aussi c'est la discrétisation de la table de jeu : en 2014 on avait des pas de 10cm. Si on veut des pas de 5cm, ça va forcément impacter de façon non négligeable les temps de calcul.

astralien3000, 2015-06-01:
Dans les cas extrèmes, je connais une équipe de Nancéens qui utilise un pas de 1mm ^^.
Et ils se demandaient pourquoi l'algo leur explosait leur RAM. (Il m'ont regardé avec des gros yeux quand je leur ait dit qu'on avait fait tourné l'algo sur un AVR) XD

Juste un truc, j'ai un petit problème avec la trajectoire directe (si tu vois ce que je veux dire, Nico). Faudra pas se foirer sur ce point.

D'ailleur j'ai une petite question pour l'implémentation, si on en fait un module DDS. Je vois bien la tête des messages d'entrée (deux champs : source, destination), mais pour le message de sortie...
DDS gère pas trop les messages à taille variable, je suppose. Il faudra trouver une interface propre pour envoyer la trajectoire.

Elioty, 2015-06-01:
Un pas de 1mm, je pense que c'est un peu overkill quand même ! Ça sera à tester selon les performances mais je pense que si on peut descendre à 1cm, ça sera largement suffisant.

Je pense que l'idée de tester la trajectoire directe n'est pas mal quand même, j'y ferai bien attention au moment de l'implémentation.

Concernant DDS, aucun problème pour les messages à taille variable ! En DDS, tu peux mettre des champs dans tes struct de type séquence (d'entier, de float, etc... et aussi d'un type complexe), et ça marche comme un vecteur de la lib standard C++ donc taille variable. Sachant que l'on peut aussi préciser la longueur maximale de la séquence mais ça peut marcher sans limite aussi. Donc pour le message de sortie, il suffit de faire une séquence de (source,destination) où `elem[1].source == source && elem[n].destination == destination && (for i in 1,n-1 : elem[i].destination == elem[i+1].source)` (gros pseudo code type post-condition que vous devriez comprendre sans problème).

astralien3000, 2015-06-01:
Impec ! DDS, c'est super bien en fait XD !

Elioty, 2015-06-01:
Beh pour que ça soit utilisé en aérospatiale et dans la défense, heureusement que c'est super/ça marche bien ! D'ailleurs, il faut que je prototype tous les types de données qu'on va échanger avec DDS mais faut avancer les spécs pour certains systèmes avant.

astralien3000, 2016-10-17:
Je reprends le flambeau de l'A* !
J'ai déjà un [algo fonctionnel](https://github.com/astralien3000/astar), mais il n'est pas encore capable de tourner sur microcontrôleur (utilisation de la STL).

darwikey, 2017-03-05:
C'est parfait ton algo, tu penses que ça demanderais beaucoup de travail d'enlever la STL ?
On pourrait utiliser les conteneurs d'aversive++?

astralien3000, 2017-03-05:
Tu l'a vraiment regardé ?

> C'est parfait ton algo

Merci ! ^^

Il ne devrai pas y avoir beaucoup de travail. Une aprem de weekend max. Il y a quelques fonctions difficiles à ne pas foirer (genre la création de voisins, et le calcul de coût, dont l'API est pas géniale), mais le reste est assez simple.

darwikey, 2017-03-05:
Ok cool, quand Ceres aura une asserv vraiment fonctionnelle, on pourra tester

astralien3000, 2017-04-06:
A quel point on peut dire que c'est bon pour cette tâche ?

darwikey, 2017-04-07:
C'est fonctionnelle, je close

closed 2017-04-07 by darwikey