xenomorphales/xenomorphales #16: Solutions de coupleur d'axe

astralien3000, 2015-06-02:
J'ai trouvé trois solution pour coupler les axes.

Le bien connu : 
http://www.selectronic.fr/coupleur-d-axe-6x6mm-919d-1.html
http://www.gotronic.fr/art-coupleur-d-axes-4-6mm-cl46-11766.htm

Aussi nommé le coupleur "qui a toute les paires de taille possibles mais pas celle que tu cherche".
Mais on peu agrandir un des côté pour faire passer notre axe, éventuellement.

Deuxième choix : 
http://www.gotronic.fr/art-coupleur-d-axes-5-8-mm-cl58-19704.htm

D'après ce que j'ai compris on a tout les choix entre 5mm et 8mm. Mais ça se fait par serrage, et il y a pas de vis de pression... J'ai pas trop confiance.

Et dernier choix : 
http://www.gotronic.fr/art-machoire-d-accouplement-6-mm-cl8-25m6-20811.htm

Les machoires d'accouplement (j'adore le nom) ! Au moins on a la taille exacte, il y a une vis de pression, et on peut composer n'importe quelle paire de taille.
Par contre, je suis pas sur de la façon dont on assemble le tout, et de l'élasticité finale.

Elioty, 2015-06-02:
Ooooh. Ça à l'air cool avec la mâchoire d'accouplement. Le principe a l'air simple pourtant Loïc. Les deux mâchoires sont identiques (au diamètre de l'axe près) et tu utilises un joint qui ressemble à une étoile à quatre branches. Ta première mâchoire va positionner ses deux dents entre deux inter-branches opposés, l'autre mâchoire se positionnera dans les deux autre inter-branches. (Il y a une photo du système monté pourtant)
La solution revient pas mal plus chère par contre. Le prix de la liberté certain diront, mais aussi de la propreté. À condition également que ça tienne bien et que ça ne se démonte pas aux premières vibrations. J'essaierai de voir s'il y a des retours sur le net.

astralien3000, 2015-06-02:
Nan, ça j'avais compris, je suis pas débile, mais c'est juste que j'ai l'impression que les deux mâchoires ne sont pas vraiment solidaires une fois montées. Je veux dire par là que le système opposera pas de résistance si on tire sur les deux parties. C'est pas grave au final, mais je pensais qu'il y aurait quelque chose pour au moins les retenir entre elles.

Si non j'ai pas fait gaffe au prix, mais en effet, ça fait très mal !

Elioty, 2015-06-02:
Perso, pour avoir un truc bien propre, je pense qu'on peut bien mettre ~24€ par coupleur. Sachant que ça a l'air de bien tenir quand même.
Et d'après les caractéristiques de celui que tu as présentés, il peut encaisser largement le mouvement sorti par nos moteur :
> Couple maxi: 60,7 kg.cm
> Vitesse maxi: 16000 tours/min

Sinon on peut aussi mail-(all, eirbot, anciens) pour voir s'il y a des gens qu'on des avis sur le coupable par mâchoires. J'ai cherché succinctement sur le net, ça à l'air utilisé dans l'industrie ce principe donc ça doit bien marcher.
Et pour la question de tirer sur les deux parties pour les désolidariser, ça doit se faire je pense (faut bien que ça se démonte quand même s'il faut changer une pièce) mais une fois monté, rien ne devrait tirer dessus de toute manière, il ne devrait que tourner. Puis si on monte bien des suspensions sur le bloc moteur, le coupleur sera même "sous pression" donc je ferrais vraiment des gros yeux étonnés si le truc tombe en pièces !

astralien3000, 2015-06-03:
Je suis aussi pour les mâchoires.
Mais bon, on a encore un peu de temps avant de devoir faire un choix.

astralien3000, 2015-07-06:
Une meilleure solution serai donc un système de courroie poulies.
J'ai une proposition qui ne nécessite pas d'utiliser un frigo ou une plaque électrique : 
http://www.gotronic.fr/art-poulie-crantee-44-dents-tp44t6-19611.htm

astralien3000, 2015-08-03:
La proposition précédente est trop épaisse. On peut pas la faire tenir.
J'ai une solution qui est de la bonne épaisseur et qui semble avoir une vis de serrage.
Il y a pas d'image pour le vérifier, mais c'est écrit en plusieurs endroits.
http://www.conrad.fr/ce/fr/product/209515/Roue-pour-courroie-dentee-MODELCRAFT-6mm-30-dents

Elioty, 2015-08-03:
Hum, tu es sûr de toi pour la vis de serrage ? Il a y a bien quelques références à une vis (sans tête) mais ça ne précise pas de serrage et sur les schéma de la datasheet, elle n'est pas représentée non plus. La datasheet précise juste M3 en 5mm (merci Google translate pour traduire l'allemand).
Après effectivement, je suppose aussi que c'est une vis de serrage mais bon, on ne sait jamais. Moyen de les contacter à Conrad pour leur demander des détails ? (sachant qu'il semblerait qu'on va leur commander pas mal d'articles)

astralien3000, 2015-08-03:
Je leur ai posé la question sur le site, on verra s'ils nous répondent. Mais je suis confiant.

astralien3000, 2015-08-03:
Pour les axes de roues, j'ai pas trouvé grand chose de bien.
Sur conrad, pas d'axe avec méplat, semblerait il. Et en plus c'est un peu cher...
http://www.conrad.fr/ce/fr/product/237086/Arbre-en-acier-Modelcraft-10576?ref=list

Je crois que je vais tenter ma chance avec des vis de 6mm avec un méplat à la main pour commencer.

astralien3000, 2015-08-21:
Hey !

https://www.dropbox.com/s/zagteimm77257fi/2015-08-21%2022.00.18.jpg?dl=0

Elioty, 2015-08-22:
Ça a l'air d'être du solide ! :yum:

astralien3000, 2015-08-23:
Bon, j'ai 2 mauvaises nouvelles : 
 - La courroi que j'ai acheté semble vraiment courte. Je crois que je me suis planté dans les calculs... Donc je valide pas.
 - L'impression de la pièce a foiré. Elle est trop petite. Je doit encore déterminer si c'est ma faute (conception trop "serrée") ou si c'est l'imprimante qui est mal calibrée. 4h perdues ce samedi...

astralien3000, 2015-08-27:
J'ai avancé du côté de l'impression !
Bon déjà je me suis carrément acheté une imprimante 3D (ouai, je suis comme ça), à 350€ et elle marche pas trop mal !
En suite, c'était bien la pièce imprimée qui était redimentionnée (soit par l'imprimante, soit par le logiciel), et mon schéma autodesk inventor était bon.

https://www.dropbox.com/s/3yztsqiqh0sdbne/2015-08-27%2002.20.54.jpg?dl=0

La pièce a aussi foiré un peu avant la fin, mais on peu voir que c'est pas mal.
Je pense qu'il faut juste la solidifier au niveau du roulement.
Et j'ai trouvé une façon de la couper en 2 qui va faciliter le montage, diminuer les risques de foirage d'impression, et sans ajouter de visses !

astralien3000, 2015-08-27:
Pièce imprimée et validée !

Elioty, 2015-08-27:
Wow, ça rend super sur la photo :)

astralien3000, 2015-09-13:
Le block pour le galet tendeur est imprimé et validé.
J'ai juste un petit doute parceque ça appuie sur la courroie même dans la configuration "tension minimale", mais que ce soit avec une roue achetée ou les objets que j'utilise actuellement, je doute de faire mieux.

En tout cas, reste plus que la commande RS pour valider l'ensemble.

Elioty, 2015-09-14:
Ouais on verra avec l'ensemble mais on a des courroies de 230mm alors qu'il y a 220mm "d'espace" normalement, c'est bien ça ? Si oui, faudra donc dans tous les cas que ça tende un minimum, je ne vois pas donc pas de problème particulière (sauf si la dite "tension minimale" n'est pas si "minime" que ça).

astralien3000, 2015-09-29:
Wah ! Enfin fini !

closed 2015-09-29 by astralien3000

astralien3000, 2017-04-01:
Bon rappel NE PAS ACHETER DES COUROIES CONRAD 200mm !