xenomorphales/xenomorphales #17: Modélisation mécanique de la base roulante de Céres

astralien3000, 2015-06-03:
Comme on avance pas mal sur le choix de moteurs, roues, encodeurs, etc, il faut mettre en ligne les anciens schéma mécanique de Cérès et les mettre à jour.
Il sont sur mon Windows, que je démarre une fois tout les 3 mois. Il va falloir que je le fasse, même si c'est dur...

astralien3000, 2015-06-04:
Done !
Allez voir de la dépôt ceres_meca.
Je sais pas si on va séparer les dépôt meca, elec et soft, mais pour l'instant c'est le cas.

astralien3000, 2015-06-20:
J'ai une idée d'architecture pour la ase roulante qui peut résoudre certains de nos problèmes.
https://www.dropbox.com/s/ux2xsh8kkj5bz51/2015-06-20%2018.46.56.jpg?dl=0

A gauche, on a l'architecure telle qu'on la pensait. Le moteurs sont au milieu, avec un revoi d'angle. Les roues motrices sont à l'intérieur, et les roues codeuses sont à l'extérieur. Le problème est que les roues codeuses doivent être très petites pour tenir dans cette architecture.

A droite, on inverse la position des roues codeuses et des roues motrices. Les encodeurs doivent alors respecter les mêmes contraintes que les moteurs dans la première version de Céres, ce qui est beaucoup plus facile. Par ailleurs, les moteurs sont déportés au dessus, et liées aux roues motrices par une poulie. Par rapport à l'architecture de gauche, le centre de gravité est plus bas, même s'il rest relativement haut.

nbelin, 2015-06-21:
Effectivement une poulie est sûrement beaucoup plus simple à faire qu'un renvoi d'angle !
Par contre je ne vois pas vraiment pourquoi tu mets les encodeurs à l'intérieur ? C'est pas mieux de les mettre à l'extérieur ?

astralien3000, 2015-06-21:
C'est parceque j'ai pas la place, l'encodeur que je veux mettre est trop gros.
Il fait la taille des moteurs que j'ai mis cette année, ce qui est petit pour un moteur mais énorme pour un encodeur. Et même les encodeurs plus petits sont trop gros pour l'autre configuration.

Si non, pour le renvoi d'angle, on aait une solution toute faite, mais le problème est qu'elle prend inutilement de la place.

Elioty, 2015-06-22:
Ça à l'air pas mal comme système/disposition mais question suspension, on fait quoi ?
Pour l'encodeur, ce n'est pas dur, les suspensions doivent le pousser vers le bas. Mais pour le bloc moteur pour maintenir la courroie tendue, on pousse les roues vers le bas ou les moteurs vers le haut ? Je ne sais pas trop vraiment là :X

astralien3000, 2015-06-22:
Déjà le problème de pousser les roues vers le base est qu'après il y aura le sol qui poussera dans l'autre sens, donc on peut pas les utiliser pour tendre la courroie. Et pour les moteurs, ça m'enchante pas trop de les mettre sur une plaque mouvante par rapport au robot. Il y a une troisième solution : une galet tendeur. C'est une troisième roue sur ressort qui a pour unique but de maintenir la courroie tendue.

Donc, ce qu'il me paraît pas mal, c'est de mettre les roues sur suspensions (comme on voulait le faire à la base), mettre les moteurs fixes, et mettre un galet tendeur.

Par contre, j'ai peur que le système ne commence à devenir complexe...

Elioty, 2015-06-22:
Ah ton histoire de galet tendeur, c'est un peu comme les dérailleurs à l'arrière des vélos. Et j'avais effectivement pensé aux mêmes problèmes que ceux que tu évoques, mais cette solution ne m'était pas venu à l'esprit.

Ça complexifie le tout, c'est vrai, mais ça nous permet d'avoir un encodeur performant sans être trop encombrant, n'est-ce pas ?

astralien3000, 2015-06-22:
Ouai je pense.
En gros cette disposition est la meilleur que j'ai trouvé pour mettre les encodeurs de 1024 ppr. Si non il faut prendre soit le truc plus petit à 200 ppr, soit des encodeurs beaucoup plus chers, et ils seront peut être pas assez petits.

J'ai déjà commencé à regarder pour faire le "bloc roue". C'est chaud, mais ça me semble faisable.
En gros, il faut la roue pololu, le monting hub, une poulie, et deux roulements à bille. Ca fait pas mal épais (pour une roue), mais c'est toujours moins épais qu'un encodeur, et normalement, il y aura la place pour la deuxième roue avec l'encodeur.

Elioty, 2015-06-22:
C'est vrai qu'un encodeur 1024ppr pour pas trop cher, ça serait cool :smile: 

Pour l'équipement, je te fais confiance, je ne m'y connais pas là.

astralien3000, 2015-07-20:
Pour la base roulante, j'ai commencé un design qui rentre dans le périmètre (j'ai quelques millimètres de marge mais c'est juste). J'ai les roues, les encodeurs, mais pas les moteurs. Il faudra aussi que je change la poulie de la roue qui est pas bonne, mais ça devrai pas changer le reste.

Elioty, 2015-07-20:
Super boulot ! Avec tout ce que tu fais là, j'ai plutôt intérêt à faire une asserv qui dépote sinon, tu te seras embêté pour rien !

astralien3000, 2015-08-04:
Alors, j'ai trouvé une solution pour les batteries.
A la place d'en mettre 1 de 4S, 2 de 2S, on peut en mettre 4 de 1S !

http://www.conrad.fr/ce/fr/product/517997/Accu-LiPo-CONRADENERGY-37-V-2100-mAh-20-C-XH-974356-2100-20C?ref=searchDetail

2100mAh, 17€
Ca fait le jeu de 4 batteries à environs 80€, contre 70€ pour 2 x 2S.

Un truc cool, vu que c'est des 1S, il existe des chargeurs de LiPo !
https://learn.sparkfun.com/tutorials/lipo-usb-charger-hookup-guide
(On en trouve pas pour 2S et plus)

En dehors de ça, les batteries sont vraiment petites et tiennent parfaitements dans la base roulante.

Tu en penses quoi ?

Elioty, 2015-08-04:
Ah beh c'est vraiment pas con, surtout si ça te facilite la meca ! Pour le tarif, ça va encore, pas de soucis. Faudra acheter un second jeu de batterie avant de partir à la coupe mais on a le temps encore pour ça. Et question caractéristique (principalement l'ampérage moyen et en pic), ça suit encore donc c'est bon :smile: 

Et le fait qu'il existe des chargeurs aussi, c'est super ! On risque moins de tout se faire péter à la figure. Enfin ça existe dans le commerce ou tu n'as trouvé que ce DIY ?

astralien3000, 2015-08-04:
Euh, en fait le seul avantage de ce chargeur c'est qu'il est USB, si non je possède un chargeur pro qui est 10x plus sécurisé, et qui peut charger plus vite aussi.

Mais ça permettrait de pas avoir à sortir les batteires du robot. Après, a-t-on besoin de cette fonctionnalité ?

Elioty, 2015-08-04:
Tout dépend si tes batteries sont facilement "extractible" de ta méca.

astralien3000, 2015-08-04:
Normalement oui, elles seront faciles à enlever et à remettre. Normalement. ^^

Elioty, 2015-08-04:
Dans ce cas là, on peut se contenter de ton chargeur pro alors. Puis ça va être assez chargé comme ça déjà, on ne va pas rajouter inutilement des choses non plus.

astralien3000, 2015-08-04:
Pour archives, je penses à mettre 2 lipo avec un sac dans une boite comme ça : 
http://www.conrad.fr/ce/fr/product/522635/Boitier-universel-Strapubox-2024-ABS-noir-72-x-50-x-26-1-pcs/?ref=detview1&rt=detview1&rb=1

Bon, faut monter/démonter avec un tournevis, mais ça prendrait pas trop de place.

Elioty, 2015-08-04:
Ça termine en poupée russe ton histoire :p Mais c'était l'idée que j'avais aussi hier soir :) Au moins ça évite au sac ignifugé, en tissu, de trainer dans tous les sens selon les plis qu'il fait. Après pourquoi aurait-on besoin de la démonter fréquemment ? Peut-être à l'homologation pour que les arbitres voient le sac ignifugé mais les fils vont en sortir pour pouvoir se brancher dans le robot, et dans le cas de la charge, on pourrait très bien sortir toute la boite (mais sans la démonter). Faut étudier le système de fixation de la boite par contre.

Mais le rendu final pourrait être pas mal (point de vue esthétique je parle), ça fait genre gros bloc comme une batterie de voiture dans le robot :D

Et histoire d'être modulaire, on se fait un jeu de batteries avec son jeu de boites. Une fois les batteries misent dans leur sac, puis le sac mis dans sa boite, on y touche plus et on ne manipule que la boite. Et lorsqu'on achètera d'autres batteries, on achètera aussi leurs sacs et boites qui vont avec, pour avoir, une fois monté, un système de batterie avec sa protection et ses contraintes mécaniques (je veux dire des dimensions physiques "rigide") clé en main.

Donc si tu peux faire rentrer ça et qu'on peut faire ce que j'ai écris au dessus, ça serait vraiment super cool ! Manque plus qu'à intégrer dans la boite un beeper pour nous avertir du niveau faible de la batterie, mais je commence peut-être à en demander beaucoup là ^_^

astralien3000, 2015-08-04:
Les boites pourront théoriquement pas sortir du robot (elles seront directement vissées).
J'ai trouvé une autre boite intéressante : 
http://www.conrad.fr/ce/fr/product/521981/Boitier-module-Strapubox-2028-ABS-noir-85-x-50-x-29-1-pcs/?ref=detview1&rt=detview1&rb=1

Là je suis plus sûre de faire rentrer les batteries.

Elioty, 2015-08-05:
D'accord mais fonctionnellement parlant, ça sera quoi le processus pour sortir les batteries du coups ?
- La boite reste statique dans le robot, on peut juste l'ouvrir simplement et sortir les batteries ?
- La boite est facilement dévissable de la structure du robot ?

Elioty, 2015-08-19:
Je vois dans ton suivi que tu as déjà placé le système de motorisation et les encodeurs. Cela pourrait constituer un bon jalon à faire valider par Black Wolf, non ? À l'exception des batteries peut-être, le reste devrait pouvoir se modéliser/positionner sans trop de problème je pense.

astralien3000, 2015-10-17:
@Elioty, j'ai besoin de la taille de ta carte.
Il me faut la longueur/largeur, la distance entre les trous de vis M3 (elles sont disposées en carré ?), et l'épaisseur max.

astralien3000, 2015-10-17:
La modélisation est terminée !
C'est sur le dépôt si vous voulez voir.

closed 2015-10-17 by astralien3000