xenomorphales/xenomorphales #18: Choix des roues pour Céres

astralien3000, 2015-06-03:
On a déjà repéré ce choix de roues : 
https://www.pololu.com/product/1430

Qui seront montable sur un axe de 5mm avec ça : 
https://www.pololu.com/product/1203

8cm pour 1cm d'épaisseur me paraît correct.

Après il y a une possibilité de prendre une de 90x10mm, étant donné que pour le moment on a un surplus de couple grâce au moteur.

Attention : Il faudra vérifier que la roue est bien compatible avec la fixation à l'axe !

Elioty, 2015-06-03:
Je cite la description de la roue sur sa page :
>  Six additional mounting holes for 4-40 screws make it possible to use the wheel with any of our universal mounting hubs.
Et le mounting hub que tu présentes fait partie de leur "universal mounting hubs" donc en principe ça devrait aller. À la vue des photos et du schéma du mounting hub, je ne suis pas totalement convaincu par contre. Peut-on les contacter pour demander plus de renseignements ? Enfin surtout leur poser la question cash si les deux "collent" bien.
Autre point, les vis de serrage de l'axe sont bien livrées avec les mounting hubs mais pas les vis de fixation avec les roues. On en achète un paquet par précaution ou on se risque à en taxer à Eirbot (avec le risque de ne pas en trouver, ou d'en trouver mais pas en bon état) ?

astralien3000, 2015-06-03:
Il y a quasiment que du M3 à Eirbot, peut être quelques 4-40, mais elles sont rares et trop courtes pour être utilisées comme ça. Je me souviens d'en avoir croisé une ou deux qui pourraient faire l'affaire, mais pas de quoi faire deux roues. Faudra peut être les prendre avec.
Ces américains et leur système impérial...

Elioty, 2015-06-03:
Ah beh en "norme" de vis, je ne m'y connais pas mais américains ou pas, il suffit de faire la correspondance entre les spécs de la roue, du "mounting hub" et du paquet de vis.
Puis le [paquet de 25 \#4-40, 1/2" de long](https://www.pololu.com/product/1962) (effectivement, la longueur est en pouce là... 1.27cm, merci Google) est à 0.99$ donc je crois qu'on ne risque pas grand chose à en acheter un (si ce n'est en s'en faire un stock parce qu'on n'en aura pas besoin d'autant). Elles ne sont pas trop longues celles-ci d'ailleurs ?

astralien3000, 2015-07-07:
Roues commandées !
J'ai profité d'une promo... Vu qu'on a arrêté depuis longtemps le choix et qu'on a pas besoin de mieux.

Voila les mounting hub : 
https://www.pololu.com/product/1999
(les vis seront des M3, rien à carrer !)

Les roues codeuses : 
https://www.pololu.com/product/1420
(Vaut mieux un encodeur 1024 ppr avec ça... Comme le gros qu'on avait du mal à placer avant de les mettre au milieu)

Les roues motrices : 
https://www.pololu.com/product/1430
Tout ce qu'il faut, 8cm de diamètres, 1cm d'épaisseur. Il faudra tester l'adhérence.

closed 2015-07-07 by astralien3000