xenomorphales/xenomorphales #19: RTC dans le système micro-informatique

Elioty, 2015-06-04:
Les Raspberry Pi 2 ne disposent pas de RTC (ni de batterie ou de pile embarquée). L'OS n'a donc aucun moyen de se garder (relativement) à l'heure dans un système autonome comme c'est le cas lors de la participation à la coupe.

J'ai donc pensé à deux solutions :
- la première consiste à régler manuellement l'heure sur l'une des boards pendant le temps de préparation
- la second consiste à utiliser une RTC
Dans les deux cas, une des cartes fait "le maitre du temps" et dispose d'un service NTP pour partager le temps et rester synchronisées avec les autres boards.
La première solution est moins chère (pas besoin de matériel) mais pose des problèmes de contraintes de temps, surtout s'il faut redémarrer la boards maitresse du temps.

Personnellement, je suis évidemment plutôt pour utiliser une RTC.
Après une recherche rapide, je suis tombé sur [ce post de blog](http://blog.remibergsma.com/2013/05/08/adding-a-hardware-clock-rtc-to-the-raspberry-pi/) qui utilise [cette clock](http://afterthoughtsoftware.com/products/rasclock) (qui coute une douzaine d'euro frais de port exclus).
Gros avantage pour Raspbian, le support semble totalement intégré avec les dernières versions, juste un tout petit peu de config pour dire au système que la clock est présente.
Pour les anciennes version de Raspbian, la méthode pour setup la clock sont données donc ça devrait pouvoir s'adapter à ArchLinux et Gentoo.

Si jamais on décide qu'on a besoin d'une RTC et que Gentoo et ArchLinux ne peuvent supporter la clock, on devra (malheureusement ou pas, selon les goûts de chacun) se rabattre sur Raspbian. Je vais quand même réaliser quelques tests dès que possible pour voir si on peut compiler les sources de ce qu'il faut sur Gentoo et ArchLinux pour supporter la clock que j'ai donné en exemple plus haut.

Remarque : j'ai mis le milestone "Summer" vu que le choix de la clock peut impacter le choix de l'OS selon le support de la clock par un OS ou un autre.

astralien3000, 2015-06-04:
Je suis à 100% pour la 2e solution !

Je sais pas à quel point DDS est résistant, mais si on faisait des trucs pas catholiques sur la clock en utilisant ROS, ça faisait des trucs pas géniaux (genre freeze des application en cours jusqu'à rattraper l'heure d'avant le changement)... Ouai, j'ai fais des truc pas catholiques avec la RTC sous Ubuntu, parce que je devais aussi faire le driver de la SYNCLOCK (une board qui avait une précision temporelle de malade !)

Si non, j'ai confiance en toi, tu va arriver à la mettre sous Gentoo !
Idem pour le serveur NTP, j'espère que t'aura pas de problème, parce que pour le coup, je suis incompétent !

Elioty, 2015-06-04:
Pour NTP, ça ne devrait poser aucun problème à mettre en place, le net doit plutôt être garni de tuto et de doc à son sujet (le net incluant le wiki Gentoo qui détaille plutôt bien la configuration des différents softs possibles et imaginables supportés par Gentoo).
Pour le driver de la clock en elle même, ça devrait pouvoir se faire sans problème aussi mais je ne garanti juste pas à 100% que ça marchera sous Gentoo.
Mais c'est vrai que j'adorerais pouvoir mettre un petit logo Gentoo plutôt que Debian/Raspbian sur notre poster, ça fait plus \#unlike :yum: 

Concernant DDS, en lui même, la synchronisation du temps entre les noeuds n'est pas un problème. Par contre revenir dans le passé sur un OS est certainement un problème pour les apps qui tournent dessus. Mais ça peut poser des problèmes d'un point de vue fonctionnel de ne pas avoir une horloge synchronisée si on veut faire de l'échange de messages genre discussion (question/réponse).

astralien3000, 2016-09-02:
La carte RTC n'a pas été achetée, je pense ? Si c'est bien la cas, abandon.

closed 2016-09-02 by astralien3000