xenomorphales/xenomorphales #20: Innorobo 2015

astralien3000, 2015-06-04:
Il y a le forum innorobo 2015 qui se déroule à Lyon le vendredi 3 juillet.
On pourra y rencontrer Optoforce, le futur sponsor de buggybot.
Et on pourrai peut être trouver d'autres sponsors ?

Enfin bon, voila. Vu que tu es pas trop loin Nico, ça peut t'intéresser ?
Clément, si je me rappelles bien, ça peut t'intéresser aussi ?

Voila, qu'en dites-vous ?

nbelin, 2015-06-04:
Heu oui ça a l'air intéressant, on peut peut-être même y amener son CV ;)
Par contre c'est dans la journée ? Vu que j'ai 1h30 de trajet pour aller à Lyon depuis mon boulot, faut que je prenne l'aprem voire la journée entière...
On peut en discuter si vous voulez, j'ai peut-être pas tout compris :)

astralien3000, 2015-06-04:
C'est de 9h à 17h ascii paraît.
[les info](http://innorobo.com/practical-infos/)

Faudra sûrement prendre la journée pour y aller. En tout cas pour nous c'est réglé si on y va.
En tou cas j'ai vu certains participants, et rien que pour les voir, je trouve que ça vaut la peine.

nbelin, 2015-06-04:
Écoute j'ai pas de jours de congés sur ma convention du coup je demanderai mais je ne peux pas promettre d'avoir la possibilité d'y aller. Ensuite en fonction de si vous y allez ou pas, et des participants, je verrai si j'y vais ou pas ^^

astralien3000, 2015-06-04:
Ouaip, pas de problème ^^
C'est une proposition.

Mais demande quand même ça coûte rien.

Elioty, 2015-06-04:
Je serais bien venu mais franchement, je ne sais pas si je peux avoir un jour :/
Concernant Optoforce, ils seront dans quel hall, tu sais ? Parce que le prix du ticket est différent selon le hall que tu veux visiter donc fait bien attention d'aller là où ils sont quoi !

astralien3000, 2015-06-05:
Ils sont dans le hall d'exhibition normalement, j'ai vérifié sur la floormap. Au pire je leur demanderai quand même pour être sûr. Et normalement le prix du visiteur étudiant (ne donnant accès qu'au hall d'exhibition) est de 10€.
Par contre, c'est le déplacement qui est chiant... Le train, c'est même pas la peine si on est à 2. Ca serait plus en voiture.

nbelin, 2015-06-05:
Bordeaux -> Lyon en train c'est effectivement pas le plus pratique :/

Elioty, 2015-06-05:
Ouais mais en voiture, faudrait limite partir la veille ou très tôt dans la nuit :X Puis se taper la route aller/retour pour un jour, ça fait lourd quand même >_<

astralien3000, 2015-06-05:
Ouai, bon, je sais pas si ça vaut le coup finalement. ><
J'aurai proposé...

nbelin, 2015-06-05:
Après c'est un vendredi, je dois vérifier que je serai bien là ce weekend, mais vous pourriez dormir chez nous le soir (70m² :p)

astralien3000, 2015-06-11:
Alors ?
Des nouvelles de ce côté ?

nbelin, 2015-06-11:
Zut j'ai toujours pas demandé si je pouvais prendre la journée (en même temps, mon maître de stage n'a jamais de temps pour moi...). Faudrait vraiment que je me motive à me bouger le cul, mais c'est sûr que si j'ai la possibilité d'y aller et que vous y allez, j'y vais. Si j'ai la possibilité et que vous y allez pas, ça va être compliqué ^^

astralien3000, 2015-06-29:
Donc, après de nombreuses discussions, nous allons à innorobo avec Clément en voiture !
On peut acheter les tickets d'entrée sur place ou sur le site web, il faut qu'on en discute encore un peu.
Mais je pense qu'on peut clore la discussion !

On va voir notre sponsor Optoforce ! (j'ai le contrat entre le mains) !
Et on va voir plein de trucs marrants !

closed 2015-06-29 by astralien3000