xenomorphales/xenomorphales #21: Site web

astralien3000, 2015-06-29:
Il faudrait faire un site web (simple !) pour notre communication.
En effet, si j'arrive à obtenir plusieurs sponsors, ça serait bien de les tenir informés, tout en leur faisant de la pub.
Et puis même pour nous ça serait cool d'avoir un joli site à montrer aux gens ^^ !
github.io devrait faire amplement l'affaire en tant qu'hébergement.

Elioty, 2015-06-29:
Ouais, ça serait bien effectivement. Et je pense qu'on pourrait utiliser jekyll comme le suggère Github pour n'avoir qu'à éditer le contenu sans trop avoir à se soucier du reste.

astralien3000, 2015-07-18:
J'ai commencé le travail !
[lien](http://astralien3000.github.io/projet2016/)

Jekyll marche bien, et est bien pratique.
Une fois tout mis en place, on pourra mettre du vrai contenu.

astralien3000, 2015-07-18:
Voila ! C'est déjà pas mal complet là !
Il faut faire un peu de contenu maintenant, quelqu'un pour m'aider ?
Déjà faudrait que vour remplissiez vos fiches de membre.
Pour le titre, vu qu'on est fnialement pas d'accord pour le nom d'équipe, j'ai laissé celui de l'année passée.

Elioty, 2015-07-20:
Cool tout ça :) Et le template que tu as trouvé rend très bien avec notre "charte graphique/chromatique". Manque plus que le nom, et à remplir (en français et en anglais d'ailleurs).

Reste à voir pour un nom de domaine tout propre une fois qu'on aura bien choisi le nom d'équipe évidemment. Doc pour paramétrer tout ce qu'il faut : https://help.github.com/articles/setting-up-a-custom-domain-with-github-pages/

astralien3000, 2015-07-20:
En parlant d'anglais/français, il y a pas de système natif dans Jekyll qui permette de le faire proprement. Il y a un plugins, mais github le gère pas. Ils donnent la solution d'exécuter le plugins sur son ordi puis de mettre le fichier "build" de jekyll sur le site, mais ça me plaît moyen.
Il y a peut être un moyen de faire un truc avec un système de catégories (géré par jekyll).
Ou si non on met tout sur la même page mais ça fait crade...

Elioty, 2015-07-20:
Ou sinon on écrit tout en anglais dessus sans se caser la tête avec ça. Mais ça parait effectivement moyen de push le build sorti par Jekyll alors que github-pages gère Jekyll tout seul comme un grand (sauf le plugin qui nous intéresserait évidement).

Autre solution, on écrit tout en anglais, et on fait juste les news en double (anglais/français) dans deux onglets différents (on garde le "news" pour l'anglais, on en rajoute un "nouvelles" juste à côté pour les news en français), ça serait possible ?

Sinon question design/ergonomie, je pense qu'il faudrait que la page d'accueil soit les news, plus de page de bienvenu avec le logo en géant. Et à propos du logo, il faudrait le mettre à gauche du titre en haut à gauche de la page je pense.

Elioty, 2015-07-24:
Maintenant que le nom est décidé, on peut le mettre sur le site à la place du titre et se décider pour l'extension du nom de domaine. Je propose .com histoire de rester classique (à 11.99€ la création et 13.19€ pour le renouvellement).

Sinon, en intéressant/fun il y a aussi :
- .cat (création en promo à 6€ et renouvellement à 22.79€), juste pour le fun celui là
- .band (création en promo à 7.19€ et renouvellement à 22.79€)
- .guru (création à 28.79€ et renouvellement à 31.19€)

Bon en fait, on va rester sur le .com je pense (question tarif mais aussi parce qu'il n'y en a pas de vraiment intéressant, le .bot ou .robot n'existent pas :sob:)

J'ai indiqué tous les tarifs en TTC chez OVH. On pourrait éventuellement jeter un coup d'oeil du côté de Gandi aussi.

En fait, il semblerait que le .bot existe mais il ne serait pas encore disponible à la vente, il peut être pré-commandé sur Gandi. Et j'ai du coups jeter un coup d'oeil, xenomorphales.com est à 15.05€ TTC chez Gandi.

astralien3000, 2015-07-24:
xenomorphales.com c'est bien.
On termine de remplir le site avant de s'en occuper par contre.

Elioty, 2015-07-24:
Oui oui, pas de problème.

astralien3000, 2015-07-26:
Vous pouvez remplir vos fiches membres [ici](https://github.com/astralien3000/xenomorphales/tree/gh-pages/_members).
Prenez la mienne comme modèle. Mais si vous avez des suggestions/améliorations à apporter, allez-y !

astralien3000, 2015-09-29:
Bon, le site est plutôt bien mis en route, en plus on l'a présenté à la liste Eirbot, je pense qu'on peut clore l'issue.

closed 2015-09-29 by astralien3000