xenomorphales/xenomorphales #22: Sac à LiPo

astralien3000, 2015-07-07:
Vu que l'intégration du sac est problématique, il faut en trouver un qui soit pas trop encombrant.
Ce site paraît intéressant : 
http://www.hobbyking.com/hobbyking/store/__1114__85__Batteries_Accessories-Lipo_Safe_Bags.html

astralien3000, 2015-08-03:
Céres va aussi avoir des batteries LiPo.
Donc il faut aussi trouver des sac à lipo pour céres.
Grosses contraintes de taille : 
moins de 3 mm d'épaisseur autour de la batterie.

astralien3000, 2015-08-04:
Un sac à LiPo pour des batteries 1S pour Cérès : 
http://www.hobbyking.com/hobbyking/store/__41876__Trackstar_Fireproof_Lipo_Storage_Case_105_x_55_x_30mm_.html

Pas cher, et pas encombrant. Et en plus je vais aussi acheter des servo sur le même site ! Tout ce qu'il faut ?

Elioty, 2015-08-04:
Je croyais que tu voulais embarquer deux 2S dans Cérès, ce n'est pas possible en fin de compte ? Du coups quatre 1S ?!?!

Sinon, je ne sais pas quoi commander de plus sur ce site donc tu peux y aller.

Elioty, 2015-08-05:
Je poste ici vu qu'on n'a pas vraiment d'issue à ce sujet et que ça concerne quand même les LiPo (et que c'est aussi sur hobbyking.com). J'ai parcourus les forums de planète-sciences et principalement [un topic sur les fameux sacs ignifugés pour les LiPo](http://www.planete-sciences.org/forums/viewtopic.php?f=1&t=16526) et donc ils discutent de différentes problématiques de sécurité, le confinement dans un sac permet bien de confiner un départ de feu mais augmente aussi les chances qu'un départ de feu se déclare, vu que la/les batteries chauffent à l'utilisation et qu'elles ne sont pas ventilées cloitrées la dedans (donc à faire attention si on fait de longue session de tests). Autre point qui a été soulevé, la décharge trop importante d'une batterie lithium qui peut aussi déclencher un départ de feu. Et pour cela, quelqu'un a posté ceci : http://www.hobbyking.com/hobbyking/store/__18588__

Tu as déjà un chargeur pro pour la charge (donc en principe bien sécurisé), il nous faudrait juste ça pour nous protéger de la décharge. D'autant plus qu'il ne fait pas que simple beeper, il permet aussi d'afficher la tension et donc d'éventuellement savoir à peu près où en est le niveau de charge.
Par contre, il semblerait que celui ci n'accepte que des 2 à 8S. À voir si on peu en trouver un pour des 1S (vu qu'on parte pour des 1S).

astralien3000, 2015-09-29:
Bon, un petit topos sur les batteries.
Pour Cérès, on a opté pour 4 batteries 1S en série, mais il manque les connecteurs (ce qui est bien en fait, vu qu'on a changé d'avis récemment).
Pour les sacs à lipo, on a les sacs, mais il faut s'assurer qu'ils seront acceptés à la coupe.

Elioty, 2015-09-30:
Pour le connecteur entre les batteries pour les mettre en série, aucune idée (surtout à cette heure-ci).
Par contre pour le connecteur sur lequel on aura le 15V, il faut prendre un XT60 femelle ([celui-ci pour rappel](http://www.conrad.fr/ce/fr/product/209210)) et le positif sur le côté plat et la masse à la pointe. C'est ce qui semble être le "standard" sur toutes les photos que j'ai pu trouver sur le net où ils utilisaient ce connecteur. (C'est la même chose pour les connecteurs en T d'ailleurs)

Elioty, 2015-10-04:
### Liste d'articles (sur Hobbyking)

- 3* [Alarme de décharge](http://www.hobbyking.com/hobbyking/store/__18588__)
- 3* [Packs de connecteurs XT60](http://www.hobbyking.com/hobbyking/store/__9572__Nylon_XT60_Connectors_Male_Female_5_pairs_GENUINE.html)
- 1* [Connecteur/fil pour la mise en série](http://www.hobbyking.com/hobbyking/store/__10264__XT60_Harness_for_2_Packs_in_Series_1pc_.html?strSearch=XT60%20series)
- 2* [Batteries pour Cérès](http://www.hobbyking.com/hobbyking/store/__9499__Turnigy_3300mAh_2S_30C_Lipo_Pack.html)
- 2* [Sac à LiPo](http://www.hobbyking.com/hobbyking/store/__41871__Super_Strong_Flame_Retardant_Silicone_Fiberglass_Bag_180x40x35mm_1pc_.html)

Elioty, 2015-11-20:
Les batteries ont enfin leur connecteur de soudé. Les alarmes de décharge mesurent bien les tensions. Et le câble pour mettre en série les deux batteries fonctionne bien, tension bien mesurée de 15.35V en sortie au multimètre. Je clos l'issue.

closed 2015-11-20 by Elioty