xenomorphales/xenomorphales #23: Surcouche OpenDDS

astralien3000, 2015-08-23:
Il serait intéressant de faire une petite lib de surcouche à OpenDDS pour simplifier la création d'un nouveau projet.

Perso j'aimerai bien reprendre les concepts et la facilité d'utilisation de ROS : 
[Comme dans ce tutoriel](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29)

J'ai un exemple de Publisher/Subscriber qui fonctionne, mais il est pas propre.

Par contre, j'arrive à le fair fonctionner avec la découverte centralisée, mais pas avec le RTPS...

Elioty, 2015-08-24:
Le "problème" de la surcouche avec les middlewares de ce genre est qu'on doit écrire un parser qui prend tes .idl en entrée pour sortir ta surcouche qui te permettra de travailler plus simplement par la suite. Ce n'est pas tellement une difficulté, c'est parfaitement réalisable. Néanmoins, à moins de réaliser une surcouche très large, elle limite généralement les possibilités de configuration. Et en voyant toutes les combinaisons de paramètres de qualité de service possibles avec DDS, si on veut bien cacher toutes la lib OpenDDS tout en gardant tous les paramètres configurables, ça fait du boulot. Sachant que ces paramètres seront à bien _fine-tuner_ pour tirer le maximum de performance (sans que tout s'écroule) de noter réseau Wifi. Personnellement, je pense que mettre tous les pointeurs vers les différentes entités de DDS dans une structure (ou une classe) devrait suffire avec des fonctions (ou des méthodes) du genre `dds_init` à "l'ancienne".

Pour le RTPS, il suffit selon la doc de faire un ficher rtps.ini à côté de tes exécutables avec le contenu suivant :
```
[common]
DCPSGlobalTransportConfig=$file
DCPSDefaultDiscovery=DEFAULT_RTPS
[transport/the_rtps_transport]
transport_type=rtps_udp
```
Tu lances ensuite tes exécutables publisher et subscriber avec les arguments suivants : `-DCPSConfigFile rtps.ini`

astralien3000, 2015-08-24:
Pour le RTPS, justement, j'ai essayé avec cette configuration. Mais les deux programmes se voient pas. En fait si, on peut voir dans les logs (DCPSDebugLevel=10) que le second programme lancé essaie de prendre le même port unicast que le premier, mais n'y arrive pas vu qu'il est déjà utilisé. Une fois les deux participants lancés, ils ne se voient pas. Et ça m'a pris la tête toute une soirée.
Je me demande si c'est pas mon ordi/ma box qui supporte pas le multi/unicast. Je comprends pas grand chose sur ce genre d'adresse ip (239.255.0.1).

Si non pour la surcouche, c'est pas pour garder les paramètres QoS configurables. En fait, je considère qu'on utilisera qu'un nombre restreint de configuration différentes. L'utilisateur devra juste dire "je vais envoyer des données capteurs, donc j'ai besoin de cette classe", ou "je vais envoyer une alarme, donc j'utilise cette autre classe", etc...
Un petit utilitaire pour ne pas avoir à choisir à chaque fois les QoS, et avoir des modèles.

Elioty, 2015-08-24:
Le message de log te dit le numéro de port exact qui pose problème ? Que j'essaie de le trouver dans les valeurs par défaut de la configuration pour savoir quel champs il faut changer.
Pour savoir si le multicast fonctionne sous ton Linux, fais juste un `ifconfig eth0 | grep MULTICAST`  (en remplaçant eth0 par le nom de ton interface évidemment). S'il n'est pas actif, tu peux l'activer avec `ifconfig eth0 multicast` et si ça ne marche pas, c'est que ton noyau est compilé sans le support du multicast mais ça, ça m'étonnerait vraiment beaucoup ! Pour ton équipement réseau, rien ne devrait empêcher le multicast de fonctionner dans ton réseau local.

Pour la surcouche, je pensais que tu parlais de carrément faire une bibliothèque complète qui t’offusque au maximum que ça utilise OpenDDS dessous. Moi je vois ça plus comme une simple encapsulation de bout de code dans une ou des classes. Après, rien ne nous empêche de faire de la méta-programmation à coups de template ou de macro-define pour nous faciliter le boulot pour supporter un type de plus.
Pour les paramètres de QoS, je suis d'accord avec toi qu'on va avoir besoin d'au maximum 2 ou 3 profils différents. Et pour s'en servir, je pensais simplement faire des fonctions (ou méthodes) comme la méthode `DomainParticipantFactory::get_default_participant_qos` par exemple à laquelle tu donnes ton profile de QoS non initialisé et la fonction (ou méthode) te le remplis pour toi.

astralien3000, 2015-08-27:
Voici ce que me donne le subscriber :
```
(11656|140648859277120) NOTICE: "DCPSPersistentDataDir" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "DCPSPendingTimeout" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "DCPSPublisherContentFilter" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "ORBLogFile" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "ORBVerboseLogging" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "FederationRecoveryDuration" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "FederationInitialBackoffSeconds" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "FederationBackoffMultiplier" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "FederationLivelinessDuration" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "scheduler" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "scheduler_slice" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "queue_messages_per_pool" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "queue_initial_pools" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "max_packet_size" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "max_samples_per_packet" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "optimum_packet_size" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "thread_per_connection" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "datalink_release_delay" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "datalink_control_chunks" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "local_address" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "multicast_group_address" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "multicast_interface" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "nak_depth" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "ttl" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "nak_response_delay" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "heartbeat_period" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "heartbeat_response_delay" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: "handshake_timeout" is not defined in config file - using code default.
(11656|140648859277120) NOTICE: Service_Participant::load_domain_configuration failed to open [domain] section - using code default.
(11656|140648859277120) NOTICE: Service_Participant::intializeScheduling() - no scheduling policy specified, not setting policy.
(11656|140648859277120) Service_Participant::get_discovery: returning default RTPS discovery for domain 42.
(11656|140648859277120) Spdp::SpdpTransport::open_unicast_socket() - opened unicast socket on port 17910
(11656|140648859277120) Spdp::SpdpTransport::SpdpTransport joining group  239.255.0.1:17900
(11656|140648859277120) DataLinkSet send_control_element_allocator 20afb58 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20afd08 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20b00d8 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20b0288 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20b0658 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20b0808 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20d5ba8 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20d5d58 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20d8158 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20d8308 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20da5f8 with 20 chunks
(11656|140648859277120) DataLinkSet send_control_element_allocator 20da7a8 with 20 chunks
(11656|140648859277120) ip interface 127.0.0.1:0 maps to hostname localhost.localdomain
(11656|140648859277120) found fqdn localhost.localdomain from 192.168.0.11:0
(11656|140648859277120) DirectPriorityMapper:thread_priority() - mapped TRANSPORT_PRIORITY value 0 to thread priority 0.
(11656|140648859277120) TransportSendStrategy replaced_element_allocator 20f1088 with 20 chunks
(11656|140648859277120) TransportReceiveStrategy-mb Cached_Allocator_With_Overflow 20f3b08 with 1000 chunks
(11656|140648859277120) TransportReceiveStrategy-db Cached_Allocator_With_Overflow 20f3b80 with 100 chunks
(11656|140648859277120) TransportReceiveStrategy-data Cached_Allocator_With_Overflow 20f3bf8 with 100 chunks
(11656|140648859277120) TransportImpl::configure()
   transport_type                : rtps_udp
[... cut ...]
(11672|139908204554048) TransportClient::send_i: no links for publication 01030024.542abe6f.2d980000.00000002(3898fcf1), not sending element 0x130cf10 for transaction: 2.
(11672|139908204554048) WriteDataContainer::data_delivered 130CF10 
(11672|139908204554048) WriteDataContainer::data_delivered: domain 42 topic test_topic publication 01030024.542abe6f.2d980000.00000002(3898fcf1) pushed to HISTORY.
(11672|139908204554048) WriteDataContainer::data_delivered: Inserting acked_sequence: 2
(11672|139908204554048) WriteDataContainer::data_delivered - broadcasting wait_for_acknowledgments update.
write
```

Elioty, 2015-08-28:
Alors oui il semblerait bien le second tente d'utiliser le même port mais il se rend bien compte qu'il est utilisé et en essaie donc un autre, avec succès :
```
(11672|139908204554048) Spdp::SpdpTransport::open_unicast_socket() - failed to open unicast socket on port 17910 ACE_SOCK_Dgram::open: Address already in use.  Trying next participantId...
(11672|139908204554048) Spdp::SpdpTransport::open_unicast_socket() - opened unicast socket on port 17912
```

Donc pour le fait que le reader ne reçoit/voit pas le message, j'ai deux idées qui me viennent rapidement en tête :
- Es-tu sûr d'avoir des paramètres de QoS compatibles entre ton reader et ton writer ? Tu peux le déterminer en mettant un DataReaderListener (si ce n'est pas déjà fait) qui surcharge la méthode `on_requested_incompatible_qos`. Si la méthode est appelée, c'est qu'un writer a été trouvé sur le même topic mais avec des paramètres de QoS incompatibles.
- Combien de message écrit ton writer ? Selon les paramètres de QoS et s'ils se découvrent après que le message ait été envoyé, le reader peut ne jamais le recevoir. Si tu veux que le writer délivre les messages écrit par le passé à de nouveaux arrivants, il faut mettre un historique avec une profondeur non nulle et la durabilité au minimum à TRANSIENT_LOCAL.

Sinon point de vue documentation, celle d'OpenDDS est "vide" dans le sens où toute l'API est bien passé à la moulinette Doxygen mais il n'y aucun commentaire explicatif. Mais DDS étant un standard (et je tend à croire que même l'API est standardisée), tu peux te référer à la document de RTI DDS (une implémentation propriétaire) qui est bien plus complète en terme d'explication.
http://community.rti.com/rti-doc/510/ndds/doc/html/api_cpp/group__DDSDcpsModule.html
De même, 99% des problèmes que l'on peut avoir avec DDS sont de l'ordre de la compréhension du standard en lui même donc tu peux aussi lire les forums de RTI lorsqu'ils sortent dans tes recherches Google.

astralien3000, 2015-08-30:
J'ai utilisé les paramètres QoS par défaut pour les deux. Ça m'étonnerai que la lib les changent d'un programme à l'autre.
Et le writer écrit des messages indéfiniment. Donc même si on en rate, on est sencé en recevoir. Et j'ai essayé d'en envoyer pendant plusieurs minutes.

kazord, 2015-09-15:
Probleme 'basique' : openDDS doit utiliser l'ip ou la mac de l'interface dans la fonction 'discovery', du coup il n'est pas capable de se voir lui même.
Faudrai que je creuse un peu plus, mais pour l'instant, le seule moyen que j'ai trouver pour rtsp en 'local' c'est de crée une deuxième interface virtuel avec un ip différente du même réseau

Elioty, 2015-09-15:
Sérieux ? Tu as trouvé ça où dans la doc de OpenDDS ? Ou tu as fouillé le code et/ou les logs comme un barbare ? Et le format de l'identité d'un DomainParticipant n'est pas standardisé ? (ça pourrait être simplement `ip-pid` ou `mac-pid` (peu importe le séparateur) et OpenDDS a décidé de faire autrement évidement...)

Et l'idée quand même derrière notre utilisation de DDS est que le système soit totalement réparti sans point central. Par exemple, cela devrait permettre, dans le cas de code qui publie des données uniquement en fonction des données auxquelles il est abonné (comprendre qu'il n'extrait pas d'info depuis du matériel), d'exécuter ce code sur n'importe quelle Rasp de notre système. Ce qui pourrait être le cas d'un processus type IA stratégie qui décide qui fait quoi quand par exemple pour coordonner un peu mieux les robots que juste chaque robot dit ce qu'il est en train de faire pour éviter que l'autre le fasse aussi en même temps.

astralien3000, 2015-09-15:
Wow, ça pose un gros problème. J'espérais pouvoir faire une architecture en "noeuds", qui peuvent être sur une seule machine. Donc avec plein de participants sur la même machine, qui communiquent entre eux, et avec les autres machine de façon trasparente. Si il faut créer une interface virtuelle par participant ça me parraît pas génial.

C'est une utilisation anormale de DDS ? Ou c'est juste OpenDDS qui fait n'importe quoi ?

kazord, 2015-09-15:
j'ai écrit 'doit' dans le sens je suppose, si je teste mon code (exemple ou modifier) sur la meme machine avec opendds ca marche pas, sur 2 machine différente avec les memes programmes, ca marche niquel
DDS sur la même machine est normaliser avec passage par de la mémoire partagé (a voir si tu peux utiliser ca pour des noeuds multiple de nombre (in)connu

avec rti par contre, 2 applis sur la meme machine ca marche sans changement des valeurs par défault (faudrai que je reteste avec wirshark voir si ca passe par le réseau ou la mem)

Elioty, 2015-09-16:
Je confirme avec RTI DDS (implémentation proprio), ça marche nickel sur une seule machine sans bidouille (j'ai dû aller jusqu'à facile 10 processus en même temps en stage sans problème), ce qui est censé être le fonctionnement de base de DDS vu que le middleware est normalement là pour te cacher toute la complexité du partage sur le réseau (ou en mémoire partagée).

Concernant la mémoire partagée et OpenDDS, ce moyen de transport est disponible de mémoire, à voir comment écrire le fichier de config pour qu'un processus puisse utiliser en même temps RTPS et la mémoire partagée.

kazord, 2015-09-16:
comparaion de ma version pour le cours de réparti et la paf, 2 lignes en plus : 

    // Block until Publisher completes
    DDS::StatusCondition_var condition = reader->get_statuscondition();
    condition->set_enabled_statuses(DDS::SUBSCRIPTION_MATCHED_STATUS);
 
    DDS::WaitSet_var ws = new DDS::WaitSet;
    ws->attach_condition(condition);

Ce code est placé juste avant ma boucle de réception des messages (info sur ce que ca fait en fin de post)
modification et retest avec openDDS, ca marche bien en faite (ouf) (erreur de débutant/ignorant powa)

PS: http://community.rti.com/rti-doc/45f/ndds.4.5f/doc/html/api_java/classcom_1_1rti_1_1dds_1_1subscription_1_1SubscriptionMatchedStatus.html]subscriptionmatched

Elioty, 2015-09-16:
Hum, ceci permet juste de bloquer le processus subscriber jusqu'à qu'un "match" est fait, ça ne devrait changer en rien le fonctionnement de la découverte et du match des publishers/subscribers sur le même topic avec QoS compatible.

D'autant plus que Loïc arrive très bien à faire marcher son code avec le système DCPS centralisé. Le middleware est là pour faire abstraction, le même code devrait tout aussi bien marcher en RTPS+SharedMemory sous réserve que la config soit correcte.

astralien3000, 2016-05-10:
Alors, OpenDDS, c'est pas génial.
Il faut trouver d'autres libs (open sources) implémentant DDS.
En attendant, j'ai utilisé ZMQ sur buggybot pour la communication entre les noeuds.
C'est pas la même chose que DDS, mais ça faisait le travail.

closed 2016-05-10 by astralien3000