xenomorphales/xenomorphales #27: Fair l'asserv de Céres

astralien3000, 2015-10-04:
Pour toi Clément
Asserv en deux étape : 
 - [ ] Asserv des moteurs en vitesse
 - [ ] Asserv en angle/distance ou angle/vitesse
 - [ ] Bonus : Positionnement en XY

astralien3000, 2016-05-22:
Je ne peux que conseiller d'utiliser cette base de code : 
https://github.com/AversivePlusPlus/project-2wheel-asserv

Elioty, 2016-05-22:
Oui, c'est bien l'idée et merci beaucoup pour tout ce travail de fait déjà :)

astralien3000, 2016-05-22:
De rien, je l'ai fait pour m'amuser ^^. D'ailleurs je continuerai sûrement à l'améliorer avec toi.
Ca me fait penser qu'on a "oublié" une tâche : les drivers pour Céres. Même si c'est un module Aversive++, c'est plus pour Xenomorphales qu'on le réalise. Et en plus elle est pas listée dans le dépôt Aversive++.

astralien3000, 2016-09-02:
Bon, l'asserv est faite, il reste juste à la lier au hardware (kinetis).

closed 2016-09-02 by astralien3000