xenomorphales/xenomorphales #32: Noeud DDS récupérant les données des capteurs optoforce

astralien3000, 2015-10-04:
## Entrées

Les capteurs communiquent par port série, normalement il ne devrait pas y avoir de problème sous linux.
Optoforce donne une lib avec ses capteurs, je ne l'ai pas testée...

## Sorties
```C
struct Force {
  double x;
  double y;
  double z;
}
```

On a un topic par patte : 
 - LF_force
 - RF_force
 - LB_force
 - RB_force

Elioty, 2015-10-04:
Est-il vraiment nécessaire de faire un topic par patte ?
Je pense qu'il serait mieux de faire un seul topic et de rajouter un ID dans ta structure. Ce sera aussi bien plus simple à faire de la factorisation comme cela en ayant plusieurs instances dans le même topic que d'avoir plusieurs topics.

astralien3000, 2015-10-04:
Ha ouai, ça serai mieux en effet !

```C
enum Leg {
  LF,
  RF,
  LB,
  RB
};

struct Force {
  Leg leg;
  double x;
  double y;
  double z;
}
```

Avec un seul topic : 
`optoforce`

Elioty, 2015-10-04:
Oui est n’oublie pas de mettre le commentaire/tag "clé primaire" sur le champs leg.
Et pour le nom du topic, je préciserai buggybot dedans si jamais on utilise d'autres capteurs Optoforce dans le futur, sait-on jamais !

astralien3000, 2015-10-04:
Vu le prix... Chaud !
Mais si on devient riche avec Aversive++ pourquoi pas ! XD
Mais ok pour le nom.

astralien3000, 2016-05-10:
C'est pas DDS, mais c'est fait.

closed 2016-05-10 by astralien3000