xenomorphales/xenomorphales #33: Nom de domaine

astralien3000, 2015-10-05:
xenomorphales.com

Il faudrai l'acheter.

Elioty, 2015-10-05:
Je pense que ça va de pair avec l'hébergeur. Tant qu'à garder le choses simples, autant prendre le nom de domaine via la même entreprise qui nous fournira l'hébergement.

Et comme on en a discuté hier, je pense que OVH avec ses VPS SSD répondrait très bien à notre besoin.
https://www.ovh.com/fr/vps/vps-ssd.xml

Pour les .com, ils sont à 11,99€ TTC pour la création puis 13,19€ TTC pour le renouvellement chaque année, ce que je ne trouve pas excessif.
https://www.ovh.com/fr/domaines/tarifs/

astralien3000, 2015-10-05:
Je pensais pas utiliser un serveur OVH pour Xenomorphales, mes interrogations étaient pour Aversive++. Tu penses que c'est mieux d'héberger le site de xenomorphales aussi sur OVH ? (Donc se taper 2 serveurs ?)

Elioty, 2015-10-05:
Ha oui c'est pas faux. Le site sur Github suffit largement pour Xenomorphales. Mais quitte à éviter de se disperser, autant prendre le nom de domaine pour Xenomorphales au même endroit que celui pour Aversive++.

astralien3000, 2015-10-05:
Oui, pour le coup.

astralien3000, 2015-10-09:
Alors, le nom de domaine est acheté, mais ils faut mettre en place la redirection.
ASCII paraît, c'est long...

astralien3000, 2015-10-09:
[Done !](http://xenomorphales.org)

Elioty, 2015-10-09:
Je ré-ouvre parce que c'est "moche" une "vraie" redirection (je pensais que tu parlais d'autre chose en parlant de redirection), j'entends par là que l'URL du site au final reste http://xenomorphales.github.io/xenomorphales/

Voici la doc de Github pour avoir son propre nom de domaine comme URL :
https://help.github.com/articles/adding-a-cname-file-to-your-repository/
https://help.github.com/articles/tips-for-configuring-a-cname-record-with-your-dns-provider/

Avec xenomorphales.org dans le fichier CNAME.

astralien3000, 2015-10-09:
Bon, je t'avouerai que c'était pas la même config quand j'ai fermé l'issue...
En gros vu que le site est dans un "sous dossier", j'avais le chois entre une redirection invisible et une redirection visible.

J'ai tenté l'invisible. Tout le site était sous xenomorphales.org, c'était bien, sauf que je me suis rendu compte que si on clique sur un lien externe, ça reste sous xenomorphales.org... Pas bon.
Donc ça m'a gonflé, et j'ai mis une redirection visible.

Elioty, 2015-10-09:
Si j'ai bien compris la doc de Github, il te faut le fichier CNAME comme tu as fait déjà.
Ensuite sur ton DNS il te faut l'une ou l'autre ou les deux (à voir mais je pencherais plus pour les deux) :
```
xenomorphales.org.              3592    IN      CNAME   xenomorphales.github.io.
www.xenomorphales.org.          3592    IN      CNAME   xenomorphales.github.io.
```

Cela permet de dire que les deux adresses à gauche pointent vers la même chose que xenomorphales.github.io. Et le fichier CNAME dans le dépôt permet d'indiquer au serveur HTTP de Github que telle adresse correspond justement à notre dépôt.

astralien3000, 2015-10-10:
Bon, normalement, c'est bon, peux-tu vérifier s'il te plais ?
[xenomorphales.org](http://xenomorphales.org)
[www.xenomorphales.org](http://www.xenomorphales.org)
(si tu tombe sur la même adresse avec les deux liens, c'est normal, www.xenomorphales.org redirige vers xenomorphales.org)

Elioty, 2015-10-10:
L'adresse sans le www fonctionne très bien pour moi. Celle avec le www pas encore, je suppose juste que le DNS ne s'est pas propagé encore.

Elioty, 2015-10-10:
Ok c'est bon ça fonctionne pour moi, tu peux fermer l'issue si ça marche pour toi aussi ;)

closed 2015-10-10 by astralien3000