xenomorphales/xenomorphales #34: Électronique Buggybot

astralien3000, 2015-10-07:
# Besoins

## Servomoteurs

### Contrôle en position

La carte doit pouvoir contrôler 4 servomoteurs.
La spécificité est que les servomoteurs peuvent être soit hackés soit non modifiés.
On doit pouvoir utiliser les deux configuration de servomoteurs juste en changeant de connecteur.

Quand le servomoteur est hacké, le microcontrôleur doit pouvoir lui envoyer une commande de vitesse et de sens. Quand le servomoteur est non modifié, le servomoteur doit pouvoir lui envoyer une commande de position. Dans les deux cas, il doit y avoir la possibilité de brancher un potentiomètre avin de déterminer la position du moteur.

### Mesure de force

On doit pouvoir mesurer le courrant pour chaque servomoteurs afin de pouvoir déterminer la force développée par le moteur.

### Alimentation

Les servomoteurs doivent pouvoir être alimentés avec une tension entre 5 et 7.5 Volts. Il vaut mieux que la tension d'alimentation soit facilement modifiable. Chaque servomoteur peut avoir un voltage différent
L'alimentation doit pouvoir tenir 2A par servomoteurs.

## Communication

### Programmation

La programmation doit se faire à l'aide d'un cable USB (mini ? micro ?).
Cela implique soit la présence d'un programmateur sur la carte, soit la présence d'un bootloader.

### Communication Rasp

La communicaion avec la raspberry doit se faire via un cable USB.
Le port USB peut être le même que le port de programmation.
Dans ce cas, il y a la possibilité de mettre un jumper pour interchanger les deux fonctions dans le cas d'un programmateur.

### Réseau I2C

La carte doit comporter un port I2C.
Le port I2C permetra la communication avec des modules tels que un accéléromètre ou gyroscope.

## Alimentation

### Logique

La logique sera alimentée par une batterie USB Lithium-Ions.
Le connecteur d'alimentation ne sera pas nécessaiement un cable USB.

### Puissance

La puissance sera alimentée par une batterie LiPo 2S 7.5V

astralien3000, 2015-10-17:
J'ai une petite information qui peut être utile : 
J'ai parlé à un roboticien bidouilleur chevronné, qui m'a expliqué qu'il fallait éviter de mettre des régulateurs de tensions pour alimenter des servomoteurs. En effet, les servomoteurs ont une demande très erratique en puissance, et les régulo aiment pas trop ça.

Il m'a aussi dit qu'il utilisait la plupart des servo directement sur 7.5/8V (soit la tension d'une batterie LiPo 2S), même les servo chinois à 7€, qui sont censés supporter 6V max !

Je viens de tester les vieux servos de buggybot avec du 7.5 (alors qu'ils étaient régulés en 5V), et j'ai une jambe complète qui est hors service. Mais cette jambe avait déjà un servo crammé, et le deux autres, par contre, je sais pas... Le reste des jambes ont tenu plusieurs minutes en 7.5 sans problème visible, et fonctionnent toujours.

Enfin bref quoiqu'il en soit, je comptais m'acheter de nouveaux servos qui sont censé tenir du 7.5 (ils écrivent même qu'ils sont faits pour être directement branchés sur une LiPo 2S. Donc, pour ceux là, de toute façon, pas de régulo ?

Elioty, 2015-10-18:
Beh si les servo que tu comptes utiliser supportent bien 7.5V et sont même faits pour être directement alimentés par une LiPo 2S, on les alimente direct par une LiPo 2S, pas de question à se poser :p Juste un peu de découplage au plus près du servo (enfin, du connecteur sur lequel il est branché) et ça ira ;)

astralien3000, 2015-10-19:
Ouai, mais je n'utiliserai certainement pas que ces servos.
Donc je voulais savoir si des servos de moins bonne qualité pouvaient survivre, ce qui semble être en partie le cas.

astralien3000, 2016-02-29:
Cette issue n'est plus d'actualité, j'en recommence une

closed 2016-02-29 by astralien3000