xenomorphales/xenomorphales #37: Discussion architecture logicielle BigBrother

astralien3000, 2015-10-07:
Comme @nbelin a commencé la discussion par mail, j'en fais un résumé ici : 
On pensait donc que faire 4 instances de classes "Balises" dont le but était de trouver la balise associée était une bonne idée.

Il faut par contre penser à mutualiser l'utilisation des filtres OpenCV afin de ne pas faire deux fois le même calcul.

Nicolas a aussi pensé à faire une recherche en parallèle sur plusieurs coeurs afins de l'accélèrer.
Attention aux problèmes de concurrence cependant.

nbelin, 2015-10-12:
C'est parti pour un petit pavé sur mes dernières idées :)

Objectif principal : avoir un code exploitant le mieux possible les différents coeurs de la Raspberry !

Les différents threads que j'ai pensés :
- thread principal : récupère les images de la caméra (voire les sauvegarde sur une carte SD), les traite (RGB -> HSV, redimensionnement, etc.), ET les met dans un buffer circulaire d'images (tout en signalant aux autres threads qu'une nouvelle image n°X est disponible
- un thread par balise à détecter : récupère le pointeur vers l'image la plus récente, et cherche la balise associée dans l'image. Applique une recherche localisée de balise, et étend la recherche si celle-ci n'est pas trouvée. Dès que la balise est trouvée, le thread indique au réseau (cf. Elioty et ses protocoles) la position de la balise ET l'ID de l'image (ou l'âge de l'image) dans laquelle la balise a été trouvée. Si la recherche a été étendue à toute la largeur de l'image et que la balise est restée introuvable, le thread PEUT indiquer au réseau qu'il n'a pas trouvé la balise. Dans tous les cas, le thread récupère le pointeur vers la nouvelle image la plus récente.

FAQ :
- pourquoi un buffer circulaire d'images ? Pour laisser chaque image reçue un certain temps en mémoire et donc laisser du temps aux threads pour la traiter (typiquement si un thread ne retrouve pas sa balise, il prendra plus de temps)
- quelle taille de buffer ? A voir expérimentalement, mais je pense qu'un buffer de 40 images (soit environ 40 Mo) reste raisonnable en mémoire, et laisse un peu plus d'une seconde à un thread pour traiter l'image (et on peut considérer qu'au delà, le thread doit récupérer une nouvelle image).

Avantages :
- si un thread perd de vue sa balise, il gênera pas la détection des autres balises qui resteront réactives
- si une balise tombe, un thread plante, ou quoi que ce soit, l'ensemble du système peut toujours fonctionner correctement
- ça fait un code sexy à écrire ;)

Inconvénients :
- est-ce que c'est pas un peu de l'overkill ?
- attention aux sources de bugs !

astralien3000, 2015-10-14:
J'aime bien l'idée qu'un thread ne ralenti pas les autres si il met du temps pour trouver sa balise.
Par contre, faudra faire un bon horodatage de l'information.

Et il faudra, mais tu t'en doutes, bien blinder tes threads, parce que quand un thread plante, il y a des chances que les autres suivent !

Si non, ça ressemble pas trop à de l'overkill, tant que tu gère bien tes classes, etc...
Il suffit que tes algos soient thread safe, et après tu ajoute une couche de multithreading par dessus.
(Il faut que tes classes soient indépendantes du fait qu'il y ai un thread ou pas).

closed 2016-09-02 by astralien3000