xenomorphales/xenomorphales #39: Détection de balise

nbelin, 2015-10-10:


nbelin, 2015-10-11:
Rectangle AABB simple coloré bientôt détecté. Ça commence à marcher mais tourne à 75 Hz pour un seul rectangle suivi :/
Quand la détection sera fiabilisée, je cocherai la tâche et passerai à la mise en commun de 2 rectangles colorés :)
Enfin, faudra optimiser !

astralien3000, 2015-10-11:
Ouai mais elles font quelle taille tes images ? Parceque du 75Hz avec des images de 1600x1200 c'est pas la même que 75Hz avec 640x480.

nbelin, 2015-10-12:
Effectivement hier soir j'ai écrit ça un peu en catastrophe avant de me coucher sans préciser grand chose :
- images de 640x480
- "flood fill" non cache-friendly (= "region growing")
- seg fault si je commente "cv::imshow("img", frame);"

Pour les commentaires à froid :
- les images sont de 640x480, mais ce que j'ai écrit ne dépend pas trop de la résolution de l'image, mais surtout de la taille apparente de la balise (car plus de pixels à parcourir lors du flood fill). J'envisage de diviser la taille de toutes les images reçues par 4, et d'appliquer la détection d'une balise sur l'image originale si celle-ci est loin, et sur l'image réduite si celle-ci est proche.
- pour l'instant le flood fill rajoute les pixels un par un dans une pile. J'ai essayé de faire en sorte qu'en moyenne une ligne de pixels soit traitée entièrement avant de passer à la suivante (dans la façon dont j'ajoute les pixels à la pile) mais c'est pas parfait (notamment sur une même ligne ça peut aussi revenir en arrière, et donc à chaque fois recharger le cache). Du coup il y a encore des améliorations possible pour rendre le flood fill plus cache-friendly. (CEPENDANT, optimiser des histoires de cache alors que je suis pas sur la machine cible, ça me parait prématurée comme démarche, du coup j'y reviendrai probablement bien plus tard quand l'ensemble fonctionnera sur Raspberry)
- alors oui le code segfault si je commente la ligne "cv::imshow("img", frame);" alors que celle-ci ne devrait avoir absolumen AUCUNE influence sur l'exécution du code ! (pour le coup je comprends pas trop xD)
- je compile en -O3, du coup aucune amélioration "gratuite" des performances à attendre juste en optimisant la compilation
- dans le 75 fps que j'ai mesuré, ça comprend quelques manipulations d'image communes à toutes les balises (notamment transformation de l'espace de couleur), AINSI qu'une transformation CONTRE performante que je me suis rendu compte aujourd'hui (allez savoir pourquoi je me suis amusé à séparer mes cannaux de couleur ?)
- la détection des rectangles n'est de toute façon par encore bien finie (ça ne marche pas si le rectangle se déplace vers la droite dans l'image par exemple ^^)
- si le rectangle disparait de l'image, c'est fini !

Résumé :
- un frame rate pas aussi bon que j'espérais, malgré des optimisations évidentes restantes que j'ai en tête et que je vais essayer de mettre en place ces prochains jours
- les optimisations de cache seront laissées de côté, je vais en priorité intégrer cette détection de rectangle avec la détection plus haut niveau de balise (détection de balise à temps=0, ne pas être "trop" perturbé si la balise disparait, etc.)

kazord, 2015-10-12:
ne connaissant pas trop la detection de carrée dans une image par opencv je lance l'idée comme ca :
- Est il possible de partir de la position supposé (image d'avant ou déduit de la coordonée du robot) dans l'image pour 'trouver' le carrée et donc sa taille ? (partir du centrer et élargir ligne par ligne)

nbelin, 2015-10-13:
C'est justement ce que fait le code pour l'instant : il connait la position du rectangle dans l'image t-1, et cherche à étendre une région de la bonne couleur à cette même position. Après pour bien faire les choses, il faut pas étendre bêtement la région, mais faire attention à être gentil avec son cache dans son parcours de l'image :)

nbelin, 2015-10-19:
J'ai approfondi les tests sur une autre balise de la vidéo : j'arrive à suivre le rectangle jaune à 150 Hz ! Peut-être qu'enlever l'option -g de la compilation a un peu aidé ;)
(en effet, l'option -g peut désactiver certaines optimisations du genre déroulement de boucles, fonctions inline, etc. qui gênent le débogage...)

Le souci est qu'à un moment, la balise est perdue pour aucun raison apparente, ce qui montre que le suivi de rectangle coloré n'est pas encore tout à fait au point.
Le code ne sait pas encore retrouver une balise perdue, mais ça, c'est pour une autre histoire...

astralien3000, 2016-10-17:
Le message de départ est encore d'actualité ? Si c'est pas le cas, il vaut mieux le modifier.
Allez, tu nous ponds un truc pour décembre ? ^^

astralien3000, 2017-04-07:
Selon le premier message, cette tâche est finie, non ?

closed 2017-05-04 by astralien3000