xenomorphales/xenomorphales #44: Électronique Buggybot

astralien3000, 2016-02-29:
# Besoins

Je veux pouvoir brancher (au moins) 12 servomoteurs [Feetech SCS15](http://www.lextronic.fr/P36125-servo-avec-retours-dinformations.html). Pour cela, je vais utiliser 2 [boards](http://www.lextronic.fr/P36132-module-de-conversion-seriettl.html) qui transforment une communication UART en communication half-duplex pour les moteurs. Tant qu'à faire, un 3eme uart de sorti pour les moteurs, c'est mieux. Et un 4e pour la communication avec la Rasp !

Il me faut aussi 4 entrées ADC pour des GP2, 4 sorties PWM (pour des servomoteurs plus conventionnels). Un port de communication I2C (pour une centrale inertielle).

Les servomoteurs SCS15 seront alimentés par une tension de 7.5 volts (venant d'une batteire LiPo 2S)
Les servomoteurs plus conventionnels seront alimentés pas une tension de 5 volts (venant de la même batterie LiPo, mais avec une [régulateur à découpage](https://www.pololu.com/product/2851)).
L'alimentation logique sera une batterie USB pour téléphones portable.

La board de contrôle sera une arduino Mega2560. J'ai bien vérifié qu'elle comporte les 4 uarts. Le reste allant de soit.

# Tâches

- [ ] Spécifications
- [ ] Schématique
- [ ] Routage
- [ ] Achat des composants
- [ ] Tirage du PCB
- [ ] Soudage

astralien3000, 2016-02-29:
Tu le sent le truc qu'on va pas arriver à faire en 2 mois vu qu'on a tout le reste à faire ?

Elioty, 2016-03-01:
Donc si je comprend bien, on partirait sur un shield pour ADM2560 ?
Avec sur ce shield tous les connecteurs nécessaires mais :
- pour le module "régulateur à découpage", tu veux le fixer sur le shield ou ailleurs ?
- pour les deux modules UART <> TTL, tu veux les fixer comment ? les scratcher ailleurs sur le robot ?

Sinon point de vue spécification, tu as quasiment tout, le matos que tu as donné semble répondre à ton besoin, reste juste à faire la "pin map" pour les spéc a priori, et le choix des types de connecteur.

Sinon, c'est jouable en deux mois je pense ;) Une fois ces quelques interrogations répondues, je pourrais me lancer dessus (je ne sais pas pourquoi mais je suis plus enthousiasmé à l'idée de faire ça que de faire les drivers Aversive++ pour les MK20D7) :laughing:

astralien3000, 2016-03-03:
> (je ne sais pas pourquoi mais je suis plus enthousiasmé à l'idée de faire ça que de faire les drivers Aversive++ pour les MK20D7) :laughing:

Ok, on fait un échange de tâches XP !
Mais push les sources de kinetis, même si ça compile pas !

astralien3000, 2016-03-03:
> Donc si je comprend bien, on partirait sur un shield pour ADM2560 ?

Oui

> pour le module "régulateur à découpage", tu veux le fixer sur le shield ou ailleurs ?

Sur le shield

> pour les deux modules UART <> TTL, tu veux les fixer comment ? les scratcher ailleurs sur le robot ?
Ils doivent être dépluggables, donc les 2

Et c'est 3 modules UART <> TTL

> EDIT : pour l'UART, je suppose que tu veux communiquer avec la Rasp à travers l'Atmega16U2 de la board elle-même, pas avec un adaptateur USB <> UART quelconque ?

À travers l'USB de la board Arduino, pas besoin de sortir l'UART pour un adaptateur quelconque.

Elioty, 2016-03-03:
> Ok, on fait un échange de tâches ! XP
> Mais push les sources de kinetis, même si ça compile pas !

Nan, je vais les faire quand même, faut bien que je mettes la main là dedans un peu quand même.

Ok, un shield. Et l'ADM2560 étant déjà pas mal grande en surface, je vais essayer de limiter le shield aux même dimensions et superposer les trous de fixation, tu pourras donc traverser tout le sandwich pour bien le fixer :)
Question pas si conne que ça, en plus de la connectique donc tu as besoin, je remonte tous les pins autour de la carte comme ça tu pourras stacker un autre shield pas dessus ou pas ? (Quoi que ça risque être chaud avec le module "régulateur à découpage" monté dessus)

> Et c'est 3 modules UART <> TTL

Tu as dis "2 boards" UART <> TTL dans ton premier message donc je pensais que tu voulais le dernier UART juste comme ça, au naturel. Mais ok pas de problème. Et je n'ai pas bien compris où tu veux les fixer ensuite (sachant qu'ils n'ont pas de trous pour passer des vis). La question subsidiaire étant, dois-je leur laisser de la surface sur le pcb comme pour le régulateur à découpage ?

> À travers l'USB de la board Arduino, pas besoin de sortir l'UART pour un adaptateur quelconque.

Ok par l'USB.

astralien3000, 2016-03-04:
> Question pas si conne que ça, en plus de la connectique donc tu as besoin, je remonte tous les pins autour de la carte comme ça tu pourras stacker un autre shield pas dessus ou pas ? (Quoi que ça risque être chaud avec le module "régulateur à découpage" monté dessus)

Non merci, pas besoin ^^

> Tu as dis "2 boards" UART <> TTL dans ton premier message donc je pensais que tu voulais le dernier UART juste comme ça, au naturel. Mais ok pas de problème. Et je n'ai pas bien compris où tu veux [...]

En fait tu a raison, vaut mieux que le 3e UART reste généraliste (TX, RX, GND).
Par contre, il serait peut être intéressant de mettre une sortie puissance sur des connecteurs pourris (pas des XT60) en plus, au cas où.

Et si non, les boards UART <> TTL sont vendues avec des connecteurs (sécables) déjà montés, et un "dessous de carte" en plastique pour les protéger. Les connecteurs sont coudés, de sorte qu'il serait pas pratique de les plugger directement sur le shield. Et j'ai pas envie de foutre la protection en l'air juste pour les dessouder. Donc je les fixerai ailleurs.

Si non pour la fixation, le velcro c'est la vie ^^ !

Elioty, 2016-03-04:
Juste pour noter avant d'oublier et de faire les spécs au propre un peu plus tard :
- entrée puissance en XT60 avec fusible
- seulement le module "régulateur à découpage" à monter sur le shield, les modules UART <> TTL étant positionnés ailleurs dans le robot
- pluger uniquement les deux bandes latérales de l'ADM2560
- faire au moins 5 entrées logique avec anti-rebond, de préférence à router sur les pins supportant les interruptions
- l'ADM2560 sera alimentée par la Raspberry
- les GP2 sont les même que ceux sur Cérès

Dernières (je pense) questions :
- les servo plus classiques seront alimentés en 5V par le régulateur, devront les GP2, systèmes anti-rebond et l'éventuelle centrale inertielle être alimentés par le régul aussi ou par l'ADM2560 (elle même alimentée par la Rasp ; sachant que le régul sort max 5A et que tu prévois max 4 servo classiques) ?
- pour la centrale inertielle i2c, dois-je incorporer un fil +5V à son connecteur ou me limiter à l'i2c et la masse ?

astralien3000, 2016-03-04:
> les servo plus classiques seront alimentés en 5V par le régulateur, devront les GP2, systèmes anti-rebond et l'éventuelle centrale inertielle être alimentés par le régul aussi ou par l'ADM2560 (elle [...]

Par défaut, les GP2, centrale intertielle et antirebonds seront sur l'ADM2560.
Mais il y a pas un moyen de faire un petit système de jumpers pour arrêter des les alimenter par l'arduino et mettre une alim externe ? (C'est en option, sit tu le sent de le faire ^^)

> pour la centrale inertielle i2c, dois-je incorporer un fil +5V à son connecteur ou me limiter à l'i2c et la masse ?

Ajoute le +5V

Elioty, 2016-03-04:
> Mais il y a pas un moyen de faire un petit système de jumpers pour arrêter des les alimenter par l'arduino et mettre une alim externe ? (C'est en option, sit tu le sent de le faire ^^)

Si, ça doit pouvoir se faire sans problème ;)

Elioty, 2016-03-06:
### Spécifications

#### Documents

- [Datasheet de l'Atmega2560](http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf)
- [Schématique de l'ADM2560](https://www.arduino.cc/en/uploads/Main/arduino-mega2560_R3-sch.pdf)
- [Pin map de l'ADM2560](https://www.arduino.cc/en/Hacking/PinMapping2560)

#### Servos intelligents

Pour [les servos des 4 pattes de Buggybot](http://www.lextronic.fr/P36125-servo-avec-retours-dinformations.html), [deux modules UART <> TTL](http://www.lextronic.fr/P36132-module-de-conversion-seriettl.html) seront utilisés. Pour cela, l'UART 1 et 2 de l'Atmega2560 seront sollicités. Ils sont routés sur les pins 18 (TX), 19 (RX) et 16 (TX), 17 (RX) de la board. Les deux modules UART <> TTL seront fixés ailleurs dans le robot, il faut donc juste deux connecteurs 4 pins sur le shield (GND, Vin (puissance), RX, TX).

#### Servos classiques

Des servos plus classiques devront aussi pouvoir être utilisés avec le shield.
**pins à déterminer, voir question au message suivant**

#### Centrale inertielle

Une centrale inertielle I2C doit pouvoir être utilisée. Pour cela, le shield routera les pins 20 (SDA) et 21 (SCL) de l'ADM2560. Le connecteur fournira également du +5V (à 4 pins donc avec la masse).

#### GP2

Le shield fournira également de quoi connecter quatre GP2 ([les mêmes que pour Cérès](https://www.pololu.com/file/download/gp2y0a21yk0f.pdf?file_id=0J85)). Ils seront routés sur les pin A0 à A3 de la board (ADC0 à ADC3 du point de vue de l'Atmega2560).

#### Entrées digitales classiques

Cinq entrées digitales classiques (pour fins de course éventuelles et tirette) devront être disponibles. Elles seront équipées de système anti-rebond matériel comme sur Cérès. Elles seront routés sur les pins A6 à A10 de la boards, disposant de Pin Change Interrupt (PCINT 14 à 18).

#### Autres fonctionnalités

##### UART

Le dernier UART (le 3) de l'ADM2560 sera accessible sur le shield en tant que simple UART (connecteur trois broches TX, RX, GND).

##### Diodes de debug

Cinq diodes de debug seront disponibles sur le shield, routés sur les pins A11 à A15 de la board.

#### Communication système embarqué/Raspberry

La communication avec la Raspberry se fera classiquement par le connecteur USB de l'ADM2560, et donc de l'UART 0.

#### Alimentation

L'ADM2560 en elle même sera alimentée par la Raspberry à travers l'USB.

Le reste des composants 5V (GP2, systèmes anti-rebond, servos classiques, centrale inertielle) sera alimenté (par défaut) par un module "[régulateur à découpage](https://www.pololu.com/product/2851)" qui devra être fixé sur le shield.

À noter qu'un jumper trois états devra être disponible afin de pouvoir alimenter tous les composants logiques cités au paragraphe ci-dessus soit par le régulateur à découpage, soit par l'ADM2560, soit directement depuis une source +5V externe (avec un connecteur deux pins comme sur Hermès donc).

Pour la puissance, un connecteur XT60 sera utilisé, connecté à une Lipo 2S (~7.5V). Il alimentera les deux modules UART <> TTL et le régulateur à découpage.

#### Résumé

_À faire_

Elioty, 2016-03-06:
Voilà les specs, il reste le tableau récapitulatif avec toutes les correspondances "pin board", "pin mcu", "réf du module en interne" (genre OC3A) à faire.

Des remarques particulières ?

Pour les 4 PWM, je les ai toutes mises sur des timers 16 bits et j'ai évité le 0, pour que tu puisses bien le configurer comme tu veux pour le scheduler.

astralien3000, 2016-03-06:
Cool ! Pas de remarque importante.

Peut être, pourquoi 5 leds de debug ? et pas 4 ni 6 ? C'est pour compléter un port déjà entamé ?

Ah, je viens de voir sur les specs du servo intelligent que le courant max est de 1.5A. On va plus tabler sur 2A pour prendre large.

Elioty, 2016-03-06:
> Peut être, pourquoi 5 leds de debug ? et pas 4 ni 6 ? C'est pour compléter un port déjà entamé ?

Les pins A6 à A10 sont pris et j'ai mis les LED à la suite (de A11 à A15) et il ne reste que 5 pins sur la "lignée" là. Après, je peux t'en tirer d'autres d'ailleurs si besoin. C'est juste que là au moins, elles sont toutes à la suite et pas besoin de tirer une piste depuis l'autre bout de la carte :p

On y verra mieux une fois le résumé et l'infographie faits ;)

> Ah, je viens de voir sur les specs du servo intelligent que le courant max est de 1.5A. On va plus tabler sur 2A pour prendre large.

Ça me fait penser que j'ai justement oublié de parler du fusible puissance !! Je regarde ça demain.

Elioty, 2016-03-09:
Je reviens encore sur la consommation pour le dimensionnement du fusible.
Le régulateur par découpage tire max 3.5A sur notre 7.5V (il sort max 5A en 5V, soit 25W, soit 3.33A à 7.5V). Mais pour les servos, 2A max chacun, mais les douze pourront-ils vraiment tirer 2A en même temps ? Histoire d'essayer de dimensionner le fusible au mieux, afin de protéger au mieux (on ne manque pas forcément d'argent mais quand même). Une idée, ordre de grandeur ?

astralien3000, 2016-03-10:
Franchement, c'est difficile de prévoir quoi que ce soit.
C'est vrai que : 12*2A + 3.5A = 27.5A ça paraît beaucoup
Mais c'est uniquement en pique, pas très longtemps. 
Peut être qu'on peut faire confiance à la spec de moteurs et faire 12*1.5A+3.5A = 21.5A.
En sachant qu'il est presque impossible qu'ils tirent tous au max en même temps (sauf si quelqu'un s'asseoit sur le robot), on peut arondir à 20A.
Et on peut aussi considérer que un des 3 servo de la patte ne sera pas trop utilisé (un servo de l'épaule), on peut faire : 8*1.5A+4*0.25A+3.5A = 16.5A.

Donc, voila, on peut dire entre ~15-16A et ~20A pour le fusible ?
Si ça te paraît aberrant, dis moi.

Elioty, 2016-03-10:
Non, ça me parait bien, je pense qu'on va partir sur 20A alors. Mais je vais te mettre un "socket à fusible" sur ton shield comme sur Hermès donc on pourra simplement changer le fusible, dans le cas où il pète, mais aussi dans le cas où on veut changer la valeur.
Par contre pour du 20A, le fusible fera du 6.3mm x 32mm alors que sur Hermès il fait 5mm x 20mm. (On peut prendre un 5mm x 20mm en 20A aussi mais avec les filtres que j'ai mis sur Mouser, il n'y avait qu'une seule référence, à presque 10€ l'unité !!!...)

Elioty, 2016-03-12:
Une dernière chose, une idée de nom pour la carte ? :)

Elioty, 2016-03-12:
Je viens de compléter le résumé avec tableau (complet, le rêve du développeur, tu devrais avoir tout dessus) et infographie.

Sachant qu'on ne veut pas utiliser les 2*18 pins de l'extrémité de l'ADM2560, il reste à disposition les pins 2, 3, 4, 9 à 13, A4 et A5 (les 0 et 1 ne peuvent pas être utilisés comme cela, ils sont également routés vers le bridge USB). Mais je pense que tu n'as pas besoin de plus ? Si oui, c'est le moment de le dire avant de valider ces spécs :)

Et il reste à lui donner un petit nom aussi.

astralien3000, 2016-03-12:
ça, c'est du travail de pro !

> Mais je pense que tu n'as pas besoin de plus ? Si oui, c'est le moment de le dire avant de valider ces spécs :)

Je pense pas en avoir besoin, mais il serait eut être intéressant de laisser les pads quand même, même si on soude rien dessus. En parlant de ça, tu a les lib kicad pour l'arduino dans le dépôt de buggybot. Faudra surement les modifier, mais au moins tu n'aura pas à tout faire ?

> EDIT : La bonne question aussi est : est-ce que je route le pin reset sur un bouton pressoir ? (vu que le bouton de la board elle même ne sera plus accessible ; il suffit de tirer le pin à la masse pour faire le reset)

Je pense pas que ce soit utile, vu que l'arduino se reset automatiquement à chaque ouverture d'une connexion série via USB. C'est d'ailleurs un truc qui peut devenir dangereux si on fait pas gaffe (mais bon, c'est pas comme un robot à roues qui peut partir en toupie).

> Et il reste à lui donner un petit nom aussi.

J'y réfléchis, et je te dis ça rapidement (si possible). D'ailleurs si tu a des suggestions.

Petite question, au niveau mécanique, tu pense arriver à ne pas dépasser les limites en surface de l'Arduino. C'est pas grave que ce soit pas le cas, mais il faut que je fasse gaffe pour l'intégration.

astralien3000, 2016-03-13:
> Petite question, au niveau mécanique, tu pense arriver à ne pas dépasser les limites en surface de l'Arduino. C'est pas grave que ce soit pas le cas, mais il faut que je fasse gaffe pour l'intégration.

En fait, je viens de regarder un peu pour le placement des cartes (qui va être chiant !!). Tu peux dépasser en largeur sans trop de problème, mais pas en longueur. Et il vaut mieux pas utiliser l'espace en hauteur au niveau de la double rangée de pins.

Elioty, 2016-03-13:
> Je pense pas en avoir besoin, mais il serait eut être intéressant de laisser les pads quand même, même si on soude rien dessus. En parlant de ça, tu a les lib kicad pour l'arduino dans le dépôt de buggybot. Faudra surement les modifier, mais au moins tu n'aura pas à tout faire ?

Oui je pensais bien mettre des sécables mâles qui s'accouplent sur tous les femelles dessus, même si certains pins ne sont pas utilisés. On aura donc bien les pads avec, on pourra toujours y souder un fil si besoin. Je vais voir pour les lib ;)

> Je pense pas que ce soit utile, vu que l'arduino se reset automatiquement à chaque ouverture d'une connexion série via USB. C'est d'ailleurs un truc qui peut devenir dangereux si on fait pas gaffe (mais bon, c'est pas comme un robot à roues qui peut partir en toupie).

Ok, pas de reset sur le shield alors. Et à toi de voir pour ce comportement heureusement connu maintenant. Tu peux toujours reprogrammer l'Atmega16u2 si cela pose problème mais tu risques d'y laisser ta carte haha

> J'y réfléchis, et je te dis ça rapidement (si possible). D'ailleurs si tu a des suggestions.

Je n'ai pas fait de recherche encore mais je pensais à pourquoi pas une divinité/créature de la mythologie grecque ou d'une mythologie asiatique qui a plein de bras (je ne sais pas s'il existe une telle chose dans la mythologie grecque, je ne suis pas un as dans le domaine).

> Petite question, au niveau mécanique, tu pense arriver à ne pas dépasser les limites en surface de l'Arduino. C'est pas grave que ce soit pas le cas, mais il faut que je fasse gaffe pour l'intégration.

` `
> En fait, je viens de regarder un peu pour le placement des cartes (qui va être chiant !!). Tu peux dépasser en largeur sans trop de problème, mais pas en longueur. Et il vaut mieux pas utiliser l'espace en hauteur au niveau de la double rangée de pins.

Alors pour l'aspect méca, je vais essayer de me limiter à la largeur de la board, mais ok, je peux dépasser si besoin.
Côté longueur, c'est sûr que je ne vais pas dépasser le PCB de la board côté USB, et du côté du 2*18 pins, je vais arrêter le PCB juste avant, vu que, 1 on ne s'en sert pas, 2 tu pourras facilement y mettre des fils si besoin futur.

Elioty, 2016-03-13:
Et du coups, rien à rajouter sur ces spec ? Si oui, tu peux les valider en cochant la petite case correspondante :blush:

astralien3000, 2016-03-15:
Validé !

Elioty, 2016-04-16:
Je viens de finir schématique et empreintes, j'ai bien fait le tour au moins deux fois pour tout vérifier et je ferais le routage demain. Mais j'ai une dernière question qui m'est venu à l'esprit.

Tu peux me dire exactement comment tu branches les cartes UART <> TTL actuellement pour tes tests ?

PS : je viens de penser que je n'ai pas fini l'empreinte du régulateur en fait, me manque les deux pas de vis à mettre.

astralien3000, 2016-04-17:
Cool !

Et bien, j'utilise les pins pour l'UART1 et 2 de l'arduino mega.
Si non, pour la puissance, c'est direct branché sur la batterie lipo 2s.

Mais justement j'ai eu un léger problème avec la puissance. Je ne passe pas par la board uart/ttl, mais pas une board de test spécifiquement faite pour laisser passer de la puissance.
Après, cette board de test ne sera peut être plus nécessaire avec le shield.

astralien3000, 2016-05-10:
Tout avait été testé avant la coupe, et rien n'a cramé pendant.
Impec !

closed 2016-05-10 by astralien3000

Elioty, 2016-05-10:
Je note juste que la commande enable du régul ne fonctionne pas, certainement mon circuit qui déconne. Le régulateur reste tout le temps activé.

astralien3000, 2016-05-10:
Ha oui j'avais oublié.