xenomorphales/xenomorphales #51: Préparation de 2016-2017

astralien3000, 2016-08-16:
Salut à tout ceux qui suivent ce sujet.

Les "vacances" d'été arrivent dans leur dernière partie, et il faut penser à la suite. Il y a quelques gros changements dans la directions que je souhaite prendre. Et vous pouvez aussi me dire ce que vous voulez prendre comme direction pour cette année.

Alors, pour ma part : 
 - Je ne présenterai pas Buggybot cette année à la coupe.
 - Je ne veux plus travailler seul sur un robot
 - A savoir que j'ai faillis abandonné l'année passée, et que le plus gros argument qui me pousse à continuer pour le moment est le thème "Moon Village" de cette année. Pour un passionné de science fiction comme moi, ça se loupe pas ! =)
 - Je souhaite travailler sur un robot à roues, mais pas forcément en tant que "dev principal". Mais si il n'y a pas le choix, je prendrai les responsabilités qu'il faut.
 - On va surement instaurer une séparation des projets en prototypes pour les robots afin de laisser les personnes voulant travailler sur de petits systèmes pas trop complexe se faire plaisir.

Je met aussi mes objectifs relatifs à Aversive++ : 
 - Je considère l'année passée à travailler sur Aversive++ comme un semi échec...
 - J'ai déjà abandonné depuis longtemps le fait de recruter des gens pour bosser dessus. Voila, ça sert à rien mais je le dis quand même.
 - Je vais abandonner complètement l'interface HAL, et vais faire en sorte d'interfacer les devices avec des bibliothèques préexistantes (RIOT, XPCC, Arduino, et Jean-Passe).
 - Je vais abandonner les proof of concept du HDL, qui n'a pas vraiment de raison d'être sans le HAL maison.
 - Je vais fusionner la lib de cinématique inverse ikpy avec mon module de cinématique inverse. Le plus gros du travail est déjà fait, et ça va être cool !
 - En règle général, mon but sera de créer des ponts entre les différentes lib existantes pour l'embarqué, plus que de créer des choses à partir de zéro.

Voila, ce n'est qu'une partie des idées que j'ai eu dernièrement. J'aimerai vraiment savoir les envies pour l'année qui arrive des gens qui prendront part au projet, ou qui pensent prendre part au projet. Please !

Sokario, 2016-08-17:
Viens travailler sur mon robot, viens, viens il est bien ;-) 
RIP buggybot :-/

astralien3000, 2016-09-02:
Discussion qui s'est faite par mail finalement.

closed 2016-09-02 by astralien3000