xenomorphales/xenomorphales #58: API de l'asserv de silver

astralien3000, 2016-10-17:
Il faut définir les registres que l'application haut niveau poura utiliser pour gérer l'asserv.

Sokario, 2016-11-21:
Je suis d'accord mais les registres dépendent du design que je réalise lors de l'implémentation du bitstream sur la carte. Il faut donc que je termine tout les blocs de l'architectures avant de pouvoir te sortir les registres qui seront utilisé.

Par contre, je m'engage à ce que tu puisses à tout moment overider la boucle d'asservissement logic du FPGA. En soit en prenant le contrôle sur certaines étapes, tu auras toujours la possibilitée de récupérer les données des autres blocs qui sont indépendants.

astralien3000, 2016-11-22:
Même sans avoir les blocs déjà codés, on peut déjà prévoir les registres qui seront au minimum accessibles, non ? Au pire, il y aura plus de registres que prévu ^^

Sokario, 2016-11-22:
Pour le nombre de registre employés, c'est envisageable par contre je ne pourrait pas te dire exactement leur adresse.

astralien3000, 2016-11-22:
On ne s'occupe pas des adresses pour le moment, on s'occupe juste des noms des registres ^^

Elioty, 2016-11-22:
Et sinon l'idée final est toujours de faire un driver pour faire un device Linux (genre /dev/monFPGAdelamort) et puis de l'open (en rw) et de le mmap dans une application ? (avec certainement une bibliothèque applicative pour faire un API plus haut niveau)

astralien3000, 2016-11-23:
Pas de changement de ce côté là, à priori.
Mais après le fait d'avoir un linux est un plus, sans être une priorité. Si il n'y a pas de Linux, on programmera en mode bare-metal. La priorité est d'avoir l'asserv.

astralien3000, 2017-04-06:
La discussion n'est plus d'actualité, et est ouverte sur le [dépôt de silver](https://github.com/Sokario/Silver/issues/1).

closed 2017-04-06 by astralien3000