xenomorphales/xenomorphales #62: T-Shirt

astralien3000, 2016-10-17:
Voici un apreçu : 
https://www.spreadshirt.fr/personnaliser-soi-meme?product=201934875

Ça serai 18€ l'unité si on en commande plus de 6.

Elioty, 2016-10-17:
Classique, simple, sobre, très bien :p

astralien3000, 2016-10-18:
Le prix aussi est acceptable ?

Elioty, 2016-10-18:
J'ai payé des tshirt 15€ à Londres... Alors du sur-commande à 18€ en petit volume ne me parait pas exorbitant (même comparé aux tshirt à 10€ qu'on peut trouver en France).

Sokario, 2016-10-18:
Pas de "poste" sur le t-shirt?

astralien3000, 2016-10-18:
@Sokario : Sachant que les "postes" changent d'année en année et ne sont pas super bien définis, je pense pas que ce soit utile. En plus j'ai pas spécialement besoin qu'on sache mon poste, perso, l'appartenance à Xenomorphales me suffit ^^. Après si tout le monde veux le poste, ça peut se faire ! Il faut juste me le dire. La réduc a l'aire de marcher aussi pour des textes différents.

@Elioty : Je comptais en prendre 2, pour essayer de pas garder le même T-Shirt pendant 4 jours XD. Et puis, il y a moyen que si je demande à ma famille, quelques uns en voudront un aussi ^^ !

astralien3000, 2016-10-18:
D'ailleurs, la réduc a l'aire de marcher pour des produits différents ! On peut aussi prendre des sweat, ou des polo, par exemple, et avoir encore plus de réduc. Mais les sweat/polos sont plus chers, bien sûre !

Elioty, 2016-10-19:
Je n'avais que peu de doutes que tu en prendrais au moins deux :p

Ah beh à voir les prix mais peut-être tshirt et polo alors pour moi :) Puis je suppose que la réduc marche toujours sur des produits différents... tant que tu utilises le même masque d'impression. On c'est totalement décorrélé ?

darwikey, 2016-11-14:
Ma version :
https://www.spreadshirt.fr/personnaliser-soi-meme?product=145719495

darwikey, 2016-11-14:
Lien pour s'incrire :
https://docs.google.com/spreadsheets/d/1TUJ5O72AldJs2PED9vxAQsy1CHGUhorkqEucO2OmTLA/edit?usp=sharing

closed 2017-03-05 by darwikey