xenomorphales/xenomorphales #63: Électronique Céres (robot principal, avec bras)

astralien3000, 2017-04-06:
Comme on a une 2e Céres finalement avec une élec à l'arrache, on va essayer de faire quelque chose pour pas tout cramer pas inattention (comme hier soir)

Donc voila les éléments électroniques : 
 - une board de dev sous MCU ARM (3V3 de tension logique) : [SAMR21](http://www.atmel.com/tools/atsamr21-xpro.aspx)/Nucleo-*
 - 5 ou 10 (ou plus, si je trouve d'autres choses à faire) [SCS15](http://www.feetechrc.com/product/digital-servo/feetech-scservo-multi-purpose-uart-bus-robot-servo-for-arduino-diy/), mais (contrairement à buggybot) seulement 2 ou 3 seront actifs en même temps.
 - 2 ou 4 (ou +) [XL320](http://support.robotis.com/en/product/actuator/dynamixel_x/xl_series/xl-320.htm)
 - un [TTL-Linker mini](http://www.lextronic.fr/P36132-module-de-conversion-sriettl.html) pour communiquer avec les SCS15 et les XL320
 - un [Level shifter/converter (je n'ai pas la ref exacte, mais c'est le plus ressemblant)](https://www.amazon.com/XCSOURCE-Converter-Bi-Directional-Module-TE291/dp/B0148BLZGE) pour la communication UART entre la board ARM et le TTL Linker mini.
 - un [gyro (je l'ai pas acheté là, mais c'est le plus ressemblant)](http://robokits.co.in/sensors/gyroscope/l3g4200d-3axis-gyro).
 - Une [carte L298N (toujours pas la bonne ref)](https://www.amazon.com/Leegoal-Stepper-Driver-Controller-Module/dp/B008BSGA7K), en gros (1 PWM sur EN, 2 GPIO sur la direction) * 2 canaux.
 - Deux LiPo 2S à brancher en série. Attention, j'ai besoin de 7.4V pour les servos.
 - Une tirette
 - Un arret d'urgence
 - 4 capteurs de distance ultrason ( = 2 gpio digitaux par capteur)
 - 4 GP2 ( = 1 pin ADC par capteur)
 - Une RASP ?
 - Potentiellement un carte de communication wireless (en gros : un UART)

Voila, j'espère ne rien oublier.
C'est vrai que mon matos commence à dater, impossible de retrouver les refs...

Elioty, 2017-04-10:
> - une board de dev sous MCU ARM (3V3 de tension logique) : SAMR21/Nucleo-*

Comme on a dit, pas sûr que ce soit la carte finale, donc je ne vais pas faire un shield ou un truc trop intégré mais juste un gros connecteur.

> - 5 ou 10 (ou plus, si je trouve d'autres choses à faire) SCS15, mais (contrairement à buggybot) seulement 2 ou 3 seront actifs en même temps.
> - 2 ou 4 (ou +) XL320

C'est quoi le type de connecteurs pour eux et combien la carte devraient en avoir (vu que tu peux les mettre en série, les SCS15 tout du moins, je ne sais pas pour les XL320) ?

> - un TTL-Linker mini pour communiquer avec les SCS15 et les XL320

Un seul pour adresser tous les servo ? Je pense que je referais venir le bus TTL sur la carte pour pouvoir sortir la puissance sans passer par la petite carte.
En tout cas, cette carte n'est pas stackable.

> - un Level shifter/converter (je n'ai pas la ref exacte, mais c'est le plus ressemblant) pour la communication UART entre la board ARM et le TTL Linker mini.

On le stack sur la carte ou tu le mettra ailleurs dans le robot ? Et si on le stack, directement soudé ou on met un "socket" avec des sécables femelles ? (comme la Teensy sur Hermès)

> - un gyro (je l'ai pas acheté là, mais c'est le plus ressemblant).

Comme on a dit, il doit être placé à un endroit précis dans le robot donc pas de question pour le stacker, on met juste un connecteur.

> - Une carte L298N (toujours pas la bonne ref), en gros (1 PWM sur EN, 2 GPIO sur la direction) * 2 canaux.

On peut la stacker avec des vis aussi. Mais dans tous les cas faut tirer des fils, peut pas mettre des sécables pour celui là. Donc stackée ou placée ailleurs dans le robot ?
Autre question, c'est pour driver la base roulante ?

> - Deux LiPo 2S à brancher en série. Attention, j'ai besoin de 7.4V pour les servos.

Ouais donc on peux acheter le Y (ou le faire) qui met les deux LiPo en série comme sur le gros Cérès avec du XT60 pour brancher la batterie et les moteurs de la base roulante (si la carte L298N est bien là pour les driver).

> - Une tirette

Juste un GPIO en input avec anti-rebond comme on a déjà fait.

> - Un arret d'urgence

En mode adhoc directement sur le câble comme sur le gros Cérès ?

> - 4 capteurs de distance ultrason ( = 2 gpio digitaux par capteur)

8 GPIO sans anti-rebond ici. Je suppose qu'il faut un connecteur 4 pôles par capteur ? (Vin (5 ou 3.3V ?), GND, Input, Output)

> - 4 GP2 ( = 1 pin ADC par capteur)

Comme pour la tirette, on a déjà fait.

> - Une RASP ?

Hmm, la com RASP <> carte embarquée, tu gères ça directement du côté de la carte embarqué donc normalement pas d'impact sur la carte à réaliser.

> - Potentiellement un carte de communication wireless (en gros : un UART)

Ok, on peut sortir un UART sans problème.

astralien3000, 2017-04-12:
>> - 5 ou 10 (ou plus, si je trouve d'autres choses à faire) SCS15, mais (contrairement à buggybot) seulement 2 ou 3 seront actifs en même temps.
>> - 2 ou 4 (ou +) XL320

> C'est quoi le type de connecteurs pour eux et combien la carte devraient en avoir (vu que tu peux les mettre en série, les SCS15 tout du moins, je ne sais pas pour les XL320) ?

Il semble que ce soit des molex (molex n'est pas donné dans les spec du servo, donc c'est peut être pas obligatoire) 5264 3P (3 Positions). Les mâles ressemblent en tout cas.
Il faut au minimum 2 connecteurs femelles (1 pour chaque bras). Et on va sûrement en mettre 4 au cas où (funny action, autre...)

> Un seul pour adresser tous les servo ? Je pense que je referais venir le bus TTL sur la carte pour pouvoir sortir la puissance sans passer par la petite carte.
En tout cas, cette carte n'est pas stackable.
 
👍 

>> -  un Level shifter/converter (je n'ai pas la ref exacte, mais c'est le plus ressemblant) pour la communication UART entre la board ARM et le TTL Linker mini.

> On le stack sur la carte ou tu le mettra ailleurs dans le robot ? Et si on le stack, directement soudé ou on met un "socket" avec des sécables femelles ? (comme la Teensy sur Hermès)

Dans le meilleur des cas, vaudrait mieux le stacker. Après qu'on mette un sécable (femelles) ou pas, ça change rien. mais par défaut j'en mettrai un.

>> Une carte L298N (toujours pas la bonne ref), en gros (1 PWM sur EN, 2 GPIO sur la direction) * 2 canaux.

> On peut la stacker avec des vis aussi. Mais dans tous les cas faut tirer des fils, peut pas mettre des sécables pour celui là. Donc stackée ou placée ailleurs dans le robot ?
Autre question, c'est pour driver la base roulante ?

C'est pour la base roulante, et elle sera ailleurs dans le robot.

> Ouais donc on peux acheter le Y (ou le faire) qui met les deux LiPo en série comme sur le gros Cérès avec du XT60 pour brancher la batterie et les moteurs de la base roulante (si la carte L298N est bien là pour les driver).

Acheter les Y ? Je préferrai que les batteries soient individuellement branchées sur la board, ainsi que l'AU. En effet, il faut aussi que la batteries dont le "-" est à la masse soit branchée au fil puissance des servos. On peut le faire en dehors de la carte, mais ça va faire dégueu.
Peut être faut il donner un choix en permettant de brancher 3 batteries ? (2 pour les moteurs, et 1 pour les servos)... Et on aurai  quand même une config pour 2 batteries.

>> - 4 capteurs de distance ultrason ( = 2 gpio digitaux par capteur)

> 8 GPIO sans anti-rebond ici. Je suppose qu'il faut un connecteur 4 pôles par capteur ? (Vin (5 ou 3.3V ?), GND, Input, Output)

http://www.gotronic.fr/art-module-de-detection-us-hc-sr04-20912.htm
5V il semble.
Par contre leur histoire de plantage ma fait un peu peur...

astralien3000, 2017-04-12:
Petite précision pour le gyro, il faut tirer les pins : 
 - GND
 - Vcc (3.3V)
 - SCL
 - SDA
 - DR
 - INT

Les 2 dernière sont peut être pas obligatoire mais le driver que j'utilise a l'aire de les utiliser, donc je préfère pas prendre de risque.

astralien3000, 2017-04-18:
Level shifter pin mapping : 
```
5V0 GND LV1 LV2 LV3 LV4

3V3 GND HV1 HV2 HV3 HV4
```

Elioty, 2017-04-18:
Alors la carte aura :
* Un connecteur pour brancher la carte embedded avec :
  * +5V venant de l'embed
  * +3.3V venant de l'embed
  * GND
  * servos (en passant par le level shifter) :
    * UART RX
    * UART TX
  * gyro :
    * SCL
    * SDA
    * DR
    * INT
  * moteur (x2) :
    * PWM
    * DIR x2
  * pin tirette
  * capteur ultrason (x4) :
    * Trigger
    * Echo
  * GP2 (x4) :
    * ADC
  * UART supplémentaire (sans passer par le level shifter) :
    * RX
    * TX
* Un socket pour le level shifter :
  * d'un côté :
    * +5V en direction du level shifter
    * GND
    * IN/OUT x4
  * de l'autre côté :
    * +3.3V  en direction du level shifter
    * GND
    * IN/OUT x4
* Un connecteur pour le TTL-Linker :
  * RX
  * TX
* Un connecteur Molex vers le TTL-Linker :
  * S
  * +7.4V en direction du TTL-Linker
  * GND
* Des connecteurs pour brancher les servos x4 :
  * S
  * +7.4V
  * GND
* Un connecteur pour le gyro :
  * GND
  * SDA
  * SCL
  * +3.3V
  * DR
  * INT
* Un connecteur XT-60 pour fournir la puissance au driver moteur :
  * +7.4V en direction du driver moteur
  * GND
* Un connecteur XT-60 LiPo 2S :
  * +7.4V venant de la batterie
  * GND
* Un connecteur XT-60 pour l’arrêt d'urgence
* Un connecteur pour driver les moteurs :
  * +5V en direction du driver moteur
  * GND
  * Par moteurs x2 :
    * EN
    * DIR x2
* Des connecteurs pour les capteurs ultrasons x4 :
  * +5V en direction des capteurs
  * Trigger
  * Echo
  * GND
* Des connecteurs pour les GP2 x4 :
  * +5V en direction des GP2
  * GND
  * ADC (avec diviseur de tension derrière pour ramener sur l'intervalle [0 ; 3.3V])
* Un connecteur pour la tirette (en +3.3V), deux pôles, avec anti-rebond derrière
* Un connecteur UART supplémentaire :
  * RX
  * TX
  * GND

Elioty, 2017-04-18:
Voila @astralien3000. Est-ce que j'ai oublié quelque chose ou, plus grave, est-ce que j'ai foiré quelque chose dans ce listing de tout ce qu'aura la carte ?

Elioty, 2017-04-20:
@astralien3000 pas de remarque sur mon récapitulatif, dois-je le considérer all green ?

astralien3000, 2017-04-20:
Désolé, oui tout est bon, pas de fusible par contre.

Elioty, 2017-04-25:
Avancement schématique :
- [ ] Un connecteur pour brancher la carte embedded
- [ ] Un socket pour le level shifter
- [ ] Un connecteur pour le TTL-Linker
- [ ] Un connecteur Molex vers le TTL-Linker
- [ ] Des connecteurs pour brancher les servos x4
- [ ] Un connecteur pour le gyro
- [x] Un connecteur XT-60 pour la LiPo 2S
- [x] Un connecteur XT-60 pour l’arrêt d'urgence
- [x] Un connecteur XT-60 pour fournir la puissance au driver moteur
- [x] Un connecteur pour driver les moteurs
- [x] Des connecteurs pour les capteurs ultrasons x4
- [x] Des connecteurs pour les GP2 x4
- [x] Un connecteur pour la tirette (en +3.3V), deux pôles, avec anti-rebond derrière
- [x] Un connecteur UART supplémentaire

Elioty, 2017-05-02:
Je viens de finir la schématique. Reste à faire la liste des composants.
@astralien3000 si je fais un .md comme pour Hermès (sans lien vers un quelconque marchand par contre, tu trouveras celui qui te va le mieux), ça te convient ?

astralien3000, 2017-05-02:
Ça me va ^^

Elioty, 2017-05-03:
J'ai poussé le Components.md, ça te va @astralien3000 ? Je dois te sortir les schématiques en pdf ou tu les visualiseras directement sur KiCad pour la soudure et le montage ?

astralien3000, 2017-05-03:
Merci !!
Si ça te prends pas beaucoup plus de temps je veux bien le PDF, si non j'utiliserai Kicad.

Elioty, 2017-05-03:
@astralien3000 envoyée par email 😉

Elioty, 2017-05-04:
Bon du coups, on pourra fermer l'issue une fois que tu auras soudé tout ça. Attention, j'ai mis une capa électrolytique pour la puissance (celle de 470µF) et c'est polarisée. Ne te trompe pas de sens si tu ne veux pas qu'elle t'explose à la figure lorsque tu brancheras le jus 😋

astralien3000, 2017-05-04:
Ouai j'ai vu. C'est en partie pour ça que j'en ai pris de rechange (et puis j'avais pas trop le choix en fait). ^^

Elioty, 2017-05-04:
Au pire, si tu ne sais pas, tu les soudes toutes en alternant à chaque fois de sens ^-^' blague à part, avec cette 470µF de base, tu devrais avoir une bonne tension bien fixe lorsque tu tireras sur tes servos. Je pense qu'ils ont quelques capa de comprises à l'intérieur mais il en faut mieux trop que pas assez.

Autre point, pour les petites capa céramiques (pas de sens pour elles, sauvé haha), elles servent surtout à filtrer les retours et toutes ces merdes là donc faut essayer de les mettre au plus proche de la source, donc dans notre cas, en général des connecteurs :wink: 

Je suppose que tu as déjà commandé alors. Pour les connecteurs, tu as pris du sécable classique ou des trucs avec des détrompeurs comme sur Hermès ? Et le gros 2x15 ne te dérange pas ? haha
Pour le gyro, j'ai mis un 2x4 aussi même si on ne sert que de 6 pins pour l'instant, tu pourras donc câbler les deux autres IO de ton gyro si jamais tu veux t'en servir plus tard.

astralien3000, 2017-05-04:
Je les ai même déjà reçus ! Ils sont rapides les gars.
J'ai tout pris en sécable, les détrompeurs c'est pour les faibles.
Le connecteur 2x15 est toujours mieux que 2x15 cables séparés ^^

Elioty, 2017-05-05:
Wow, ça c'est de la livraison expresse ! Bon week-end soudure alors :yum:
Haha ok, n'hésite pas à mettre des petites pastilles de couleur (genre une goute de blanco puis la colorier au stylo ?) quand même pour pas te tromper de connecteur et pour t'indiquer le sens.
C'est sûr, vaut mieux UN connecteur 2x15 que TRENTE fils séparés 😆

astralien3000, 2017-05-11:
Juste pour information, j'ai pris quelques libertés avec le schématique.
 - J'ai pa pu faire les capteurs ultrason (ma carte de contrôle a pas assez de pins)
 - Tout les connecteurs 3 pins ont leur VCC au milieu et GND sur le côté (vu que je n'ai pas de détrompeur, c'est plus sécurisant comme ça).

astralien3000, 2017-05-17:
Fini et testé !

closed 2017-05-17 by astralien3000