xenomorphales/xenomorphales #65: Base roulante Céres 3.0

astralien3000, 2017-05-31:
http://www.francerobotique.com/moteurs-classiques/567-moteur-m%C3%A9tal-plan%C3%A9taire-premium-hd-12v-612rpm-avec-encodeur.html

Elioty, 2017-05-31:
Ah ouais, ce n'est pas le même prix que les moteurs de la base roulante de Cérès 2.0 quand même >_< mais ils envoient du lourd quoi !!!

astralien3000, 2017-05-31:
Surtout que je vais peut être ajouter un rapport d'engrenage qui va multiplier la vitesse par 2 (vu qu'on est large niveau couple) ! XD

astralien3000, 2017-07-07:
Pour les archives, après avoir rapidement vérifié la faisabilité mécanique de la base roulante, j'ai acheté le materiel suivant : 
 - [nouveaux moteurs](http://www.francerobotique.com/moteurs-classiques/567-moteur-m%C3%A9tal-plan%C3%A9taire-premium-hd-12v-612rpm-avec-encodeur.html)
 - [encodeurs (les même que Céres 1 et 2.0)](http://www.robotshop.com/eu/fr/encodeur-rotatif-6mm-1024-p-r.html)
 - [poulie D20mm](http://www.conrad.fr/ce/fr/product/209513/Poulie-dentee-Reely-aluminium-percage-4-mm-Diametre-20-mm-Nombre-de-dents-20/SHOP_AREA_32457)
 - [poulie D38mm](http://www.conrad.fr/ce/fr/product/209516/Poulie-dentee-Reely-aluminium-percage-6-mm-Diametre-38-mm-Nombre-de-dents-40/SHOP_AREA_32457)
 - [courroie qui va bien (j'espère)](http://www.conrad.fr/ce/fr/product/209519/MODELCRAFT-Courroie-dentee-160-mm-64/SHOP_AREA_32457)

Je vais vampiriser le reste de Céres 1.1.

astralien3000, 2017-12-21:
Done !
Un méca au poil !

closed 2017-12-21 by astralien3000