xenomorphales/xenomorphales #66: Hemes 2

astralien3000, 2017-05-31:


Elioty, 2017-05-31:
Du coups, quels besoins d'IO ? Déjà une idée pour le microcontrôleur ? Dimension de la carte et position des pas de vis ? Quels genres de connecteurs ?

astralien3000, 2017-05-31:
> quels besoins d'IO ?

Déjà, j'aimerai entamer une réflexion sur l'architecture électronique générale du robot, pas que sur la carte Hermes. Je sais pas si ça vaut le coup de le mettre sur une autre issue.

> Déjà une idée pour le microcontrôleur ?

Non

> Dimension de la carte et position des pas de vis ?

Je fais la méca avant ^^, mais vu les permiers schémas que j'ai dessiné, on devrait avoir une carte de largeur entre 3 et 4 cm de large pour une longueur > 10 cm.

> Quels genres de connecteurs ?

XT60 pour la puissance (ils reste bien).
Il faudra peut être une compatibilité avec le bus TTL des feetech.
Les mêmes embases avec détrompteur que Hermes 1 si non ?

Elioty, 2017-06-01:
> Déjà, j'aimerai entamer une réflexion sur l'architecture électronique générale du robot, pas que sur la carte Hermes. Je sais pas si ça vaut le coup de le mettre sur une autre issue.

On peut garder une seule issue et donc renommer celle-ci avec un nom plus englobant ?!

> Je fais la méca avant ^^, mais vu les permiers schémas que j'ai dessiné, on devrait avoir une carte de largeur entre 3 et 4 cm de large pour une longueur > 10 cm.

Je lançais un peu en l'air tous les sujets qui me pensaient par la tête et c'est vrai que de toute manière, l'aspect mécanique de la carte ne sera important pour moi que lorsque je passerai au routage, donc pas pour tout de suite encore. Mais sinon longueur > 10 cm, ça devrait être pas mal pour poser tout plein de connecteur sur les deux côtés, mais je suppose que tu veux une carte embarquant son proper microcontrôleur, pas un shield alors ? Ou alors un shield pour une carte mince genre Teensy comme avec Hermes 1.

> XT60 pour la puissance (ils reste bien).
> Il faudra peut être une compatibilité avec le bus TTL des feetech.
> Les mêmes embases avec détrompteur que Hermes 1 si non ?

Oui, les XT60 sont satisfaisants pour la puissance donc autant rester avec eux 😃 
Pour les connecteurs comme sur tes câbles des feetech, pour le connecteur ce n'est pas un soucis mais faudra voir comment le connecteur femelle se monte. Sinon les embases avec détrompeur comme sur Hermes 1 sont relativement simples à monter et ont "l'avantage" d'être compatibles avec les bêtes sécables (dans les deux sens, mâle de l'un avec femelle de l'autre et inversement) au cas où on n'ait pas de stock à la coupe, on peut peut-être trouver une équipes qui nous passerait des bouts de sécables (mais si on en arrive là, on est vraiment dans la mouise de toute façon). 😆

kazord, 2017-06-01:
La partie mécanique me plaît bien, j pourrai participé (sachant que je suis sur bordeaux a partir de juillet pour la coordination)

Côté câblage modulaire, l idée est pas mal, mais faut un connecteur standard qui a suffisamment de pin pour tout les étages inférieur,  e donc savoir combien de port vont du contrôleur vers les shield inferieur

astralien3000, 2017-06-01:
@Elioty ouai, on va garder cette issue pour le développement spécifique d'Hermes (il y a déjà du boulot pour les prochains pont en H, [le lien de ma trouvaille, en passant](https://www.pololu.com/product/128)). Mais pour l'instant, c'est en pause, jusqu'à ce qu'on soit d'accord sur l'archi élec générale.

@kazord Le problème de la mécanique, c'est que c'est moi le responsable et je suis à Paris... C'est pas pratique. Aussi, faut mieux poster sur #67 pour parler de ça maintenant.

astralien3000, 2017-07-10:
Désolé pour le temps d'attente.

# Besoins initiaux

## Périphériques

Comme on avait discuté de mettre 2 Hermes dans le robot, je donne la config minimale d'une Hermes, et j'ajoute des éléments pour la sûreté si on veut pouvoir utiliser Hermes comme board principale.

### Minimum vital

 - 2 [encodeurs](http://www.robotshop.com/eu/fr/encodeur-rotatif-6mm-1024-p-r.html)
 - 2 [moteurs](http://www.francerobotique.com/moteurs-classiques/567-moteur-m%C3%A9tal-plan%C3%A9taire-premium-hd-12v-612rpm-avec-encodeur.html)
 - 2 fin de course
 - 3 GP2
 - BUS de comm (#67)
 - Alim en 14V (pas de batterie USB) 

### Extension sécurité

 - 1 à 3 GP2 de plus (donc 4 à 6 en tout)
 - 2 fin de course de plus (4 en tout)
 - Sortie UART ?
 - Tirette ?

## Méca

J'essaierai de te trouver une surface de 8x8 cm, mais il est possible que tu n'ai pas besoin d'autant.
Le moins de place en "hauteur" (hauteur des composants), le mieux c'est, la max étant de 2cm.
Pour l'emplacement des vis, c'est relativement libre, mais faire une disposition carrée ou rectangulaire c'est mieux. Le support de la board sera imprimé en 3D.

Elioty, 2017-09-27:
Ça y est, c'est la rentrée (de la robotique) et je me plonge enfin sérieusement sur l'affaire. Et j'arrive avec quelques petites questions/remarques.

### 1. Moteurs et Ponts en H

En lisant la [datasheet du moteur que tu souhaites utiliser](https://www.francerobotique.com/index.php?controller=attachment&id_attachment=153), le 20A (qui m'avait fait peur !!), c'est le "max stall current", c'est-à-dire le courant maximum (à 12V, tension nominale du moteur, même si on les fera fonctionner avec 4 cellules de LiPo (sans régulateur) donc à ~14.8V) lorsque le moteur est totalement bloqué. Le "rated-load current", courant typique attendu lorsque le moteur déplace une charge (à 12V encore une fois) n'est que de 2A, c'est beaucoup plus acceptable !!

Je pense donc qu'il serait possible d'utiliser des LMD18200/LMD18201 pour driver ces moteurs, avec un fusible de 8A à l'arriver de l'alimentation sur la board (sachant qu'il y aura aussi un régulateur derrière ce fusible pour alimenter la logique) et tout le découplage nécessaire. Pour étayer mon choix, je rappelle que ces ponts en H supportent 3A en continu et jusqu'à 6A en pic (durant max 200ms) (datasheet page 3). De plus, ces ponts en H disposent d'un circuit de protection contre les pic de courant dépassant les ~10A (datasheet page 9) qui arrête complétement le pont en H lorsque cela ce produit (exemples donnés dans la datasheet, court-circuit ou rotor bloqué, exactement le cas de notre "max stall current") puis tente périodiquement de rétablir le fonctionnement (et l’interrompt immédiatement si le courant dépasse encore les ~10A). D'ailleurs, on pourrait peut-être se passer du fusible pour la puissance du coups. Enfin, ces ponts en H ont aussi une protection thermique, levant un flag en output à partir de 145°C (permettant à notre micro-contrôleur de prendre une décision) et arrêtant encore une fois le pont en H de fonctionner au dessus de 170°C (chauffe qui peut arriver si le pont en H est très sollicité avec éventuellement des pics réguliers entre 3 et 6A) (datasheet page 7). Enfin, d'après la datasheet page 16, le pont en H soudé sur la board fait 20.57mm +- 0.25 (si tu veux vérifier dans la datasheet, j'ai fait la somme des mesures 17.78mm +- 0.25 et 2.79mm). Donc pour la hauteur totale de la board, avec l'épaisseur du PCB plus les soudures et CMS en bottom, ça dépasse la contrainte des 2cm, on peut cependant coucher ces ponts en H là donc on devrait s'en sortir.

- [datasheet LMD18201](http://www.ti.com/lit/ds/symlink/lmd18201.pdf)

Est-ce que tu valides ce choix ? Et avec ou sans fusible sur la puissance ? (même si je pense que tu préféreras avec, et je pense que ça serait mieux aussi).

Puis en fait, pas d'autres sujets pour l'instant. Le reste du besoin (enco, fin de course, GP2, CAN, LED, UART, etc) a déjà été traité soit dans #67 ou #14, reste à spécifier le pin mapping, que j'ai commencé, et à rédiger une spécification propre pour le reste (choix des composants, dont le LMD1820x cité plus haut fera parti ou non selon ta réponse).

astralien3000, 2017-09-28:
Hmm...
- LMD1820x
  - (+) On a de l'expérience avec
  - (+) C'est assez sécurisé
  - (+) Prends moins de surface sur le PCB
  - (-) Assez "haut"
  - (-) Seulement 6A en pique
  - (-) C'est cher
- VNH2SP30
  - (+) Moitié du prix
  - (+) Plus petit en hauteur
  - (+) Théoriquement 30A en continu (14A sans danger selon pololu)
  - (-) Prends beaucoup de surface sur le PCB
  - (-) On a pas d'expérience avec
  - (-) "Ça supporte 30A en continu, mais ça peut surchauffer avant"

Pas super évident...
Je me dit aussi que si le moteur demande 20A, la table (ou la méca) va peut être morfler.
En regardant la datasheet du [VHN](https://www.pololu.com/file/0J52/vnh2sp30.pdf), ça a l'aire de donner les mêmes sécurités, supporter des ampérages plus élevés, et c'est moins cher.
Je viens de voir ton commentaire sur le fait que ce soit chaud à souder tel-quel, on peut l'utiliser avec la board, en effet.

Il semble quand même que le VNH soit plus intéressant, mais après la question est : est-ce que tu a envie de l'utiliser ? Peut on avoir confiance en les données qu'on a ? (généralement pololu est assez sérieux comme vendeur, quand même)

Elioty, 2017-09-30:
Alors si on prend le VNH2SP30 :
- (~) pour la partie soudure, on devrait déjà avoir un PHY CAN ([comme celui là par exemple](http://www.ti.com/lit/ds/symlink/sn65hvd230.pdf)) qui a ses pattes espacées de juste 1.27mm (juste 1 pour le VNH2SP30) mais je ferais ça à Eirbot donc au pire, on trouvera un âme charitable.
- (-) pour le 14A continu sans surchauffe/danger, si tu lis le petit 3, ça dit en utilisant les cartes toutes prêtes de Pololu et ils ont soudé les _heat slugs_ qui se trouvent *sous* la puce, or si on soude la puce à la main, on ne pourra pas les souder ceux-là.
- (-) pour les IO, le VNH2SP30 demande 3 pins pour le contrôler, deux pour le sens et un pour la PWM (conte deux pour le LMD18201), et bien qu'il ait des protections thermiques, je n'ai pas vu de thermal flag qui permettrait d'informer le µC que ça chauffe.
- (+) pour la mesure de la consommation, il supporte du current sensing comme le LMD18201, mais si on veut l'utiliser, c'est un ADC de plus à utiliser.

Enfin bref, je pense que si on veut utiliser celui-là, il vaut mieux prendre la board toute prête. Question sous-jacente donc, question méca, ça se placerait où, de la place de dispo ?

astralien3000, 2017-10-01:
Aaaaah ! Pourquoi j'ai pas vu ta réponse avant >< les notification de github sont assez random...

> (-) pour le 14A continu sans surchauffe/danger, si tu lis le petit 3, ça dit en utilisant les cartes toutes prêtes de Pololu et ils ont soudé les heat slugs qui se trouvent sous la puce, or si on soude la puce à la main, on ne pourra pas les souder ceux-là.

J'ai pas trouvé le mot "slugs" dans la page du VNH2 ou celle de [la board](https://www.pololu.com/product/706). J'ai juste vu qu'ils avaient reliés les OUTA aux trous de vis de montage pour dissiper mieux la chaleur. Par contre dans le petit 3, si j'ai bien compris, ils distent qu'ils avaient pas utilisés de ventilos ou de radiateur, ce qui est un bon point. Et ils ont dit que la board était capable de tenir les 20A dans une durée de l'ordre de la seconde, ce qui devrait être suffisant pour gérer nos piques de courant ?

> (-) pour les IO, le VNH2SP30 demande 3 pins pour le contrôler, deux pour le sens et un pour la PWM (conte deux pour le LMD18201), et bien qu'il ait des protections thermiques, je n'ai pas vu de thermal flag qui permettrait d'informer le µC que ça chauffe.

3 pins, exactement comme les L298N que j'ai utilisé cette année sur céres 1.1 ^^. Bon j'avoue ce choix est un peu nul, mais bon, on peut essayer de faire une logique avec quelques transistors pour n'utiliser que 2 pins, si on veut économiser les pins de la teensy.

Pour le thermal flag, je crois comprendre que c'est les pins DIAGA/ENA et DIAGB/ENB (qui permettent aussi de désactiver les sorties). J'ai pas compris pourquoi il y en a 2 mais bon, la datasheet donne une table de vérité pour ça.

> Enfin bref, je pense que si on veut utiliser celui-là, il vaut mieux prendre la board toute prête. Question sous-jacente donc, question méca, ça se placerait où, de la place de dispo ?

👍  Je pensais juste faire comme le régulateur de tension pour Hydra : la board est soudée à Hermes comme un composant traversant. Ca prends de la surface, mais on peut peut être s'arranger pour mettre des composants CMS en dessous ?

Elioty, 2017-10-01:
> Aaaaah ! Pourquoi j'ai pas vu ta réponse avant >< les notification de github sont assez random...

Jamais eu de problème avec les notifications sur github, bizarre.

> J'ai pas trouvé le mot "slugs" dans la page du VNH2 ou celle de la board. J'ai juste vu qu'ils avaient reliés les OUTA aux trous de vis de montage pour dissiper mieux la chaleur. Par contre dans le [...]

Sur le schématique de la board, on voit bien les trois _heat slugs_ abrégés en heatslg. Sinon c'est pas dangereux ça "d’électriser" les pas de vis ?! Sinon, oui pour le petit 3, c'est juste la board seule sans radiateur ni ventilateur. Pour le dernier point : 
> Time to overheat at 20 A: 35 s
C'est largement suffisant. Les moteurs tirent nominalement 2A avec une charge, le (jusqu'à) 20A, c'est lorsqu'ils bloquent. Et à mon humble avis, la peinture de la table et nos roues n’adhéreront pas assez pour bloquer les moteurs lorsque le robot sera contre un mur 😆 c'est plutôt nos roues et la table qui prendront cher !!!

> 3 pins, exactement comme les L298N que j'ai utilisé cette année sur céres 1.1 ^^. Bon j'avoue ce choix est un peu nul, mais bon, on peut essayer de faire une logique avec quelques transistors pour n'utiliser que 2 pins, si on veut économiser les pins de la teensy.

Je vais éviter les transistors, ça n'a pas bien marché sur Hydra pour contrôler le régulateur 🤐 je m'en sortirai avec 3 pins, ça devrait le faire 😉 

> Pour le thermal flag, je crois comprendre que c'est les pins DIAGA/ENA et DIAGB/ENB (qui permettent aussi de désactiver les sorties). J'ai pas compris pourquoi il y en a 2 mais bon, la datasheet donne une table de vérité pour ça.

C'est aussi la partie que je n'ai pas super bien compris de la datasheet. Mais de ce que je comprends c'est que DIAGA et/ou DIAGB sont tirés (dans la puce) à la masse lorsqu'il y un problème (survoltage, surampérage, surchauffe) mais ça ne précise pas lequel de ces problèmes. Et il y en a deux car en cas de surampérage, à cause d'un court circuit, ça t’indique de quel côté se trouve le court circuit. Pour la surchauffe, idem ça t'indique si c'est les transistors côté A ou côté B qui chauffent trop mais bon là franchement, je ne vois pas trop à quoi sert la distinction. Et pour la surtension, beh les deux côtés sont sur le même Vin donc encore je ne vois pas à quoi servirait la distinction à mois d'avoir un design très tordu car A et B ont chacun leur masse qui pourrait avoir des potentiels différents. Mais bon, faut vraiment être tordu dans sa tête pour faire une chose pareille.

> 👍 Je pensais juste faire comme le régulateur de tension pour Hydra : la board est soudée à Hermes comme un composant traversant. Ca prends de la surface, mais on peut peut être s'arranger pour mettre des composants CMS en dessous ?

Oui, oui, on peut très bien faire comme ça aussi. Par contre la board fait quand même 3.80cm\*3cm, faut en mettre deux ([à moins d'utiliser celle là plutôt](https://www.pololu.com/product/708) ? (qui a d'ailleurs l'avantage de pouvoir recevoir des capa supplémentaires, ça éviterait de devoir trouver leur place ailleurs et même de faire passer de la puissance sur Hermes)) et il y a déjà la Teensy à souder/à emboiter dans connecteurs femelles plus tous les autres connecteurs. Je veux bien vouloir essayer de me surpasser en routage mais ça risque être chaud chaud là.

astralien3000, 2017-10-02:
> Sinon c'est pas dangereux ça "d’électriser" les pas de vis ?!

Ouai, heureusement qu'ils ont mis un avertissement. C'est d'ailleurs pour ça qu'il est mieux de la souder directement sur Hermes, pas de tentation de la visser autre part. Ou si non faut utiliser des vis nylon.

> Mais de ce que je comprends c'est que DIAGA et/ou DIAGB sont tirés (dans la puce) à la masse lorsqu'il y un problème (survoltage, surampérage, surchauffe) mais ça ne précise pas lequel de ces problèmes.

Il semble que ce soit un simple OR logique, il suffit d'un seul problème pour que le flag passe à 0. C'est différent du LMD1820x ?

> Pour la surchauffe, idem ça t'indique si c'est les transistors côté A ou côté B qui chauffent trop mais bon là franchement, je ne vois pas trop à quoi sert la distinction.

Donc il y a moyen de tout mettre sur une pin ?

> Je veux bien essayer de me surpasser en routage mais ça risque être chaud chaud là.

J'ai confiance en toi ^^ !

Elioty, 2017-10-02:
> Ouai, heureusement qu'ils ont mis un avertissement. C'est d'ailleurs pour ça qu'il est mieux de la souder directement sur Hermes, pas de tentation de la visser autre part. Ou si non faut utiliser des vis nylon.

Pas faux ou sinon la carte avec deux ponts en H ne semble pas "électriser" les pas de vis.

> Il semble que ce soit un simple OR logique, il suffit d'un seul problème pour que le flag passe à 0. C'est différent du LMD1820x ?

Oui ça semble être un ou logique entre tous les problèmes possibles. Et c'est différent du LMD1820x dans le sens que ce dernier ne remonté que l'info de température, d'où le nom du pin : _thermal flag_.

> Donc il y a moyen de tout mettre sur une pin ?

Oui, il y a moyen. Par contre je pense que ça nécessitera peut-être un petit coup de cutter car Pololu a mis une pull-up sur chaque DIAG, et ça ne serait pas trop grave si on soude la carte sur Hermes. Je demanderai confirmation aux élecs quand même.

> J'ai confiance en toi ^^ !

Merci haha !!

astralien3000, 2017-10-05:
Petit résumé en vrac de la dernière réunion : 
 - On va plutôt utiliser la [board "Dual VNH2SP30"](https://www.pololu.com/product/708), qui est en fait plus cher que 2 LMD1820x... donc l'argument du prix n'étais pas valable.
 - le VNH2SP30 ne "prévient" pas d'une future erreur, le flag de diagnostique s'active quend l'erreur est déjà arrivée.
 - Pour la teensy 3.2, on va essayer de trouver des connecteurs qui sont pas trop "hauts"
 - Toujours pour la teensy, à la place de souder des fils en dessous de la carte, on va essayer de trouver des connecteur qui se soudent en surface.

Elioty, 2017-11-03:
Le 2cm max de hauteur tient toujours ? Car je commence un peu à regarder du côté des composants aussi et les capa pour la puissance ([celle-ci sur Hermes 1](https://www.mouser.fr/ProductDetail/Nichicon/UVZ1V331MPD/?qs=sGAEpiMZZMtZ1n0r9vR22cbP6Wf%2fHrkhX5OqxJCqtTQ%3d)) font dans les 12.5mm de haut en moyenne. On ne pourra pas en mettre directement sur la board des ponts en H donc faudra que je les case sur Hermes. Et que ce soit axial ou radial couché, ça ne change rien, elles font aussi 10mm de diamètre en moyenne, et puis ça prendre plus de place en surface ce qui va être serré aussi.

astralien3000, 2017-11-04:
Je n'ai pas encore découpé les plaques d'alu, tu peux prendre plus (genre +1cm max).
De toute façon j'étais large même pour le périmètre PMI de l'an dernier.

Elioty, 2017-11-04:
Donc je peux prendre un volume entre 8x8x2cm et 8x8x3cm si nécessaire ? Si je peux aller jusqu'à 8x8x3, ça facilitera la tache, je devrais pouvoir mettre les capa puissance directement sur la board de ponts en H (je vérifierai quand même bien sûr).

astralien3000, 2017-11-04:
Oui ^^, je viens de vérifier, tu peux même aller jusqu'à 8x8x4cm sans problème.

Elioty, 2017-11-04:
Ok super même si là, je pense que je n'atteindrais pas les 4cm de haut ^-^'

Elioty, 2017-11-05:
Pour le fusible, tu en dis quoi d'un de 30A comme [celui-ci](https://www.mouser.fr/ProductDetail/Bussmann-Eaton/BK-ABC-V-30-R/?qs=sGAEpiMZZMtxU2g%2F1juGqRXDIqpMzLsN3ttdS24t%252bVQ%3D). Et oui, c'est un axial, ça évite les problèmes avec les _fuse holders_ mais ça prend un peu plus de temps à changer du coup (faut désouder, resouder). Sinon pour prendre encore moins de place, on peut mettre des fusibles en CMS aussi mais pour le coup, c'est encore un peu plus compliqué à changer (c'est ce qu'on a sur nos cartes au boulot cela dit, ça doit mieux résister aux vibrations et il vaut mieux dans un avion ou un hélico).

Ça va être serré ce routage 😣

astralien3000, 2017-11-06:
~3,2cm de long ? C'est bien gros en effet.
Après pour souder/désouder, il faudra juste essayer de le mettre à un endroit où c'est pas trop compliqué.

Elioty, 2018-02-19:
La commande mouser a été faite jeudi dernier (15/02). Je devrais la récupérer mercredi.
Pour la commande pwbway par contre, je l'ai passé dimanche (18/02) mais ils sont en congé pour le nouvel an chinois et ils reprennent jeudi donc je ne recevrais les PCB que la semaine prochaine très certainement.

Elioty, 2018-02-21:
Commande PCBWay validée et payée.

> Estimated Finish Time: China Time Zone(GMT+8)2018-02-27

Puis 3 à 5 jours de livraison avec DHL de prévus.

Elioty, 2018-02-24:
Commande mouser reçue, j'ai fait l'inventaire est c'est bon, juste :
- pour les connecteurs puissance, ça a pris 5 minutes à remonter qu'on en a commandé que 2 (on en montera donc un par carte) car tant qu'on n'a pas le bus et les autres cartes, pas besoin de plus ;
- pour le socket à sécables qui va sous la teensy, ce n'était pas très clair et je me suis fait avoir, on a commandé des "2 rangés de 4 pins de long" alors que je pensais commander du 8 pins de long ; ça me paraissait bizarre aussi "qu'ils ne faisait pas les longueurs impaires" 😆 ; du coups, je vais quand même mettre à jour la liste des composants avec [celui-là](https://www.mouser.fr/productdetail/517-929975-01-07-rk) (7 de longs comme ce qu'il y a réellement sous la teensy), et soit on voit avec Eirbot s'ils font une commande bientôt, soit je fais une autre carte rapidement pour passer une autre commande, soit je cherche un magasin dans le coin qui pourrait avoir ça, soit tu (@astralien3000) n'as pas des connaissances à ton ancien labo (pour ne pas divulguer d'info perso) qui pourrait nous dépanner ? Je fais un mail rapidement sur les listes Eirbot et anciens aussi pour voir.

astralien3000, 2018-02-24:
Hmm... Dans le pire des cas, on peut toujours s'arranger avec 2 connecteurs (2*4)pins. Il suffir de retirer une paire de pins sur l'un des connecteurs à la dremel, et ça passe (c'est dégueu, mais ça passe). Et si non, je peux toujours tenter, mais de mon côté je doute d'en trouver.

Elioty, 2018-02-24:
C'est ce que j'avais prévu déjà de base vu que de ce que j'avais compris, je commandais du 8 de long (et non du 7 ou du 4), donc deux pins à retirer.

Je vais voir s'il y a un revendeur dans le coin qui pourrait avoir ça. Tu pourrais peut-être jeter un coups de par chez toi aussi non ?

astralien3000, 2018-02-24:
Ah oui en effet, ^^
J'aurai peut être plus de chances de trouver un revendeur que toi.

Elioty, 2018-02-26:
Les PCB ont été expédiés ce matin. Livraison prévue le 1er mars mais je le récupérerai quelques jours après à un point relai.

Et pour l'autre composant, Eirbot passe commande aujourd'hui ou demain.

Elioty, 2018-02-27:
Frais de douane payés, plus qu'à attendre le colis 😄

Elioty, 2018-02-28:
PCB reçus. J'ai testé pas mal avec mon multimètre en mode "bip bip" pour vérifier que les pistes et vias qui sont relativement proches soient bien isolés, pas remarqué de problème.
J'ai aussi testé les empreintes (tout sauf fusibles, régulateur et CMS) et tout rentre bien, le connecteur puissance à de la gueule sur la carte !

Donc pour l'instant, tout est au vert, à part quelques traces de doigts ou je ne sais quoi sur les PCBs ^-^'

Elioty, 2019-09-26:
La carte est terminée depuis longtemps maintenant, je ferme le ticket.

closed 2019-09-26 by Elioty