xenomorphales/xenomorphales #67: Idée électronique/mécanique modulaire pour Céres  3.0

astralien3000, 2017-06-01:
J'aimerai expérimenter une électronique/méca modulaire.

On part de là : 
 - Céres est découpée en étages, mais certains étages sont difficiles à démonter mécaniquement, et il y a beaucoup de fils qui passent entres les étages
 - Il manque une IHM (choisir le côté, tirette propre, leds de debug accessibles en haut du robot, affichage plus complexe pour les regles de 2018)
 - On ne peut pas charger les batteries sans les démonter du robot
 - Le robot n'a que 2 encodeurs externes on va peut être en ajouter 2 internes

Objectifs : 
 - Avoir des étages facilement séparables
 - Avoir le moins possible de fils entre 2 étages (l'idéal est d'avoir un seul fil à débrancher)
 - Avoir un système de communication commun pour ajouter facilement une nouvelle board

Idées : 
 - Entre deux étages, mettre une "gateway" : une carte à 2 faces ayant des (un ?) connecteurs pour le bus de communication des deux côtés. Elle pourait aussi permetre de choisir de couper le passage de la communication ou de l'alimentation ?
 - Avoir une carte Hermes simplifiée : seul la gestion des moteurs et encodeurs internes est gardée
 - Avoir une carte Hydra simplifiée : seul la gestion des servomoteurs est gardée
 - Avoir une carte de "localisation" gérant les encodeurs externes.
 - Avoir une cart "détection" gérant les GP2 (+ permettre ultrasons ?)
 - Avoir une carte IHM
 - Avoir une carte Alim permettant de vérifier la tension des batteries par exemple.
 - Avoir une rasp sur le bus, à la place de communiquer en USB
 - Avoir une board de communication wireless (si on utilise pas du Wifi)
 - On peut peut être utiliser plusieurs boards de la marque Teensy (2.0 -> 3.6), selon le besoin de puissance de chaque application

Voilà, c'est assez ambitieux, mais ça peut être intéressant ?

Elioty, 2017-06-01:
Wow, super boulot, ça a l'air cool 😍 

Alors première remarque, sur ton schéma, on voit bien les gateways et mécaniquement, ça serait représenter par un plaque sandwich-alu (ou autre matériau) j'imagine. L'idée de la carte qui se positionnerait dans un orifice percé quelque part sur cette plaque de séparation d'étage avec des connecteurs des deux côtés (à mon avis, il faut bien au minimum deux connecteurs par face, un pour la logique avec alim, masse et le bus de com, et l'autre pour passer l'arrivée de la batterie à l'arrêt d'urgence mais c'est à discuter en fonction de ce que l'on veut que l'arrêt d'urgence arrête, ou peut-être on peut juste l'utiliser pour commander un relais en passant par le connecteur logique). Bref, la où je voulais en venir sur ce principe de gateway est que, je pense, ça pourrait/devrait être une réflexion plus large englobant la méca. Tu y fais déjà allusion dans ton message mais je pense qu'il faut bien y réfléchir genre chaque bloc (pris indépendamment, lorsque démonté des autres blocs par exemple) ne dispose que de quatre colonnes et de la plaque du haut avec la carte/interface gateway. Mais les quatre colonnes seraient sur le même axe vertical pour tous les blocs mécaniques du robot donc faut bien réfléchir comment on vient stacker tout ça. On pourrait aussi imaginer que les plaques supérieurs proposent plusieurs "bases de fixation" pour les colonnes du bloc supérieur afin de laisser plus de liberté pour designer (l'occupation de l'espace de) chacun des blocs. Bref, faut y réfléchir aussi si tu veux que les blocs soient aussi facilement interchangeables  mécaniquement (et pas juste électrique/électroniquement) et déplaçable au sein du robot.
Désolé, première remarque plus longue que prévue 😋 

Deuxième, bien plus courte, est-ce qu'on veut vraiment que les servos d'Hydra soient sur le même bus que la com entre nos microcontrôleurs et notre rasp ? Ça me choque un peu comme cela. Et puis on va en discuter/y réfléchir mais on peut peut-être (certainement ?) trouver un bus plus efficace pour la com entre les microcontrôleurs que du TTL/UART half-duplex.

Justement sur la communication inter-microcontrôleur, ça risque être le noeud de la guerre dans toute cette archi. Faut voir quels seront les flux de données qui passeront sur le bus, qui communique avec qui, "peer-to-peer" ou uniquement "master/slaves" voir "masterS/slaves" ou encore en mode "data-centric" (je "broadcast" une information que j'ai acquise par un senseur ou calculée sur le bus et je me fiche de savoir qui consomme l'information). Voir aussi mon avant-dernière remarque un peu plus bas.

Troisièmement, c'est juste d'ordre lexical 😅 
> Avoir le moins possible de **câbles** entre 2 étages (l'idéal est d'avoir un seul **câble** à débrancher)

Avant-dernière remarque, faut peut-être aussi réfléchir à comment tu va mettre en place la boucle d'asservissement là-dessus, bref genre faire un diagramme de séquence pour savoir quelle carte fait quoi et ordonne quoi aux autres.

Dernièrement remarque, concernant le schéma, ne fait pas de boucle sur le bus de communication autour d'Hydra et de la board de détection, connecte-les une seul fois sur le bus, ça serait plus correct.

Bref, je vais m'arrêter là pour maintenant, c'est déjà pas mal je pense mais en tout cas, si on part sur quelque chose comme ça, je suis déjà enthousiasmé pour la coupe 2018 😄😍 et ça peut franchement être un bon projet à défendre/évoquer lors d'entretien d'embauche, sur LinkedIn ou même sur son CV, aussi bien la partie élec qu'informatique.

Elioty, 2017-06-01:
Oh et j'oubliais mais on va devoir chercher plein de petits noms pour toutes ces cartes 😋 
rappel des noms existants :
- Hermès pour la carte contrôlant les moteurs de la base roulante, inspiration de "[Hermès] est le messager des dieux, donneur de la chance, inventeur des poids et des mesures, **gardien des routes et carrefours, des voyageurs** et du commerce" ([source](https://fr.wikipedia.org/wiki/Herm%C3%A8s)), un peu l'aspect chance aussi pour éviter les railleries du genre "c'est l'asserve, c'est l'asserve, [...]" 😆 ;
- Hydra vient de Hydre de Lerne, "créature décrite comme un monstre possédant plusieurs têtes" ([source](https://fr.wikipedia.org/wiki/Hydre_de_Lerne)), dans notre cas, plusieurs bras robotiques.

Pour le nom Cérès, c'est la "déesse romaine de l'agriculture, des moissons et de la fécondité" mais je pense que l'inspiration originale vient plutôt de la [planète naine de notre système solaire du même nom](https://fr.wikipedia.org/wiki/(1)_C%C3%A9r%C3%A8s) (pour faire suite au thème spatial évoqué par le nom de l'équipe). Dis moi si je me trompe @astralien3000.

darwikey, 2017-06-01:
Pour la carte IHM: Janus (dieu des portes et des choix)
Pour la carte alim: Atlas (parce qu'il porte toutes les autres cartes "sur ses épaules")
Pour la carte detection: Heimdall (le gardien du Bifrost dans Thor)

astralien3000, 2017-06-01:
Longue remarque j'espère que je vais rien oublier ! ^^

Pour le premier point sur la méca.
Je suis en train de concevoir un système plsu simple à démonter qu'actuellement (enfin j'espère).
En gros, à la place d'avoir des "barres" en plein milieu du robot (où il faut que le tournevis slalome entre les actionneurs/cartes pour atteindre la structure porteuse), je voulais faire un système plus proche des bords à base d'équerres (la vis serait directement atteignable sur la paroi du robot). Mais je sais pas si ça peut s'appliquer dans tout les cas.

Autre point que tu soulève dans la première remarque : l'interchangeabilité des modules.
Déjà tout les modules ne sont pas interchangeables, ça va de soi. Le module "base roulante" sera forcément en bas. L'IHM forcément en haut (quoique, ça se discute...). Et par exemple, pour un module contenant des bras comme cette année, le changer de hauteur pourrait le rendre inutilisable (comme une grande partie des modules actionneurs, je pense).
Les seuls modules qui pourraient être déplacés dans mon schéma sont la détection, la comm sans fil et la rasp.
Donc je sais pas à quel point c'est important d'avoir une méca ultra modulaire, mais en effet, il faut essayer des trouver des constantes dans la fixation entre les modules. Pas comme actuellement, où l'étage les plu bas prend des vis M6 de 10cm, sur un carré de 10cm de côté, le suivant des M6 de 5cm sur un rectangle dont je me rapelles plus des dimentions pour finir dur des plots qui supportent des barres à base carrée ! XD

Pour ton 2e point sur le bus des servo.
Je pense que vu nos demandes, il va être difficile d'avoir les même connecteurs de toute façon (on aura vraisemblablement besoin de plus de 3 pins).
Par contre pour l'histoire du TTL/UART, il y a un gros avantage : c'est facile à coder, relativement facile à faire niveau élec (quoique ?), et presque tout les MCU gèrent l'UART, mêmes les plus simples.
Pour l'I2C et le SPI c'est déjà pas la même (mais on va considèrérer que tout ce qui est plus gros qu'un ATMega328p les aura quand même donc c'est pas une vraie contrainte).
Pour l'I2C, les anciens m'ont toujours dis que c'était une mauvaise idée, mais bon ils ont dit la même avec la tirette-jumper, et j'ai pas eu de problème depuis 4 ans...
Si tu a une autre proposition (CAN ? Ethernet ? Autre bus bourrin ?), il faut en discuter.
Il y a aussi la possibilitée de faire plusieurs bus en parallèles (genre 2 TTL/UART sur 2 lignes différentes, pour doubler le débit potentiel ?)

J'aimerai vraiment que la comm soit data-centric, ça serait pile dans ma thèse ^^ ! Et en plus ça serait cool !
J'essaie de faire une étude des flux de donnée rapidement... (tient ça vaudrait une petit issue, ça !)
Pour la diagramme de séquence, je pense qu'une certaine partie du travail est fait [ici](https://github.com/astralien3000/robot-control-tuto), mais il faut continuer, en effet.

Pour la dernière remarque je voulais juste représenter que les fils qui se croisent sont le même bus (donc il n'y a pas de composants qui les séparent, si ce n'est des connecteurs sur les boards), en oppositions aux fils qui ne sont pas croisés avec d'autres et qui ne sont donc pas dans le bus.

(ça commence à faire baucoup ! vaste sujet !)

astralien3000, 2017-06-01:
Bon, on vas pas s'étendre trop sur les noms (on verra au fur et à mesure que les cartes sont créées)

@Elioty tout juste ! ^^

@darwikey 
J'aurai plus mis Janus pour la Gateway, vu qu'il a deux faces
Mais ça va bien à l'IHM aussi, en effet !

Pour la carte Alim, j'aurai aussi pensé à Prométhée (son frère en fait), qui donne le feu (= une source d'énergie) aux hommes. En plus Prométhé, même si ça me fait pas plaisir ça fait parti de la saga alien maintenant.

Pour détection = Heimdall 👍 j'ai direct pensé à ça aussi et je pense qu'on peut difficilement trouver mieux. Mais comme tu dis, son utilitée finale peut être discutée. Il y a plein de façon de régler le même problème : 
 - On peut mettre une board par GP2 pour les brancher directement au bus comme des servomoteurs (overkill !!)
 - On peut mettre une carte détection dédiée pour pouvoir  quand même les mettre à la hauteur voulue (= plus haut que l'étage d'Hermes = plus haut que 10cm). Oui parceque c'est pas comme pour toi, Hermes serais vraisemblablement au même étage que les moteurs.
 - On met tout sur Hermes, et on détecte "au ras du sol"
 - Un mix des deux derniers ? On peut créer uen board détection, et laisser les IO sur Hermes, même si on les branche pas. ça fait de la redondance en cas de panne, ou même ajouter de l'information !

astralien3000, 2017-06-02:
J'ai une meilleur proposition pour les boards que les Teensy : 
Des Nucleo (format 32 pins) !
Spécialement la [NUCLEO-F042K6](http://www.st.com/en/evaluation-tools/nucleo-f042k6.html), qui possède [2 Timers avec lecture d'encodeurs (Ctrl-F quadrature)](http://www.st.com/content/ccc/resource/technical/document/datasheet/52/ad/d0/80/e6/be/40/ad/DM00105814.pdf/files/DM00105814.pdf/jcr:content/translations/en.DM00105814.pdf). Elle est [pas cher](http://www.mouser.fr/ProductDetail/STMicroelectronics/NUCLEO-L432KC/?qs=sGAEpiMZZMurtJ7VwBTl0QvLhrnvne9R4rEOA8s4NFPntS%2fnY7Kx3w%3d%3d), comparée aux Teensy. Et en plus elle est supportée par des vrais framework ([embed](https://developer.mbed.org/platforms/ST-Nucleo-F042K6/), [RIOT](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo32-F042)), pas Arduino-like, en somme ^^.
Elle est peut être un peu limite niveau RAM(6K)/Flash(32K), mais ça sera suffisant pour contrôler les moteurs, et lire les encodeurs tout en gérant le bus de communication.
Et au pire il y a des NUCLEO du même format qui sont plus fournie niveau mémoire (par contre c'est la seule avec 2 lecteurs de quadrature, les autres que j'ai vu n'en ont qu'un...).

Elioty, 2017-06-02:
Pour l'aspect méca, je n'ai pas précisé effectivement mais j'avais bien à l'esprit que le bloc base roulante est forcément en bas et que le bloc tout en haut n'est pas forcément IHM mais doit être capable de supporter au minimum l'arrêt d'urgence et le mat de balise. En tout cas, amuse toi bien avec la méco alors 👍 

Pour le bus des servos, c'est aussi surtout que s'ils sont sur le bus avec tout le monde, pourquoi avoir Hydra (si ce n'est pour aligner des connecteurs et des capas de découplage) ? Et si au final on utilise autre chose que de l'UART half-duplex, on n'aura pas le choix de les séparer sur un bus séparé. Car si tu veux faire du data-centric ("je partage l'information que je viens de récupérer ou de calculer"), il faut un truc qui :
- soit gère les collisions (sans risquer de faire cramer quoi que ce soit) pour les détecter et ré-émettre un peu plus tard (temps d'attente aléatoire évidemment) ;
- soit le "droit de parole" tourne sur le bus et vaut mieux que le protocole soit robuste si un nœud plante logiciellement (dans les deux cas de figure de préférence : lorsque c'est à son tour de recevoir le jeton ; lorsqu'il doit le faire passer au suivant).
Du coups pour le premier point, ce qui vient évidemment direct en tête est l’Ethernet mais c'est quand même lourd pour notre besoin. En faisant quelques recherches, le [RS232](https://fr.wikipedia.org/wiki/RS-232)/[UART](https://fr.wikipedia.org/wiki/UART), l'[I2C](https://fr.wikipedia.org/wiki/I2C), le [SPI](https://fr.wikipedia.org/wiki/Serial_Peripheral_Interface) ou le [RS485](https://fr.wikipedia.org/wiki/EIA-485), aucun d'eux ne semblent adapter pour gérer les collisions, ils fonctionnent soit en full-duplex (1-to-1), soit en master/slave et pour ceux en 1-to-1 qu'on bidouille en mode bus, on retombe dans un mode master/slave.
Par contre, eurêka, le [CAN](https://fr.wikipedia.org/wiki/Bus_de_donn%C3%A9es_CAN) gère les collisions !! Tout le monde peut prendre la parole librement sans qu'il n'y ait de master qui donne la parole. Les contrôleurs CAN sont par contre quand même bien présents dans la plupart des microcontrôleurs ARM mais côté atmega, ce n'est pas ça. Contrairement à l'UART qui est absolument partout.
Pour le bus où le droit de parole tourne et qu'il soit robuste à la panne logiciel d'un des nœuds, je ne vois qu'une solution, une horloge commune que tout le monde reçoit (avec un top départ) et chaque nœud est programmé pour parler pendant une fenêtre de temps distincte des autres (ce n'est quand même pas très modulaire si tu veux ajouter un nœud, si tu n'as pas de spare dans ton partitionnement temporel du droit de parole...). Et je ne pense pas qu'il existe des trucs comme ça implémenté en hardware donc ça risque d'être chiant 😅 mais ça devrait poser problème quand même si un nœud rate le top départ ou reboot après coups.
Bref, je vais continuer mes recherches sur le sujet mais si le master/slave te convient (ce que je ne pense pas si tu préfère du data-centric), ça peut se faire avec de l'UART half-duplex même si je préférais qu'on utilise un truc adapté comme le SPI (quand même lourd, il faut un fil slave-select par slave) ou l'I2C (mieux, le slave est adressé par le protocole de la couche liaison). Sinon, il n'y a à mon avis pas de secret si le CAN a réussi à s'imposer dans plusieurs industrie, c'est que ça marche 😋 et dans les liens externes sur sa page wikipedia on trouve... [un cours sur le CAN d'un prof de l'ENSEIRB](http://sitelec.org/download.php?filename=cours/canbus.pdf) (bon ça date un peu, l'ancien logo avant la fusion avec matmeca).

Vaste sujet comme tu l'as dit.
Autres liens que j'ai lu jusqu'à maintenant pendant mes recherches, je les met là.
http://community.silabs.com/t5/8-bit-MCU/One-master-multiple-slaves-over-UART/td-p/57005
http://www.edaboard.com/thread302747.html
https://www.maximintegrated.com/en/app-notes/index.mvp/id/214

Pour les noms des boards, on a le temps effectivement mais je vois que vous avez plein d'idées 😃 

> On peut mettre une board par GP2 pour les brancher directement au bus comme des servomoteurs (overkill !!)
Tu veux faire des GP2 numériques ?! hahaha 😆 

Pour les nucleo, pourquoi pas, elles sont minces en plus (format Arduino nano de ce que j'ai compris). Et si on découpe toute l'archi comme ça, je pense que 6K de ram et 32K de flash seront largement suffisant pour Hermes. Puis ce petit STM32F042K6 dispose quand même niveau com d'UART, de SPI, d'I2C, d'I2S, d'USB et de CAN donc a n'a que l'embarras du choix.

Et d'ailleurs, une question qui m'est venu en tête (et je m'arrête là pour ce soir), tu comptes flasher les cartes comment ? Que ce soit de l'Arduino, du Teensy ou du Nucleo, faut passer par l'USB (et pour les Teensy faut en plus appuyer sur son bouton de flashage, pour les Nucleo, je ne sais pas). Tu veux flasher depuis la raspberry embarqué ou depuis un PC portable à côté du robot ? Mais dans ce dernier cas là il faut que les USB de programmation soient accessibles ou tirés vers les bords du robots avec une rallonge quelconque installée proprement dans le robot.

astralien3000, 2017-06-03:
Ouai, j'avais parlé avec @Titwin des différents bus.
 - Pour l'UART half-duplex, on peut mettre en place un système de multi-master qui s'échangent un Token.
 - On est pas obligé de mettre Hyrda, si elle n'a aucune autre fonctionnalité. Juste les servo branchés directement sur le bus ça irai en effet.
 - Le SPI on oublie, les histoire de Chip Select c'est pas top. Le seul avantage c'est le full duplex, mais je pense pas que ça soit si utile.
 - Perso apprendre à utiliser le CAN me plairait. Après bon, on se dis que si on veut programmer un truc rapidement avec une Arduino et l'ajouter au bus, on pourra pas trop le faire. Mais après si on est pas à l'arrache pour la coupe, je vois pas pourquoi on voudrait faire ça ! ^^
 - Problème : la Rasp ne gère pas le CAN, il faut un adaptateur. Mais l'UART half-duplex aussi gagnerai a avoir un adaptateur en fait...
 - Idée à la con : Pourquoi ne pas mettre 2 Bus différents ? Le CAN pour les boards qui le gèrent et un UART half-duplex pour le reste (servos, cartes à base d'AVR, etc...)

Le dernier point m'amène à une autre problèmatique : On tire quoi comme cable d'alim en plus des bus de communication ?
 - On aura 2 batteries lipo 2S en série pour alimenter les moteurs (donc 14-16V)
 - On veut alimenter les servo XL320 et SCS15 (donc 7-8V)
 - ça peut se discuter, mais j'aimerai me débarasser de la batterie USB 5V, j'aimerai n'avoir que les 2 LiPo 2S
 - Mais quoi qu'il arrive on aura vraisemblablement besoin de 2 autres tensions (3.3V et 5V)
 - En plus, il faut faire passer l'arrêt d'urgence. En effet, on vas pas le contôler logiciellement (pas passer pas le bus de comm),et it serait quand même mieux de passer par les paserelles communes.
 - L'arrêt d'urgence, dans l'idéal ne devrais couper que la puissance du robot (tout ce qui peut entrer en mouvement), mais si les boards de contrôle restent sous tension, ça peut avoir un intérêt (pour programmer le robot, par exemple)

Il y a plusieurs possibilités : 
 - On ne fait passer que la puissance dans le bus (7V / 14V / les deux) et chaque board aura un régulateur de tension.
   - Chaque board aura des composants en plus (plus cher)
   - Pour un régulateur à découpage, on nous a dis de nous méfier quand il y en a plusieurs en même temps.
   - On peut y aller au régulateur linéaire, mais on consommera plus de batterie.
   - Avantage : Les cartes seront assez indépendantes de la carte alim (elles pourront être branchées directement aux batteries, pour des tests par exemple)
 - On convertis toutes les tensions sur la carte alim, et on distribue ça dans tout le robot (3.3V, 5V, 7V, 14V)
   - ça va faire beaucoup de fils
   - les boards ne sont pas capable de fonctionner sans la tension adéquate (obligation d'avoir la carte alim ou des batteries à la bonne tension)
   - les régulos de la carte alim crame -> plus de robot
 - On peut aussi envisager que d'autres étages aient leur propre batterie
   - Une LiPo spécifique aux servos
   - La batterie logique 5V
   - Le nombre de batteries à charger va grandir (mais ça peut rester gérable)
   - Le poids et la place prise par les batteries peut être une gène
   - La gestion de l'arrêt d'urgence va devenir folklo !

Voilà, ce sont trois proposition auxquelles j'ai pensé, on peut en trouver d'autres ou faire des mélanges.

Pour le flashage des cartes : 
Déjà j'aimerai arriver au point où on arrive à la coupe et on a pas besoin de flasher les boards qui ne gèrent pas la stratégie (comme sur Buggybot).
Mais à part ça, oui, sortir l'USB sur les bords de la méca serait une bonne solution.
De toute façon faire passer les USB par le bus commun pour atteindre la rasp n'est pas envisageable ?
Je sais que RIOT expérimente des façons de flasher, principalement sans fil, mais on pourrait aussi bien flasher à travers un bus CAN. Ca reste pas très stable encore, donc l'USB reste le plus sûr. Et puis c'est pas un choix définitif, si un jour on arrive à flasher via le bus commun, ça sera pas un problème.
D'ailleurs c'est encore un point pour les Nucleo, pas besoin d'appuyer sur un bouton !

darwikey, 2017-06-03:
Pour la teensy, normalement y a pas besoin d appuyer sur un bouton pour programmer ( en début d année j'étais obligé à cause d un problème de driver je pense)

Elioty, 2017-06-03:
Pour le bus, on a l'air de tendre vers le CAN. Pour les Arduino, il existe certainement des shields pour avoir du CAN. Et pour la Raspberry, un adaptateur USB/CAN doit pouvoir se trouver sans trop de problème non plus.
Pour l'idée à la con, oui on pourrait tirer deux bus mais est-ce vraiment nécessaire ? Ça alourdit le tout juste pour pouvoir gérer les atmega au final (car même si on câblait les servos directement sur le bus, il faudrait une carte (sans µP) pour tous les brancher proprement et leur mettre des capa de découplage). Les nucleo ne sont vraiment pas chères il semblerait donc pourquoi pas avoir de l'ARM partout ? (et puis on peu toujours acheter une ou deux cartes/shield CAN pour les atmega).

Pour l'alimentation, c'est peut-être plus cher comme tu l'as dit mais plus modulaire de n'envoyer qu'une seule tension à côté du bus et chaque module se débrouille pour réguler la/les tension(s) qu'il lui faut avec des régulateurs supportant l'ampérage nécessaire par le bloc en question, et moins de fils aussi qui traverse tout le robot. Et libre à eux d'avoir leurs propres batteries également.
Et avec cette solution, pour l'arrêt d'urgence on peut tirer entre les étages : le bus de données (juste deux fils pour le CAN), le + et le - des batteries avec un XT60 et en plus, un autre file puissance qui est au même potentiel que le + des batteries mais derrière l'arrêt d'urgence, avec un XT60 aussi soit avec un seul fil soit on remet la masse.
Il faudra quand même penser à mettre un interrupteur genre comme ceux sur les alimentations ATX (pour PC de bureau) quelque part à côté des batteries pour couper toute alimentation.

L'autre solution que tu as évoqué est d'avoir une carte alim qui regroupe tous les régulateurs puis distribue, à l'aide de plusieurs fils, les différents niveaux de tensions dans le robot, et tire un fil jusqu'en haut pour récupérer l'arrêt d'urgence.

Bref je vais récapituler avec un tableau les deux solutions, ça sera plus clair :

Alim décentralisée | Carte alim
------------------ | ----------
Bus de données | Bus de données
+15V | +15V
GND | GND
+15V (derrière l'arrêt d'urgence) | +15V (derrière l'arrêt d'urgence)
N/A | +3.3V
N/A | +5V
N/A | +7.4V (derrière l'arrêt d'urgence)

Si je n'ai rien oublié, on gagne donc trois fils. Sachant que le 7.4V alimente des servos, il faudrait donc mieux que ça soit un fil épais. Et pour le 3.3V et le 5V, pas forcément aussi épais que des fils puissance mais faut pas prendre un diamètre trop petit non plus car on alimente toute la logique du robot avec.
Personnellement, je penche plus pour l'alim décentralisée.

Dernier point, pour le flashage des cartes effectivement j'espère qu'on peut pondre des programmes qui fonctionnent bien sans trop de bug et qui ont leurs constantes paramétrables à travers le protocole applicatif qu'on utilise sur notre bus de données.
Pour les USB, vaut mieux les rendre accessibles depuis l'extérieur du robot. Parce que oui, remonter les USB par nos gateways n'est pas envisageable : soit pas modulaire en ne provisionnant que n USB ; soit un seul USB de câblé mais faut installer un hub à chaque étage 😱.
Et pour flasher par le bus, faut faire notre bootloader (et potentiellement notre loader côté Linux), ça peut se faire mais c'est pas mal de boulot et il faut très certainement investir dans une sonde de debug hardware (genre JTAG) pour faire ça. A moins que les nucleo ont du debug embarqué (comme la stm32f4-discovery il me semble) ? Dans tous les cas, ça c'est du plus qui n'apporte pas réellement de fonctionnel si ce n'est pouvoir flasher nos cartes depuis la raspberry sans rien brancher de plus. Tu veux faire le genre d'algo d'IA qui se modifie lui-même ?! (je crois que des gars font ça avec une IA sur un FPGA, c'est fou 😆)

astralien3000, 2017-06-06:
Petite remarque rapide (mais importante !)
32Ko de Flash sur ARM c'est vraiment hardcore en fait.
Je viens de faire une petite étude de la taille de mon code actuel : 
~10Ko avec un main vide ! (bon, l'OS fait du travail derrière, et surtout appel du printf sans notre consentement...)
Rien que les drivers moteurs ça nou ajoute du ~8Ko !
moteurs+gyro+asserv angle, ça dépasse déjà les 32Ko...
Et juste pour info mon code complet (drivers gyro,moteurs,bras,... + strat + cli) ça fait ~56Ko (pour 8Ko de RAM, mais là on peut facilement diminuer)

Ça monte beaucoup plus vite que sur AVR.

Elioty, 2017-06-06:
Mouais, ça va être tendu avec cette nucleo. Il n'y en a aucune avec un plus gros µP qui a quand même deux lecteurs de quadrature ? Sinon beh Teensy ? Elles gèrent bien le CAN normalement.

Sinon remarque pour comparer ARM et AVR, le premier a des instructions de taille fixe, 32 bits, sur AVR c'est variable, 16 ou 32 bits. Par contre selon la taille des valeurs manipulées, vu que l'APU de l'AVR n'est que 8 bits, une opération simple (genre une somme) peut prendre plusieurs instructions. Pour les floats, c'est fait en soft dans les deux cas mais ce n'est peut-être (certainement) pas la même implémentation avec les mêmes optims. Et tu as aussi peut-être le float64 (aka le double en C/C++) de gérer dans l'implémentation des flottants sur ton ARM. Seul le float32 est géré avec avr-gcc.

astralien3000, 2017-06-08:
- https://www.analoglamb.com/product/seeduino-stm32f405/
 - https://www.analoglamb.com/product/maple-tree-ret6-mini/

kazord, 2017-06-08:
Question qui peut paraît bête mais qui mérite réflexion (de mon point de vue au moins) 

Comment sont connecté les gateways entre elle et aux carte :
Carte = gw = carte = gw....

Ou 

Carte = gw = gw = gw = gw
                II.      II.      II.      II
            Carte. Carte         Carte

La différence devrait se voir dans deux cas : 
Test sans un module
Fonctionnement/cablage des bus de données

astralien3000, 2017-06-08:
En effet.
Je pense que de toute façon toutes les boards devraient avoir au moins 2 connecteurs pour le bus pour pouvoir les chainer.
La gateway pourait en avoir plus pour permettre les deux cas que tu a donné.

Elioty, 2017-06-09:
Pour faire du CAN, il vaut mieux faire uniquement du chainage tout droit, cad que les gateways, comme les cartes n'aient que 2 connecteurs pour le bus de données. On pourrait mettre plusieurs connecteurs pour le bus sur la gateway mais ça serait juste pour faire revenir le bus sur la gateway pour repartir sur une autre carte, totalement débile. Par contre, il ne faudra pas oublier de faire un connecteur de "fin de bus" avec la résistance qu'il faut bien pour mettre aux deux extimités du bus CAN. Bref, je voulais faire un schéma depuis hier mais j'en ai trouvé un sur la page wikipedia qui devrait convenir avec une petite explication :
[image]
Voila donc pour notre utilisation, je pense effectivement qu'on mettra deux connecteurs (à deux pôles) sur nos cartes. Les deux connecteurs d'une même carte seront directement reliés. Les deux liens seront tirés vers le contrôleur CAN de la carte (puce externe ou directement le µP). Et faudra faire deux connecteurs avec juste une résistance pour fermer la boucle aux deux extrémités.
Aussi, d'après Wikipedia, jusqu'à 30 mètres, on peut espérer un débit 1 Mb/s ce qui devrait être largement suffisant pour notre utilisation.

Sinon pour les deux cartes que tu as trouvées @astralien3000 :
- la première est sympa, FPU, pas mal de RAM et de Flash, 168MHz (!!!), le µP a bien un contrôleur CAN, la board est au même format que la Teensy 3.1, sans contact à la noix dessous mais plutôt un vrai connecteur (c'étaient chiant/difficile de souder des fils sur un pad tout plat, sans trou au milieu...), un circuit hardware pour accélérer le calcul de CRC (toujours utile) et des lecteurs de quadrature, bref vraiment bien celle là
- la seconde est plus petite en espace mémoire, pas de FPU, plus lente (mais quand même assez rapide pour notre utilisation) mois d'IO (et pourtant plus volumineuse !!!) mais a tout le reste qu'il faut bien et surtout, moins chère
Bref à voir selon le budget et le bsoin mais elles ont l'air cool 😉

kazord, 2017-06-09:
Il existe 2mode pour le CAN, avec des connections différents/placement de résistance different
Avec 1m/bits en un bus lineaire
Ou 125kbits en mode osef ( linéaire et étoile mélangé) => bas débit securisé, un truc comme ca

Un gateway avec une carte connecté peu faire un T avec le bus CAN
Par contre je ne sais pas le comportement du bus si tu as pas de carte (=2pin ouvert sur l air)

J aurai bien vu une carte avec juste la res de 120ohm a mettre au 2 bout sur les gateways de chaque coté

astralien3000, 2017-06-09:
Ok pour le CAN, je savais pas.
Ok pour Skype, mais avant, j'aimerai qu'on liste les connecteurs qu'on pourrait utiliser.

Elioty, 2017-06-09:
@kazord
Ah je ne connaissais pas qu'on pouvait avoir une topologie en étoile avec du CAN. La page Wikipedia du CAN en français n'évoque que son nom (ISO 11898-3 : low-speed, fault tolerant) sans le détaillé, il est bien détaillé sur la page en anglais cependant.

Je ne pense qu'avoir deux pins en l'air pour le CAN ISO 11898-3 pose problème, tu vas juste capter un plus de bruit, mais juste la longueur de deux pins, ça devrait être totalement marginal.

Sinon, pour les résistances, c'est un peu la merde sur le bus en étoile et pas modulaire (!!), d'après Wikipedia, chaque nœud qui se connecte sur le bus doit embarquer une résistance (je ne sais pas si c'est pull up ou down ou autre chose) pour les deux liens du bus qui est une fraction de la résistance globale sur le bus, qui doit être proche mais supérieur à 100ohm (c'est le proche qui pose problème pour la modularité).
Pour un bus linéaire, c'est juste une résistance de 120ohm à chaque extrémité et c'est tout. C'est pour ça que je proposais dans mon message précédent de faire juste un connecteur tout con qui relie ses deux pôles par une résistance, pour fermer la boucle. Et ça ne marche pas sans, on a eu le soucis au boulot une fois 😆 

@astralien3000 
Pour les connecteurs ? Tu parles en règle général pour à peu près tout ou juste pour le bus ?

astralien3000, 2017-06-11:
Juste pour le bus. On a déjà à peu près répondu pour le reste.
Je verrai pour l'histoire du montage.

Elioty, 2017-06-14:
J'ai réfléchi un petit peu et je pense que les embases comme sur la première Hermes conviendraient. Où tu veux un autre connecteur pour bien les distinguer (ou peut-être un peu plus compact) ? À défaut, faudrait essayer de bien colorer les embases (mâles comme femelles) afin de les différencier.

astralien3000, 2017-06-15:
Ces connecteurs peuvent gérer les 14V ? Ou il faut un 2e connecteur ?

Elioty, 2017-06-15:
Aucun problème avec les embases : 250V, 3A max, on en est loin ;)

http://www.mouser.fr/Search/ProductDetail.aspx?qs=hSmm4fxMIuPWb6osY0uomA==
http://www.mouser.fr/Search/ProductDetail.aspx?qs=mrPiglD9aYKHxZkM06vJ3A%3d%3d
http://www.mouser.fr/Search/ProductDetail.aspx?qs=BLN8Q0P37WapYBZgTV5Zeg%3d%3d

astralien3000, 2017-06-16:
Pas si on utilise [des moteurs un peu plus péchus](http://www.francerobotique.com/moteurs-classiques/567-moteur-m%C3%A9tal-plan%C3%A9taire-premium-hd-12v-612rpm-avec-encodeur.html) (20A/moteur, potentiellement), si ?
Bon après je viens de voir que les connecteurs qu'il y a déjà sur les moteurs sont aussi pour 250V et 3A...

Si c'est quand même bon, on va pouvoir faire le Skype, mais pas avant Mercredi.

Elioty, 2017-06-16:
Ah non mais les embases que je proposais, c'est juste pour le bus de données. Pour l'alimentation, ça serait plutôt un XT60 à côté. Puis tes connecteurs sur les moteurs, faudra les changer par des XT60. Tu veux que je regarde s'il existe des connecteurs avec deux gros contacts pour l'alim et d'autres contacts plus classique pour du logique ?

Elioty, 2017-07-05:
# Intro et questions d'ordre général

Alors pour mes devoirs suite à la dernière réunion, il semblerait que ça soit possible d'avoir deux encodeurs et du CAN sur une seule Teensy 3.2. Devrais-je regarder aussi pour la 3.5 et/ou 3.6 ? Elles ont beaucoup plus d'IO mais prennent du coups également bien plus de place en longueur mais ont aussi un uP plus pêchu (plus puissant, FPU, plus de ram, flash, cache selon les cas, [bref je vous laisse lire le tableau comparatif des uP proposés par les teensy](https://www.pjrc.com/teensy/techspecs.html)).

# Démonstration de faisabilité sur Teensy 3.2

## Ressources documentaires

Pour revenir à la 3.2 dans un premier temps, il faut, pour confirmer mes dires, deux documents :
 - [la datasheet du uP MK20DX256](https://www.pjrc.com/teensy/K20P64M72SF1.pdf)
   - [ou le user manual mais le document est bien plus long](https://www.pjrc.com/teensy/K20P64M72SF1RM.pdf), tu en auras besoin pour faire des drivers
 - [la schématique de la Teensy 3.2](https://www.pjrc.com/teensy/schematic32.gif)

## Signaux nécessaires

À partir de la page 211 du user manual, il y a la description de tous les signaux gérer par le uP. Extrait pour les signaux qui nous intéressent :

Chip signal name | Module signal name | Description | I/O
---------------- | ------------------ | ----------- | ---
FTM1_QD_PHA | PHA | Quadrature decoder phase A input. Input pin associated with quadrature decoder phase A. | I
FTM1_QD_PHB | PHB | Quadrature decoder phase B input. Input pin associated with quadrature decoder phase B. | I
FTM2_QD_PHA | PHA | Quadrature decoder phase A input. Input pin associated with quadrature decoder phase A. | I
FTM2_QD_PHB | PHB | Quadrature decoder phase B input. Input pin associated with quadrature decoder phase B. | I
CAN0_RX | CAN Rx | CAN Receive Pin | Input
CAN0_TX | CAN Tx | CAN Transmit Pin | Output

## Table de multiplexage

En cherchant les signaux qui nous intéressent dans la datasheet, on peut les retrouver dans la table de multiplexage (en page 62, page 207 pour le user manual). En résumé les signaux sont disponibles sur les pins :

Signal | uP pin | uP pin
------ | ------ | ------
FTM1_QD_PHA | 28/PTA12 | **35/PTB0**
FTM1_QD_PHB | 29/PTA13 | **36/PTB1**
FTM2_QD_PHA | **41/PTB18** | 
FTM2_QD_PHB | **42/PTB19** | 
CAN0_TX | **28/PTA12** | 41/PTB18
CAN0_RX | **29/PTA13** | 42/PTB19

La seule combinaison possible est mise en gras dans le tableau ci-dessus.

## Disponibilité des pins du uP sur la Teensy 3.2

Ensuite, dernière vérifications à faire sur la schématique de la Teensy :

uP pin | Board pin
------ | ---------
28/PTA12 | 3
29/PTA13 | 4
35/PTB0 | 16
36/PTB1 | 17
41/PTB18 | 32
42/PTB19 | 25

## Remarques

Bref, je vous invite à vérifier quand même, c'est rapide, vous avez toutes les infos ici a priori.

Et sur la version précédente d'Hermes, le deuxième encodeur est déjà branché sur ces pins là, et le premier utilise les pins 3 et 4 qu'on réserverait plutôt au CAN dans la nouvelle version. Cependant, les pin 16 et 17 sont utilisés pour des GP2 (en ADC donc) mais le sujet de ce message était de confirmer ou non la possibilité d'avoir deux encodeurs et un bus CAN sur une même Teensy 3.2. Il faudra donc, @astralien3000, que tu complètes ton besoin pour les nouvelles cartes Hermes, mais on avait toujours des pins ADC de disponibles si besoin.

# Annexes

- [la liste des datasheet et user manual des uP des Teensy](https://www.pjrc.com/teensy/datasheets.html)
- [les schématiques des Teensy](https://www.pjrc.com/teensy/schematic.html)

ps : je kiffe le markdown 😋

astralien3000, 2017-07-06:
Bon boulot !!!

Alors, je préfère rester sur du Teensy 3.2 si possible (et ça a l'air possible), 3.{5,6} c'est trop gros mécaniquement et en terme de pussance (on a pas besoin d'autant). J'ai vérifié, et j'allais te demander pourquoi FTM0 n'est pas envisagé, puis j'ai lu le user manual.
 
+1 pour faire un document de design dans le dépôt (plutôt en markdown). Après je suis pas sûr de comment tu veux faire ça, faudra qu'on voie. Mais c'est pas le sujet de cette issue. Et justement, je vais faire le nouveau cahier des charges dans #66 sous peu.

Elioty, 2017-07-06:
> J'ai vérifié, et j'allais te demander pourquoi FTM0 n'est pas envisagé, puis j'ai lu le user manual.

Oui le premier FTM n'a pas la capacité de lire des quadrature 😅 

>+1 pour faire un document de design dans le dépôt (plutôt en markdown). Après je suis pas sûr de comment tu veux faire ça, faudra qu'on voie. Mais c'est pas le sujet de cette issue. Et justement, je vais faire le nouveau cahier des charges dans #66 sous peu.

Ok, impatient de voir le cahier des charges pour la nouvelle Hermes. Et pour le document de design, c'est vrai que le markdown a l'avantage d'être rendu directement sur GitHub, donc pas besoin d'installer de logiciel particulier pour lire le document formaté, juste l'ouvrir sur GitHub.

Elioty, 2017-11-04:
Concernant la partie alimentation des cartes, j'ai trouvé [ces connecteurs qui ont l'air vraiment pas mal](http://www.mouser.com/ds/2/185/DF63_catalog-939092.pdf) (spécial dédicace le nom de la marque ; et ils font à peu près tout, connecteurs coudés ou non, et aussi "in-line" pour connecter deux câbles, pour faire rallonge). On utiliserait certainement la version 5 pôles gérant 10A je pense, un pôle pour le 15V, deux pour le 15V derrière l'arrêt d'urgence et deux pour la masse. Vous en pensez quoi de celui ci ? On garderait les XT60 en sortie des batteries et en entrée des moteurs quand même je pense, ça serait juste pour le "bus" d'alim.

astralien3000, 2017-11-04:
Le problème est que les 20A qui passent par la ligne avec arrêt d'urgence vont potentiellement passer aussi par celle sans AU (si l'AU n'est pas directement sur la carte alim : voir #70 et #71).
Et en plus ça fait "pas propre" les 5 fils, j'aimerai mieux en avoir 3 de 20A que 5 ou 6 de 10A. Mais j'imagine que c'est pas évident à trouver.

Elioty, 2017-11-05:
> Le problème est que les 20A qui passent par la ligne avec arrêt d'urgence vont potentiellement passer aussi par celle sans AU (si l'AU n'est pas directement sur la carte alim : voir #70 et #71).

Ah oui effectivement, je devais commencer à fatiguer hier soir haha.

Et je pense plutôt qu'on se limitera au premier que tu as linké, 7.92mm de pitch, ça commence à faire assez large (27.72mm de large pour 3 contacts pour le connecteur sur la board contre 32.52mm pour le modèle avec un pitch de 10.16mm). D'autant plus qu'ils n'ont que deux ampères de différence.

Pour le prix :
- [crimp socket](https://www.mouser.fr/ProductDetail/Hirose-Electric/DF22-3S-792C28/?qs=%2fha2pyFadugsgBEUKEUl%2fwVgP1%252bt%2f8tSOFyf7Gvr97qqeVkK5nbrCQ%3d%3d) : 0.408€/p (le connecteur/boitier plastique à l'extrémité du câble)
- [right angle header (tin-copper)](https://www.mouser.fr/ProductDetail/Hirose-Electric/DF22-3P-792DS05/?qs=%2fha2pyFadugUeOwwRZmOQCC7Asx0dh41YuDR2f63w3bXtbntVH2y65NQKBi2M8H3) : 0,867€/p ; [right angle header (gold)](https://www.mouser.fr/ProductDetail/Hirose-Electric/DF22-3P-792DS06/?qs=%2fha2pyFadugUeOwwRZmOQCC7Asx0dh41X6UA4bA%2fSYv3cMLGAr22e%252bHdSFqaM5v%2f) : 1.34€/p (le connecteur sur le PCB)
- [crimp contact for socket (tin)](https://www.mouser.fr/ProductDetail/Hirose-Connector/DF22A-1012SCF/?qs=sGAEpiMZZMuyN6ZGvFhteY6wXrM1uIbb1FFrdiWtPUk%3d) : 0.315€/p \*3 ; [crimp contact for socket (gold)](https://www.mouser.fr/ProductDetail/Hirose-Electric/DF22A-1012SCFA/?qs=%2fha2pyFadujlYS3bkPPVwiJ%2feReF1wG%252bnHHz7NuksqN0podhugDkBg%3d%3d) : 0.57€/p \*3 (le contact à souder sur le fil et à insérer dans le crimp socket)
- [retainer](https://www.mouser.fr/ProductDetail/Hirose-Electric/DF22-3RS-P-792/?qs=%2fha2pyFadugjaJf1bVFcB8XaW2V0bMbLTwdBmHnPRrXS2pO4X%2frorQ%3d%3d) : 0.272€/p (a priori nécessaire pour bien fixer le câble, voir les schémas chapitre _Mated Dimensions_ de la datasheet)

Ce n'est pas si peu cher que ça je trouve. Et faut rajouter le fil 10AWG avec ça. D'ailleurs ça serait cool si quelqu'un pouvait vérifier les références pour être sûr que tout ça s'accouple bien.

Par contre, pas trouver de fil sur mouser (ça fait un moment que je cherche... je ressaierai plus tard) qui rentrerait dans les contraintes de nos crimp contact (diamètre extérieur de 3.6 à 4.9mm) mais sur farnell, avec [cette recherche ](http://fr.farnell.com/w/c/cables-fils-cables-assembles/fil-de-cablage-unifilaire?diametre-de-cable-awg=10awg&diametre-exterieur=3.66mm|3.73mm|4.52mm|4.572mm|4.65mm|4.67mm|4.7mm|4.83mm|4.85mm), j'ai trouvé ça en [noir](http://fr.farnell.com/pro-power/pp001191/tri-rated-wire-6mm2-black-1m/dp/2528087) et en [rouge](http://fr.farnell.com/pro-power/pp001275/tri-rated-wire-6mm2-red-1m/dp/2528180) qui ont l'air d'aller (sauf pour les brins, ces fils ont 84 brins de 0.3mm de diamètre alors que la datasheet pour les contacts dit 104 brins à 0.24mm de diamètre mais ça reste du 10AWG donc je pense que ça devrait passer).

Sinon pour l'idée des deux rangés de trois pôles, je n'ai pas trouvé chez ce même constructeur mais dans ce cas là, on peut certainement prendre le même type de connecteur que ceux utiliser pour alimenter les cartes PCI-E utilisées dans les PC (principalement pour les cartes graphiques) mais pas sûr que ça tiennent la quinzaine d'ampères par pôle.