xenomorphales/xenomorphales #70: Carte Alim

astralien3000, 2017-11-04:
# Cahier des charges

 - Se branche à 2 lipo 2S
 - Capacité de lire la tension des batteries (globale ou de chaque cellule ?) et de la balancer sur le bus de comm
 - Capacité de lire le courrant cosommé ?
 - Permetre de se brancher au chargeur de lipo sans débrancher les batteries ? (couper toute alim dans ce cas)
 - (si l'IHM ne s'en occupe pas) gestion de l'Arrêt d'urgence.

Elioty, 2017-11-06:
Éventuelle switch ultime :
https://www.mouser.fr/ProductDetail/NKK-Switches/SW3822-U/?qs=sGAEpiMZZMtNT9UGfLL4eMYHJNfIn19vYTYTSE87yXA%3d

astralien3000, 2017-11-21:
J'ai mis à jour le cahier des charges selon nos discussion de la dernière réunion.

À propos du switch ultime, les switch à 30A ne me plaît pas des masses.
Que pense-tu de faire un système avec un switch "logique" qui contrôle un relais ?
Il semble plus facile de trouver des relais 30A, et moins cher.