xenomorphales/xenomorphales #71: Carte IHM

astralien3000, 2017-11-04:
# Cahier des charges

 - Bouton pour choisir la couleur du robot.
 - Leds pour indiquer la couleur
 - Gestion de la tirette ? (en + d'Hermes)
 - Gestion de l'Arrêt d'Urgence ?
 - Communication sur le bus pour donner l'état (couleur, tirette, AU)

Elioty, 2017-11-18:
- Pour les boutons, je vais en mettre quatre (comme sur Hermes avec ses fins de course) mais question design et UX, tu préfères [toggle switch](https://www.mouser.fr/Electromechanical/Switches/Toggle-Switches/_/N-5g2j?P=1z0z63x), [pushbutton](https://www.mouser.fr/Electromechanical/Switches/Pushbutton-Switches/_/N-5g30?Ns=Pricing%7c0&P=1z0z63x) ou [slide switch](https://www.mouser.fr/Electromechanical/Switches/Slide-Switches/_/N-5g2x?P=1z0z63x) ?
- Pour les LED RGB, c'est un peu la merde, [Mouser n'en propose que 10](https://www.mouser.fr/Optoelectronics/LED-Lighting/LED-Emitters/_/N-8usfd?P=1z0ysweZ1z0z63x), [Farnell seulement 2](http://fr.farnell.com/c/optoelectronique-afficheurs/produits-led/led-multicolores-standard-lt-75ma?couleur-de-led=red-green-blue&montage-led=through-hole) (et ceux sont des "piranha"), et [RS juste 7](http://fr.rs-online.com/web/c/afficheurs-et-optoelectronique/led-et-accessoires/led/?applied-dimensions=4294039791,4294041476,4294041483,4294380494,4294502015). Mais le plus gros problème, c'est leur tension directe, aucune ne peut être drivé directement avec une Teensy 3.2 (elles ont toutes une tension >= 3.3V, et même 3.2V ou 3.1V, je pense que ça ferait trop juste). On peut donc peut-être faire comme le montage que j'ai à la maison avec [ces drivers de PWM SPI (fait pour contrôler des LEDs)](https://www.adafruit.com/product/1455) avec [ces LEDs RGB](https://www.adafruit.com/product/302) ?
- Pas de problème pour la tirette.
- Pas de problème pour l'arrêt d'urgence avec un XT60 je pense (tu confirmes ?) et juste une bête embase 2 pins pour le monitoring sur son deuxième channel.
- pas de problème CAN comme les autres cartes.

astralien3000, 2017-11-21:
J'ai modifié les cahier des charges en réponse à tes question, et à la réunion : 
 - boutons toggle switch
 - leds monochromes au lieu de rgb
 - shield rasp

Juste une remarque/demande sur le monitoring de l'arrêt d'urgence : 
Il y a pas moyen de regarder la tension directement sur la ligne de puissance (avec un diviseur ou autre ?), je suis pas fan du fait de brancher 2 fils différents à l'AU. Bon après si c'est trop galère autrement, on le fera...

Elioty, 2017-11-25:
Pour la partie CAN, je vais pouvoir m'inspirer de ça : http://copperhilltech.com/pican-2-can-interface-for-raspberry-pi-2-3/
Qui utilise [ce contrôleur](https://www.mouser.fr/ProductDetail/Microchip-Technology/MCP2515-I-SO/?qs=sGAEpiMZZMsUzhEcHltCuTyUJ%252bMSzD1r) et [ce PHY CAN](https://www.mouser.fr/ProductDetail/Microchip-Technology/MCP2551-I-SN/?qs=sGAEpiMZZMsUzhEcHltCudXjf0%252bf93ff).
Leurs exemples de code devraient donc être compatible avec notre propre carte. Et ça semble être compatible avec [le driver SocketCAN](https://en.wikipedia.org/wiki/SocketCAN), qui est open-source et est peut-être déjà installé avec Raspbian, au pire disponible dans les dépôts.

Autre point que l'on a évoqué brièvement il me semble mais qui n'est pas cité ici, est-ce le shield qui fournira du 5V pour alimenter la RPi ?

Et pour ta question concernant le monitoring de l'AU, on peut certainement faire ça avec un pont diviseur de tension effectivement. Par contre, le pont diviseur sera dimensionné pour du LiPo 4S, il ne faudra pas mettre plus de volt que ça sinon, au mieux, la GPIO y passe, au pire la RPi.

Elioty, 2017-12-24:
Ping @astralien3000 : trois questions en suspend (du moins ici, on y a peut-être répondu en réunion)
- XT60 pour l'arrêt d'urgence je suppose ?
- Pour le monitoring, on passe par le deuxième canal de l'AU ou pont diviseur ? (voir dernier paragraphe de mon message précédent)
- Concernant la RPi, elle aura sa propre alimentation/batterie ou je dois prévoir un régulateur 5V sur Chimera pour l'alimenter ?

astralien3000, 2017-12-24:
- Oui, XT60
 - 2e canal (mais pas d'XT60 pour ça)
 - prévoit le régulateur, mais il faut pouvoir le désactiver si besoin. A part si ça interfère pas avec l'alim usb ?

Elioty, 2017-12-26:
- Ok
- Oui, juste une embase comme les autres connecteurs logiques suffira pour ça
- Ok, je vais voir pour ça (soit avec la possibilité d'enlever carrément le régulateur, soit avec un jumper tout con)