xenomorphales/xenomorphales #72: Commande Teensy 3.2

Elioty, 2017-11-07:
Alors d'après le [site officiel](https://forum.pjrc.com/threads/23601-Official-Distributors), il n'y a que trois revendeurs officiels de Teensy en France.

Sur ces trois là, un seul vend des headers 2x7 à monter en surface (pour les pins de dessous), mais uniquement dans un kit avec des sécables "empilables". [Et mouser ne vend pas de headers 2x7 à monter en surface, que des traversants](https://www.mouser.fr/Connectors/Headers-Wire-Housings/_/N-ay0lo?P=1ytibi3Z1z0wxp6Z1ytkn06). Donc deux options (peut-être trois) :
- la [Teensy 3.2 nu](https://snootlab.com/963-teensy-32-usb-fr.html) (option CMS montés) et le [kit de headers](https://snootlab.com/sparkfun/1190-teensy-32-header-kit-fr.html) à souder soit même ; défauts :
  - ça prend de la hauteur pas forcément nécessaire (je ne vois l'utilité que pour brancher un oscilloscope à la rigueur)
  - ils mettent dans le kit deux headers de 13 pins et un de 7 pins pour les contacts à l'extrémité de la carte mais on a besoin des deux dans les angles mais pas des autres (ça aurait mieux si c'était deux headers de 14 pins)
- la [Teensy 3.2 nu](https://snootlab.com/963-teensy-32-usb-fr.html) (option CMS montés), le [kit de headers](https://snootlab.com/sparkfun/1190-teensy-32-header-kit-fr.html) (juste pour le header 2x7 en surface) et [deux barettes de sécables males](https://snootlab.com/lang-fr/connecteurs/305-barrette-secable-16-pts.html) ; défaut :
  - on achète le kit juste pour le header 2x7
- \[À CONFIRMER\] la [Teensy 3.2 avec pins](https://snootlab.com/963-teensy-32-usb-fr.html) (option totalement monté, je leur ai demandé si ça correspond bien à la [Teensy 3.2 pins](https://www.pjrc.com/store/teensy32_pins.html) du site officiel, j'ajouterai un message sur cette issue une fois que j'aurais la réponse) et le [kit de headers](https://snootlab.com/sparkfun/1190-teensy-32-header-kit-fr.html) (juste pour le header 2x7 en surface) ; défaut :
  - on achète le kit juste pour le header 2x7
  - 2€20 plus cher que l'option précédente mais on n'a pas à souder les sécables sur les côtés nous-mêmes

Je n'ai pas cherché le header 2x7 en surface au détail ailleurs, mais à moins de bénéficier de livraison gratuite ou de gros volume, vaut certainement quand même mieux d'acheter le kit complet sur ce site.

Au passage, [ce même site vend ça](https://snootlab.com/pololu/1180-aluminum-standoff-pour-raspberry-pi-11mm-m25-m-f-4-pack-fr.html), ça peut-être utile.

Dernière chose, les [schémas mécaniques](https://www.pjrc.com/teensy/dimensions.html) sur le site officiel ne donne pas les côtes pour les pads de dessous, mais [j'ai trouvé ça](https://www.tindie.com/products/loglow/teensy-32-breakout-revision-d/) qui montre qu'une fois le header 2x7 soudé (correctement) dessous, ça reste sur un quadrillage de 2.54mm. Ce site ne vend d'ailleurs pas la Teensy 3.2 seule, uniquement montée sur ce kit, pour 40$... Et il ne vend pas (encore, bientôt) non plus le kit avec les sécables seuls sans le PCB de ce kit.

astralien3000, 2017-11-13:
Ne "manque"-t-il pas les barette sécables mâles de 1x7 dant ta 2e solution ? Même si tu récupère les 3 pins coupés [du 1x16](https://snootlab.com/lang-fr/connecteurs/305-barrette-secable-16-pts.html), le compte y est pas je crois.

Elioty, 2017-11-13:
> Ne "manque"-t-il pas les barette sécables mâles de 1x7 dant ta 2e solution ? Même si tu récupère les 3 pins coupés du 1x16, le compte y est pas je crois.

Je n'utilise pas les 5 pins à l'extrémité (j'utilise bien les deux dans les angles par contre).

> EDIT: pas de news pour la dernière option ?

Non, toujours pas de réponse... Mais vu que c'est un revendeur officiel, je ne pense pas qu'ils modifient le produit, ça correspond certainement à la Teensy 3.2 pins du site officiel.

astralien3000, 2017-11-15:
J'ai commandé 4 Teensy (option 2: 4x[teensy](https://snootlab.com/963-teensy-32-usb-fr.html) + 4x[header kit](https://snootlab.com/sparkfun/1190-teensy-32-header-kit-fr.html) + 8x[sécables](https://snootlab.com/lang-fr/connecteurs/305-barrette-secable-16-pts.html))

astralien3000, 2017-11-21:
Commande reçue et vérifiée, je clos.

closed 2017-11-21 by astralien3000