xenomorphales/xenomorphales #73: Commande driver moteurs (VNH2SP30)

astralien3000, 2017-11-13:
On s'est donc arrêté sur [cette board](https://www.pololu.com/product/708).

Ça pique bien, le prix, quand même ><
Mais au moins, si elle tient sa promesse, on devrait pas avoir besoin d'en changer avant un bon bout de temps.

Pour le coups j'hésite un peu à en acheter 2 d'un coup...
Peut être 1 à la fois c'est mieux.

Elioty, 2017-11-13:
J'aimerai bien avoir une Hermes 2 de mon côté pour travailler dessus (puis ça nous en fera deux de montée, en cas de besoin, hop, changement rapide) mais je n'aurais pas de moteur et encore moins le robot pour jouer avec les ponts en H donc tu peux en commander une ou deux dans un premier temps, c'est toi qui voit.

astralien3000, 2017-11-14:
J'en ai commandé 2 finalement. Let's see.

astralien3000, 2017-12-08:
Cartes (enfin) reçues, on est bon !

closed 2017-12-08 by astralien3000